arendst/Tasmota #6715: Smaller knx. bin for Sonoff MI I?

jensgulow, 2019-10-21:
I'd like to flash a Sonoff Mini with sonoff-knx.bin . But it's about 568k.....Maximum for the Mini is 508k.
Is it possible to get a smaller version of the knx.bin?

Jason2866, 2019-10-21:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-10-21:
The sonoff mini has enough memory for a full version of Tasmota with KNX.

If you are using the DIY mode for flashing, you can start with sonoff_basic.bin (that is a small version) and then you can upgrade via Tasmota OTA to the full version.

If you need assistance on that, please, find us in the Chat. Thanks.

closed 2019-10-21 by ascillato2

AgentFire, 2021-11-06:
> The sonoff mini has enough memory for a full version of Tasmota with KNX.
>
> If you are using the DIY mode for flashing, you can start with sonoff_basic.bin (that is a small version) and then you can [...]

Where can i download sonoff_basic.bin?

barbudor, 2021-11-06:
Now it is called tasmota-lite.bin

AgentFire, 2021-11-06:
@barbudor and it still bigger than 508k and can't be flashed via DIY tool.

ascillato, 2021-11-06:
http://ota.tasmota.com/tasmota/tasmota-lite.bin is 502K

AgentFire, 2021-11-06:
> http://ota.tasmota.com/tasmota/tasmota-lite.bin is 502K

513,264 bytes.
[image]

barbudor, 2021-11-06:
Pfff. Sonoff DIY is so unreliable. 
Use serial flashing

Jason2866, 2021-11-06:
The have there own definition of kbyte 😁

barbudor, 2021-11-06:
V9.3.1 is smaller
http://ota.tasmota.com/tasmota/release-9.1.0/tasmota-lite.bin

AgentFire, 2021-11-06:
> V9.3.1 is smaller http://ota.tasmota.com/tasmota/release-9.1.0/tasmota-lite.bin

Thanks. Can I hope that you'd put the link on earlier, historic, releases on the MD file?

barbudor, 2021-11-06:
Where?
You can propose doc updates by clicking the pencil icon on the top right