arendst/Tasmota #6721: VS Code con Ubuntu 19.04 - error: forward declaration of 'struct rst_info'

snakuzzo, 2019-10-22:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Error compiling 6.6 on VS Code
  

### REQUESTED INFORMATION

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 6.6
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VS Code
- [ ] Flashing tools used: _____

- [x] Provide the output of the Console log output when you experience your issue; if applicable:
```
Console output here:
  
Processing sonoff (platform: espressif8266@1.5.0; framework: arduino; board: esp01_1m)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.5.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 1.20300.1 (2.3.0), tool-esptool 1.409.0 (4.9)
Converting sonoff.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 62 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <BearSSL> 0.6
|-- <Adafruit CCS811 Library> 1.0.0
|   |-- <Wire> 1.0
|-- <Adafruit MAX31865 library> 1.0.1
|   |-- <SPI> 1.0
|-- <Adafruit SGP30 Sensor> 1.0.3
|   |-- <Wire> 1.0
|-- <I2Cdevlib-Core>
|   |-- <Wire> 1.0
|-- <I2Cdevlib-MPU6050>
|   |-- <I2Cdevlib-Core>
|   |   |-- <Wire> 1.0
|-- <Mutichannel_Gas_Sensor> 0.0.1
|   |-- <Wire> 1.0
|-- <VL53L0X> 1.0.2
|   |-- <Wire> 1.0
|-- <C2Programmer> 1.0.0
|-- <A4988_Stepper> 0.0.1
|-- <I2Cdevlib-ADS1115>
|   |-- <I2Cdevlib-Core>
|   |   |-- <Wire> 1.0
|-- <Adafruit GFX Library> 1.5.6
|   |-- <SPI> 1.0
|-- <Adafruit ILI9341> 1.2.0
|   |-- <Adafruit GFX Library> 1.5.6
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit LED Backpack Library> 1.1.6
|   |-- <Adafruit GFX Library> 1.5.6
|   |   |-- <SPI> 1.0
[... 281 lines cut ...]
sonoff-ircustom  IGNORED
sonoff-BG        IGNORED
sonoff-BR        IGNORED
sonoff-CN        IGNORED
sonoff-CZ        IGNORED
sonoff-DE        IGNORED
sonoff-ES        IGNORED
sonoff-FR        IGNORED
sonoff-GR        IGNORED
sonoff-HE        IGNORED
sonoff-HU        IGNORED
sonoff-IT        IGNORED
sonoff-KO        IGNORED
sonoff-NL        IGNORED
sonoff-PL        IGNORED
sonoff-PT        IGNORED
sonoff-RU        IGNORED
sonoff-SE        IGNORED
sonoff-SK        IGNORED
sonoff-TR        IGNORED
sonoff-TW        IGNORED
sonoff-UK        IGNORED
============================================================== 1 failed, 0 succeeded in 00:00:04.848 ==============================================================
The terminal process terminated with exit code: 1
```

meingraham, 2019-10-22:
Does 6.6.0.21 compile OK?

As this is a question regarding compilation (see original Issue template), please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

snakuzzo, 2019-10-22:
I've cloned the project few minutes ago...and compilation is ok just with default core...using core 2.3.0 I've got errors

closed 2019-10-22 by snakuzzo

snakuzzo, 2019-10-22:
As told me on discord chat, 2.3.0 is no longer supproted :(

arendst, 2019-10-23:
Most of the current version is still supported on core 2.3.0

This issue is caused by me tidying up too much. I will restore the cause after which you must be able to compile with core 2.3.0 again now.