arendst/Tasmota #6722: Hide alexa objects

prstprst, 2019-10-23:
Hello, 

When using more than one power output in tasmota and choosing the hue emulation (providing also the dimming possibility), alexa detect ALL the "power" existing in the config of tasmota.
I know that it's possible to delete i from the Alexa app, but by making a new search for new devices, uneeded device are coming back in the Alexa app.
The goal should be to have the possibility to make some power  invisible for the alexa detection by adding, for exemple, a simple "$" (has used for hidden share in windows) in front of the frinedly name used in the "hue" config .

Thanks

s-hadinger, 2019-10-23:
This change would have some impacts in the way Alexa devices are reported and controlled. Also take into consideration that friendly names work only for the first 4 devices. Your use-case seems very specific. It will probably not be implemented in the standard version. But feel free to submit a PR if you like.

Jason2866, 2019-10-23:
If you do a PR it has a low chance to be merged.
The common wanted use case is to control all connected devices via Alexa.

PR #6798 (2019-10-31): Add hide Alexa objects with friendlyname starting with '$'
## Description:

After second thoughts, it was actually easy to hide specific objects from Alexa.
Objects with friendly-names starting with dollar sign '$' will not be discovered by Alexa. If they were already discovered previously, they will still continue to work.

As there are only 4 friendly names, if friendlyname-4 starts with '$', all devices after device number 4 (included) will not be discovered by Alexa.

**Related issue (if applicable):** fixes #6722 #6762 

## Checklist:
  - [x] The pull request is done against the latest de [... cut]
closed 2019-10-23 by ascillato2