arendst/Tasmota #6724: Insistencies in page Sonoff RF Bridge 433

bwims, 2019-10-23:
This is really a description of a bug in the web page!  The video shows how to flash the RF Bridge, it is clearly an R2, yet there appears to be no need to cut tracks or plug jumpers in the reverse of the board, as is described further down the page.  There is another video by Dr Zzzzs where he flashes an R2 without cutting tracks or plugging in jumpers to connect lines at the back of the board.

Sorry to post this here, but I did not know where else to post it!

ascillato, 2019-10-23:
As explained in the wiki, cutting the tracks are ONLY if you are going to power the bridge using the USB connector when you are going to flash. If you don't want to cut those tracks, you need to power your bridge from the pin headers and NOT using the USB connector that will interfere with the flashing process.

ascillato2, 2019-10-23:
>I did not know where else to post it

Remember that you have also the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). 

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-23 by ascillato2