arendst/Tasmota #6726: Configure logging incorrect

mscreations, 2019-10-24:
### BUG DESCRIPTION
According the the wiki command list, `WebLog 5` is a valid command to get ALL messages. The device will accept this command via console, mqtt, and from an HTTP request. However, the Web UI does not show that it is set and instead shows it as "0 None".

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): **Sonoff Basic**
- [x] Tasmota binary firmware version number used: **6.6.0.21**
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: **OTA**
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
  {"Module":{"1":"Sonoff Basic"}}
  {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"9":"Switch1"}}
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [x] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  STATUS = {"Status":{"Module":1,"FriendlyName":["Porchlight"],"Topic":"porchlight","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"porchlight","SwitchMode":[2,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
  STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T00:24:06","StartupUTC":"2019-10-23T01:43:58","Sleep":50,"CfgHolder":4617,"BootCount":59,"SaveCount":1519,"SaveAddress":"F7000"}}
  STATUS2 = {"StatusFWR":{"Version":"6.6.0.21(7ecc752-sonoff)","BuildDateTime":"2019-10-22T19:00:13","Boot":6,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
  STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":5,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AluminumFalcon",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
  STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
  STATUS5 = {"StatusNET":{"Hostname":"porchlight-6139","IPAddress":"xxx.xxx.xxx.xxx","Gateway":"xxx.xxx.xxx.xxx","Subnetmask":"255.255.255.0","DNSServer":"xxx.xxx.xxx.xxx","Mac":"EC:FA:BC:07:77:FB","Webserver":2,"WifiConfig":5}}
  STATUS6 = {"StatusMQT":{"MqttHost":"xxx.xxx.xxx.xxx","MqttPort":1883,"MqttClientMask":"SON_%06X","MqttClient":"xxxxxxxx","MqttUser":"xxxxxxxx","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
  STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 24 02:08:04 2019","Local":"Wed Oct 23 22:08:04 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":"-04:00","Sunrise":"07:39","Sunset":"18:28"}}
  STATUS10 = {"StatusSNS":{"Time":"2019-10-23T22:08:04","Switch1":"OFF"}}
  STATUS11 = {"StatusSTS":{"Time":"2019-10-23T22:08:04","Uptime":"1T00:24:06","UptimeSec":87846,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"AluminumFalcon","BSSId":"68:9C:E2:A0:DD:72","Channel":6,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:05"}}}
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

### TO REPRODUCE
Set any of the logging methods to level 5 via console or MQTT command. Configuration WebUI shows it is set to "0 None"

### EXPECTED BEHAVIOUR
Correct level would be displayed

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I'm pretty sure I know what needs changed in the code, but I don't have a computer that I can install software on in order to compile the code to try it (only have a severely locked down business laptop at the moment). 
https://github.com/arendst/Sonoff-Tasmota/blob/247355a12579790aecd14cce602fe1d77087b29a/sonoff/xdrv_01_webserver.ino#L1628
Should be `i <= LOG_LEVEL_ALL`

https://github.com/arendst/Sonoff-Tasmota/blob/247355a12579790aecd14cce602fe1d77087b29a/sonoff/xdrv_01_webserver.ino#L503
Needs to add the text for All Messages

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-10-24:
I have not tested on 6.6.0.21 (the latest released version). My "newest" version is 6.6.0.16. I can set log level `5` from the Console and through MQTT and the level gets set to `5` as expected.

mscreations, 2019-10-24:
It is getting set correctly, as after setting the log level `5`, `Status3` correctly shows it as being set to `5`. The issue is with the Web UI not displaying the level `5` option after being set to `5`.

``` 
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":5,"MqttLog":0,"SysLog":0,"LogHost":"server01.mscreations.local","LogPort":514,"SSId":["AluminumFalcon",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
```
 
[image]

arendst, 2019-10-24:
There never was a "log leve all" used within the code so from now on the max log level is "More Debug" with number 4.

Needs a wiki update, remove option 5 pls.

Jason2866, 2019-10-24:
Loglevel 5 removed in wiki

mscreations, 2019-10-24:
That works for me. I wasn't really sure if there was even anything extra gained from using all messages, so this removes that oddness.

closed 2019-10-24 by mscreations