arendst/Tasmota #6738: Water Pump 10 Seconds ON / 8 Minutes off (Infinite Cicle repeating)

webxtekagency, 2019-10-24:
Hi,

I already tried many things without any success, and i can't find any guide from anyone that already done this, so i really need a hand.
I got a water pump that i need to be cycling 10 seconds On and 8 minutes OFF, and i'm currently using a WATER TIMER IR 8A (GSE-8) for that purpose, but i costs more than 100€, and after one year it fails sometimes, if i got to replace it its hard to have it on hands because it needs to be shipped over, thats why i want to set a sonoff to do this, its cheaper and easier to get a new one if it fails.

Anyway, i got today a Sonoff Pow R2, i flashed it with the latest Tasmota, but i need some help setting it up.

Thanks

meingraham, 2019-10-24:
Going to GitHub for this isn't going to help your cause. Keep asking on Discord... in the public channels.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-25 by ascillato2

ascillato2, 2019-10-25:
From Discord
```
pulsetime1 100
rule1 1
rule1 on power1#boot do backlog power1 1; ruletimer1 490 endon on Rules#Timer=1 do backlog power1 1; ruletimer1 490 endon
```