arendst/Tasmota #6740: Module automatically restarts after setting `displaywidth 64`

ajitam, 2019-10-25:
### BUG DESCRIPTION
Module automatically restarts after setting `displaywidth 64`

### REQUESTED INFORMATION

* Device used (e.g., Sonoff Basic): `WeMos D1` + `64 x 48 pixel OLED display SSD1306 `
* Self-compiled
* IDE / Compiler used: `pio`
* Flashing tools used: `pio run -t upload`

* Console log output 
```
12:28:20 CMD: display
12:28:20 MQT: /device/sonoff-blank/stat/RESULT = {"Display":{"Model":2,"Width":128,"Height":64,"Mode":0,"Dimmer":15,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[1,8],"Rows":1}}
12:28:33 CMD: displaywidth
12:28:33 MQT: /device/sonoff-blank/stat/RESULT = {"DisplayWidth":128}
12:28:39 CMD: displaywidth 64
12:28:39 MQT: /device/sonoff-blank/stat/RESULT = {"DisplayWidth":64}
12:28:41 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v645464ca
~ld

00:00:00 CFG: Loaded from flash at F7, Count 49
00:00:00 Project sonoff-blank sonoff-X-X Version 6.6.0.21(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 ...
```

### TO REPRODUCE
in console write

```
displaywidth 64
```

### EXPECTED BEHAVIOUR
 Should not restart? :)

p.s.: never been so scared to open the issue :crossed_fingers:

ascillato, 2019-10-25:
Hi, please update to latest Tasmota version

ascillato, 2019-10-25:
If after updating, your issue remains, do a reset 5 to delete all flash settings and start over.

If that don't solve your issue, please complete the template. You have deleted all the useful information we need (Tasmota version, status 0 output, etc etc.) without that information we can't help you

ascillato2, 2019-10-25:
Looking into the code, the restart is ok to apply the new settings.

ajitam, 2019-10-25:
1. I have updated the issue description (sorry)
2. I updated the SW (to 6.7.0(sonoff))
3. i did `restart 5`

If restart is OK, then issue is that settings are not stored

ajitam, 2019-10-25:
```
13:00:54 CMD: display
13:00:54 MQT: /device/sonoff/stat/RESULT = {"Display":{"Model":2,"Width":128,"Height":64,"Mode":1,"Dimmer":1,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
13:01:02 CMD: displaywidth 64
13:01:02 MQT: /device/sonoff/stat/RESULT = {"DisplayWidth":64}
13:01:03 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v645464ca
~ld

00:00:00 CFG: Loaded from flash at F9, Count 19
00:00:00 Project sonoff sonoff-X-X Version 6.7.0(sonoff)-STAGE
00:00:04 WIF: Connected
13:01:10 MQT: Attempting connection...
13:01:10 MQT: Connected
13:01:10 MQT: /device/sonoff/tele/LWT = Online (retained)
13:01:10 MQT: /device/sonoff/cmnd/POWER =
13:01:10 MQT: /device/sonoff/tele/INFO1 = {"Module":"Generic","Version":"6.7.0(sonoff)","FallbackTopic":"cmnd/sonoff_2DB424_fb/","GroupTopic":"sonoffs"}
13:01:10 MQT: /device/sonoff/tele/INFO2 = ...
13:01:10 MQT: /device/sonoff/tele/INFO3 = {"RestartReason":"Software/System restart"}
13:01:10 MQT: /device/sonoff/stat/RESULT = {"POWER":"OFF"}
13:01:10 MQT: /device/sonoff/stat/POWER = OFF
13:01:16 CMD: display
13:01:16 MQT: /device/sonoff/stat/RESULT = {"Display":{"Model":2,"Width":128,"Height":64,"Mode":1,"Dimmer":1,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}

```

ascillato, 2019-10-25:
You did a `restart 5`,sorry that is not a Tasmota command.

Please, do a `reset 5` to clear flash memory for any wrong remaining data from Initial flash and start over with the configuration. Reset 5 will preserve your wifi settings.

ascillato, 2019-10-25:
And for testing (just yo avoid any compilation issue) please update to the precompiled bin with display support: https://github.com/arendst/Sonoff-Tasmota/releases/download/v6.7.0/sonoff-display.bin

You can update by OTA (from Tasmota web UI)

ajitam, 2019-10-28:
I tried everything with no luck (I will try a new wemos).

arendst, 2019-10-28:
I'll have a look now...

arendst, 2019-10-28:
The current code does only support 96 or 128 pixels.

I'll change it to support 64 too and let you know when to test.

arendst, 2019-10-28:
Sorry for you but the used Adafruit display driver library does NOT support width 64 x 48.

The only supported sizes are 96 x 16, 128 x 32 and 128 x 64

So I cannot solve this.

arendst, 2019-10-28:
As you compile yourself you might want to try this:

In file ``lib/Adafruit_SSD1306-1.3.0-gemu-1.1/Adafruit_SSD1306.cpp`` change lines 566 to 568 from 
```
  } else {
    // Other screen varieties -- TBD
  }
```
in 
```
  } else if((WIDTH == 64) && (HEIGHT == 48)) {
    static const uint8_t PROGMEM init4d[] = {
      SSD1306_SETCOMPINS,                 // 0xDA
      0x12,
      SSD1306_SETCONTRAST };              // 0x81
    ssd1306_commandList(init4d, sizeof(init4d));
    ssd1306_command1((vccstate == SSD1306_EXTERNALVCC) ? 0x10 : 0xCF);
  } else {
    // Other screen varieties -- TBD
  }
```
and in file ``sonoff/xdsp_02_ssd1306.ino`` lines 64 to 69 in
```
    if ((Settings.display_width != 64) && (Settings.display_width != 96) && (Settings.display_width != 128)) {
      Settings.display_width = 128;
    }
    if ((Settings.display_height != 16) && (Settings.display_height != 32) && (Settings.display_height != 48) && (Settings.display_height != 64)) {
      Settings.display_height = 64;
    }
```
and report back the results.

ajitam, 2019-10-29:
aha ok, there was no note in the documentation about this.
In the meanwhile I "guess" the offset so it is not that hard to just offset the content without changing the code, for example:

```
DisplayText [z];
DisplayText [x33y0f1]Line 1
DisplayText [x33y11r64:1]
DisplayText [x33y13f1]Line 2
DisplayText [x33y24r64:1]
DisplayText [x33y26f1]Line 3
DisplayText [x33y37r64:1]
DisplayText [x33y39f1]Line 4
```

Just `x` is offset for `33`

@mike2nl so if understand correctly you will push the code to support 64x48?

@arendst do I close this?

arendst, 2019-10-29:
Pls leave open for me to remember to put the change in the production code.

mike2nl, 2019-10-29:
@ajitam 
because a PR...
- When @arendst will add it i don't have todo a PR because he is the master of all code ;-).

ascillato2, 2019-10-30:
@ajitam

Hi,

Your request has been added. Please, test it and if it works, close this issue. Thanks.

mike2nl, 2019-10-31:
@ascillato2 
i can test it too because i have all the available hardware for the D1 Mini series shileds.
Until the weekend it must be done and i will report so far the issue is still open.

ascillato2, 2019-10-31:
Closing this issue as this feature has been added. Thanks :+1:

closed 2019-10-31 by ascillato2

ajitam, 2019-11-04:
Width and height can be set now 
```
10:17:10 MQT: /device/sonoff-blank/stat/RESULT = {"Display":{"Model":2,"Width":64,"Height":48,"Mode":0,"Dimmer":15,"Size":1,"Font":1,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":2}}
```

But when I do `DisplayText [x0y0r64:48]`, square is drawn from `x:32` `y:0`. In other words `x` axis is at the middle of the screen...

Is there any option/command for this?

krijnenr, 2019-11-08:
I can confirm x0y0 is off screen (shift 32 pixels)

Updated the Adafruit_SSD1306.cpp file with code below

void Adafruit_SSD1306::display(void) {
  uint8_t col_start = 0;
  uint8_t col_end = WIDTH - 1;
  if (WIDTH == 64) {
    col_start += 32;
    col_end += 32;
  }

  TRANSACTION_START
  static const uint8_t PROGMEM dlist1[] = {
    SSD1306_PAGEADDR,
    0,                         // Page start address
    0xFF,                      // Page end (not really, but works here)
    SSD1306_COLUMNADDR };
  ssd1306_commandList(dlist1, sizeof(dlist1));
  ssd1306_command1(col_start); // Column start address
  ssd1306_command1(col_end); // Column end address

tdemalliard, 2020-01-02:
I confirm **Krijnenr** solution is working for me : wemos d1 mini + oled 64x48 v1.1.0. 

I compiled it from release 8.1.0.1.

xbmcnut, 2020-03-11:
@tdemalliard Can you please confirm your `display` settings for me? I'm using the same display but `DisplayText [x0y0r64:48]` start the draw well of to the left of the screen. If I try `DisplayText [x33y39f1]Line 4`, I just get _Line_ and nothing else.

PR #8012 (2020-03-27): Add support for 64x48 SSD1306 OLED (#6740)
## Description:

Final change to support 64x48 SSD1306 OLED displays. The code was already by @krijnenr but never merged into development. I confirm it works with 64x48 OLED display (Wemos D1 shield from Aliex)

**Related issue (if applicable):** fixes #6740 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core Tasmota_core_stage
  - [x] The code change pass travis tests. **Your PR cannot  [... cut]