arendst/Tasmota #6741: 6.7 bug with pzem modules

Smandurlo, 2019-10-25:
New Tasmota 6.7 uploaded on 2 sonoff (first minimal), one with a pzem 004T and another one with pzem 016. Both of them are not able to use the pzem modules anymore.

INFO3 on the console: {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4000dce5 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

Everything else is working as expected.

arendst, 2019-10-25:
Pls provide more information. Need output of command ``status 0`` and ``gpio all``

Smandurlo, 2019-10-25:
15:45:09 MQT: stat/pzem2/STATUS = {"Status":{"Module":1,"FriendlyName":["Pzem Mansarda"],"Topic":"pzem2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
15:45:09 MQT: stat/pzem2/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/020502/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:40:44","StartupUTC":"2019-10-25T13:04:25","Sleep":50,"CfgHolder":4617,"BootCount":179,"SaveCount":728,"SaveAddress":"F7000"}}
15:45:09 MQT: stat/pzem2/STATUS2 = {"StatusFWR":{"Version":"6.7.0(sonoff)","BuildDateTime":"2019-10-25T12:28:03","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
15:45:09 MQT: stat/pzem2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VillaTorreZisa",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000080A9","2805C8000100060000005A00000000000000","00000000"]}}
15:45:09 MQT: stat/pzem2/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":436,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:45:09 MQT: stat/pzem2/STATUS5 = {"StatusNET":{"Hostname":"s-pzem-mansarda-20","IPAddress":"192.168.178.20","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"80:7D:3A:46:FC:8C","Webserver":2,"WifiConfig":5}}
15:45:09 MQT: stat/pzem2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.26","MqttPort":1883,"MqttClientMask":"pzem","MqttClient":"pzem","MqttUser":"Felice","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:45:09 MQT: stat/pzem2/STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 25 13:45:09 2019","Local":"Fri Oct 25 15:45:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:29","Sunset":"18:07"}}
15:45:09 MQT: stat/pzem2/STATUS10 = {"StatusSNS":{"Time":"2019-10-25T15:45:09"}}
15:45:09 MQT: stat/pzem2/STATUS11 = {"StatusSTS":{"Time":"2019-10-25T15:45:09","Uptime":"0T00:40:44","UptimeSec":2444,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"VillaTorreZisa","BSSId":"E8:DF:70:4C:2A:5B","Channel":1,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:06"}}}

Smandurlo, 2019-10-25:
15:52:32 MQT: stat/pzem2/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

If I set up the pzem as before (working), I get that error and the gpio configuration is lost

Smandurlo, 2019-10-25:
[image]

This is the config that worked in both my sonoff (of course the second one had the pzem 004t)

ascillato, 2019-10-25:
Please, go to the console and type `reset 5` and configure again your device

arendst, 2019-10-25:
Just went from development 6.7.0 to release sonoff.ino 6.7.0 and caught the exception too.

Investigating...

arendst, 2019-10-25:
Considering the fact that both PZems and Pow R2 are troublesome I expect the issue to be with the serial interface.

Still investigating why my development version ran fine but the production doesn't ...

arendst, 2019-10-25:
Right.

The issue is within energy management only. I changed margin detection (absolute power delta rings a bell) and power limit and there I made an error.

Working on it...

ascillato2, 2019-10-25:
Closing issue as it has been fixed.

closed 2019-10-25 by ascillato2