arendst/Tasmota #6742: Tasmota v.6.7.0 with Sonoff POW-R2

adharyanto, 2019-10-25:
After updating from Tasmota v.6.6.0 to v.6.7.0, my Sonoff POW R-2 does not work properly; all measurement are missing.....
It is a bug or something else?

> **GUIDE**  
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>   
> Please DO NOT OPEN AN ISSUE:  
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-10-25:
And where is the requested information?

adharyanto, 2019-10-25:
Hi,

 

Sorry, I am still confused with the reporting format.

Probably I made mistake with the report.

But what I am trying to say, is that;

-        After updating to v6.7.0 (from v.6.6.0) my Sonoff POW-R2 become “always” Sonoff Basic, and all measurement are disappeared

-        I tried to change from “basic” to “pow-r2”, and restart….it did not help….always become “basic”

 

Ragards,

Antonius

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

danmed, 2019-10-25:
Same issue for me with a Teckin SP23 power monitoring plug (Although i think i've now killed it trying to flash it back to 6.6.0)

ascillato, 2019-10-25:
@adharyanto and @danmed 

Please, provide the information requested. Without that we CAN'T HELP.

ascillato, 2019-10-25:
@adharyanto

Please, go to the console and type `reset 5` and configure again your device

@danmed

If you can't reach your device anymore, please erase all flash with esptool.py as explained in the wiki and flash again v6.7 by serial using esptool.py as explained in the wiki.

adharyanto, 2019-10-25:
Hi,
After go to Console and type RESET 5, now I cannot access online the Sonoff POW-R2.
I checked the documentation, RESET 5 is = Erase all flash from program end to end of physical flash and reset the parameters to what is configured in user_config(_override).h.........***
Now totally I cannot access the Sonoff POW R2....

I think I have to flash it manually (offline), but the Sonoff POW R2 is now running "live", and I cannot pull this out, as it will interrupt the electricity...where many equipment is running..... I will have to find the best time to take down the electricity, and minimize the outage....

danmed, 2019-10-25:
> 
> 
> @adharyanto
> 
> Please, go to the console and type `reset 5` and configure again your device
> 
> @danmed
> 
> If you can't reach your device anymore, please erase all flash with esptool.py as explained in the wiki and flash again v6.7 by serial using esptool.py as explained in the wiki.

Would love to , but i don't have the facilities to do that. i was only posting to show there was more than 1 person with the issue. Discussion about it on the discord too.

TylerDurden23, 2019-10-25:
I have the same Issue with the POW R2 after upgrading to 6.7.

ascillato, 2019-10-25:
@adharyanto

reset 5 will preserve your wifi settings. So if you can't reach your device, you are having another issue. May be corrupt OTA ? please erase all flash with esptool.py as explained in the wiki and flash again v6.7 by serial using esptool.py as explained in the wiki.

@TylerDurden23 

Please, in order to troubleshoot your issue, please post here the troubleshooting template completed with as much information as you can get. We need more information to solve your issue. My devices (including POW R2) were updated to 6.7 with no issue. That is why we need more info.

adharyanto, 2019-10-25:
Tonight (in my timezone), I updated 7x Sonoff Tasmota modules, 6x of them worked fine, several types of Sonoff and esp8266, but the Sonoff POW R2 failed, all of them via OTA.
Yes, if doing it offline (not OTA), normally I can erase & re-flash either using esptool.py or esp8266 download tool. But tonight I cannot do this, because the POW R2 is running, live, and I cannot pull this out, it will interrupt the electricity, many equipment is running using the electricity..... I need to find the best time to shut down the electricity...

blakadder, 2019-10-25:
Following these reports I took out an old POW R1 running on 6.5 and used the http://thehackbox.org/tasmota/sonoff.bin via ota upgrade.

After the upgrade was finished the device was stuck as `module 1` per above reports with this output
```
17:22:43 CMD: module 6
17:22:43 MQT: stat/pow/RESULT = {"Module":{"6":"Sonoff Pow"}}
00:00:00 CFG: Loaded from flash at F5, Count 64
00:00:00 Project sonoff Sonoff Version 6.7.0(e4b0323-sonoff)-STAGE
00:00:00 WIF: Connecting to AP2 HTPC in mode 11N as pow-3799...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on pow-3799 with IP address 10.1.1.132
17:23:02 MQT: Attempting connection...
17:23:02 MQT: Connected
17:23:02 MQT: tele/pow/LWT = Online (retained)
17:23:02 MQT: cmnd/pow/POWER = 
17:23:02 MQT: tele/pow/INFO1 = {"Module":"Sonoff Basic","Version":"6.7.0(e4b0323-sonoff)","FallbackTopic":"cmnd/DVES_92CED7_fb/","GroupTopic":"sonoffs"}
17:23:02 MQT: tele/pow/INFO2 = {"WebServerMode":"Admin","Hostname":"pow-3799","IPAddress":"10.1.1.132"}
17:23:02 MQT: tele/pow/INFO3 = {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4000dce5 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
17:23:02 MQT: stat/pow/RESULT = {"POWER":"OFF"}
```
after `reset 5` the device rebooted without issues, i entered all the mqtt data and the usual backlog and then proceeded to successfully change module to `module 6`
Energy Total and Energy Today readings persisted with same values but I didn't take note of the values before the upgrade so I cannot guarantee they're the same

arendst, 2019-10-25:
See issue #6741

TylerDurden23, 2019-10-25:
Reset 5 and then setting it to POW R2 dd help with 6.7

adharyanto, 2019-10-25:
> See issue #6741

Ok, thanks for the fixed....
But now I just cannot access remotely my POW R2, it was disconnected after the Reset-5. I will try again probably tomorrow or Sunday, to minimize outage... By offline flashing.... (not OTA)

ascillato2, 2019-10-25:
Closing issue as it has been fixed.

closed 2019-10-25 by ascillato2