arendst/Tasmota #6745: Release 6.7 on hold - removed

arendst, 2019-10-25:
Some nasty issues popped up with release 6.7

I need more time to investigate.

For now, do not install release 6.7

vampywiz17, 2019-10-25:
Yes, Unfortunately my POW R2 is dead :( I try to reflash the 6.6 but not working well...(it restart every 5 sec....)

ultrara1n, 2019-10-25:
> Yes, Unfortunately my POW R2 is dead :( I try to reflash the 6.6 but not working well...(it restart every 5 sec....)

You should try to erase memory before/while flashing, if there is no hardware error, flashing 6.6 should be successful.

blakadder, 2019-10-25:
Did you erase flash before serial flashing it so its a clean flash?

arendst, 2019-10-25:
Unstable release caused by recent energy monitoring code change. See #6741 

Will recompile and release soon.

arendst, 2019-10-25:
And Re-released at https://github.com/arendst/Sonoff-Tasmota/releases

ultrara1n, 2019-10-25:
Working fine with Gosund SP111 (BlitzWolf SHP-6) again, thanks Theo!

raymajor, 2019-10-25:
[image]

Works, Thanks Guys..

vampywiz17, 2019-10-25:
My POW R2 is gone... I try to reflash it (new 6.7) erase flash, use 4 Mb zeroed. bin (espeasy) but it not help at all... 

The ping is:

[image]

No other idea....

ascillato, 2019-10-25:
@vampywiz17 

Please, instead of espeasy zeroed 4M binary in a 2MB device, try the instructions of Tasmota wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Flashing

Jason2866, 2019-10-25:
@vampywiz17  writing a image with data is not a flash erase!
Use esptool.py for erasing flash. Follow the link from @ascillato

vampywiz17, 2019-10-25:
@ascillato @Jason2866 

I do this:

1. Start NodeMCU-PyFlasher-4.0-x64

2. check Yes wipe all data

3. select sonoff.bin

Everything ok. I connect it the network, working well., no ping problem. It say it is a Sonoff Basic. (Default) I go to menu, set Sonoff POW R2 and ping problem.... set it again to basic and ping problem is gone...

I think something is happened with the I think something is happened with the measurement chip....

Jason2866, 2019-10-25:
`It is dead Jim`

ascillato, 2019-10-25:
@vampywiz17 

Are you configuring your settings mannually or uploading an old config file?

arendst, 2019-10-25:
If it returns as basic what are the console messages?

helmar74, 2019-10-25:
Just for your information:
Updated successfully several devices to 6.7.0 (re-released) version:
Sonoff SC, Sonoff S20 & S26, BlitzwoIF SHP2 & SHP6, YTW IR Bridge, Sonoff 4CH Pro, Sonoff 433 MHz Bridge, several Wemos Mini with WS2811 GRBW, DHT21, TX-20, DS18x20, TSL-2561, BPM680, Displays.....
Sonoff T1

andrethomas, 2019-10-25:
I had a POW do fastboots until it ended up on basic when using the first release of 6.7.0 - I did not have it hooked up to a serial monitor as it is installed but this is my assumption based on the reboots and the relay clicking that was going on at the time. May or may not be related. Needless to say with the last release of 6.7.0 this problem is not occurring.

rcmcronny, 2019-10-25:
No luck for me with 6.7.0 and Blitzwolf SHP2  (first Gosund SP1 Version compatible to the Blitzwolf SHP2) works with 6.6.0 perfectly. With 6.7.0 it get reboots after some seconds if module is set to blitzwolf, if it is set to sonof basic it works (without measurement) weired .. i loaded the 2nd release file from 3 hours ago.
I will stick to 6.6.0 for now and follow this issue here

helmar74, 2019-10-25:
> No luck for me with 6.7.0 and Blitzwolf SHP2 (first Gosund SP1 Version compatible to the Blitzwolf SHP2) works with 6.6.0 perfectly. With 6.7.0 it get reboots after some seconds if module is set to [...]

Was the same for me with one of my SHP2. I noted my settins, then I reseted my configuration and did all settings again, than it worked.....

Jason2866, 2019-10-25:
Just updated my Gosund SP1 (first version) from 6.6.0.12 to 6.7.0. Works.
```
00:00:00 CFG: Loaded from flash at FB, Count 830
00:00:00 Project sonoff Verbrauch Version 6.7.0(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-16CE26-3622...
00:00:02 RSL: tele/sonoff-16CE26/MARGINS = {"PowerHigh":"ON"}
00:00:02 RSL: tele/sonoff-16CE26/SENSOR = {"Time":"1970-01-01T00:00:02","ENERGY":{"TotalStartTime":"2019-03-01T16:48:41","Total":179.717,"Yesterday":1.326,"Today":0.964,"Power":46,"ApparentPower":84,"ReactivePower":70,"Factor":0.55,"Voltage":232,"Current":0.362}}
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-16CE26-3622 with IP address 192.168.2.158
00:00:05 UPP: Multicast (re)joined
21:55:59 MQT: Attempting connection...
21:55:59 MQT: Connected
21:55:59 MQT: tele/sonoff-16CE26/LWT = Online (retained)
21:55:59 MQT: cmnd/sonoff-16CE26/POWER = 
21:55:59 MQT: tele/sonoff-16CE26/INFO1 = {"Module":"BlitzWolf SHP","Version":"6.7.0(sonoff)","FallbackTopic":"cmnd/sonoff-16CE26_fb/","GroupTopic":"sonoffs"}
21:55:59 MQT: tele/sonoff-16CE26/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-16CE26-3622","IPAddress":"192.168.2.158"}
21:55:59 MQT: tele/sonoff-16CE26/INFO3 = {"RestartReason":"Software/System restart"}
21:55:59 MQT: stat/sonoff-16CE26/RESULT = {"POWER":"ON"}
21:55:59 MQT: stat/sonoff-16CE26/POWER = ON
21:55:59 UPP: Multicast (re)joined
```

vampywiz17, 2019-10-25:
> @vampywiz17
> 
> Are you configuring your settings mannually or uploading an old config file?

I configure manually.

Now I put this to trash and order a new one...

ascillato2, 2019-10-25:
Closing this issue as the release has been fixed. Thanks.

closed 2019-10-25 by ascillato2

pierch, 2019-10-26:
I get 5 consecutive reboots with my SHP2 rev 2.3 (configured as Gosund SP1 v23, SPH configuration does not work) usually under load. Console shows exception:
`{"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4000dce5 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}`

Jason2866, 2019-10-26:
@pierch  execute command `reset 5` if you have the chance to.
Do NOT reimport saved config. Configure device manually.
If you have no chance do so, do a serial flash. Erase flash with esptool.py and flash Tasmota
Configure it (do Not used saved settings).

arendst, 2019-10-26:
@pierch pls provide output of command status 0

pierch, 2019-10-26:
> @pierch execute command `reset 5` if you have the chance to.
> Do NOT reimport saved config. Configure device manually.
> If you have no chance do so, do a serial flash. Erase flash with esptool.py and flash Tasmota
> Configure it (do Not used saved settings).

You are probably correct. I did reimport configs after doing `reset 5`. Now I've configured manually and so far it did not crash/restart. I will test it for a few days before move it back to use. Thank you!

pierch, 2019-10-26:
> @pierch pls provide output of command status 0

For reference, here is the output:
`16:07:14 CMD: status 0
16:07:14 MQT: stat/blitz/STATUS = {"Status":{"Module":55,"FriendlyName":["Blitz"],"Topic":"blitz","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:07:14 MQT: stat/blitz/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:44","StartupUTC":"2019-10-26T15:01:30","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":21,"SaveAddress":"F7000"}}
16:07:14 MQT: stat/blitz/STATUS2 = {"StatusFWR":{"Version":"6.7.0(sonoff)","BuildDateTime":"2019-10-25T18:35:38","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
16:07:14 MQT: stat/blitz/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["****",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
16:07:14 MQT: stat/blitz/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:07:14 MQT: stat/blitz/STATUS5 = {"StatusNET":{"Hostname":"****","IPAddress":"*****","Gateway":"****","Subnetmask":"****","DNSServer":"****","Mac":"******","Webserver":2,"WifiConfig":4}}
16:07:14 MQT: stat/blitz/STATUS6 = {"StatusMQT":{"MqttHost":"*****","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C3530C","MqttUser":"*****","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:07:15 MQT: stat/blitz/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 26 15:07:14 2019","Local":"Sat Oct 26 16:07:14 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:27","Sunset":"17:41"}}
16:07:15 MQT: stat/blitz/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:07:15 MQT: stat/blitz/STATUS10 = {"StatusSNS":{"Time":"2019-10-26T16:07:15","ENERGY":{"TotalStartTime":"2019-10-26T15:56:43","Total":0.071,"Yesterday":0.000,"Today":0.071,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
16:07:15 MQT: stat/blitz/STATUS11 = {"StatusSTS":{"Time":"2019-10-26T16:07:15","Uptime":"0T00:05:45","UptimeSec":345,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"****","BSSId":"********","Channel":8,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:05"}}}`

arendst, 2019-10-26:
@pierch upgrade to release 6.7.1 ASAP

pierch, 2019-10-26:
> @pierch upgrade to release 6.7.1 ASAP

OK, Done! Seems to work OK. Thanks for the great work!

Bigdude65, 2019-10-31:
Some advice needed.
I went from 6.6 to 6.7.1. I did about twenty devices at once using TasmoAdmin. All the Sonoff T1 & T2 switches Sonoff dual's worked fine but all my Sonoff Basics failed. They all came up in AP mode and after entering correct WiFi details continued to reboot as per wifimode 2. So I did a Reset Configuration from WiFi manager and that was the end of them. Cant get into WiFi or flash mode, no data seen on serial connection. They just sit there with endlessly slow flashing led.
Two of my Sonoff s26 plugs also went offline after power cycling them. same thing, slow flashing blue light.
There must be some logic still working as the button still switches the relay
Any help appreciated.

ascillato, 2019-10-31:
Please, address this to the Tasmota support chat for help. Thanks.