arendst/Tasmota #6746: Tasmota 6.7.0 - PZEM-004T (v2) - only received Voltage data

amagr0, 2019-10-25:
Hello,

I'm using the latest Tasmota firmware 6.7.0 (Build Date:20191025 Time:19:00 GMT+2) in a D1mini1, and I'm only receiving Voltage data. Energy Today and Energy Total is also updating.

[image]

Already tried with another PZEM-004T (v2), and with a NodeMCU board.

Thanks

Jason2866, 2019-10-25:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help

amagr0, 2019-10-25:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I'm using the latest Tasmota firmware 6.7.0 (Build Date:20191025 Time:19:00 GMT+2) in a D1mini1, and I'm only receiving Voltage data. Energy Today and Energy Total is also updating.
  

### REQUESTED INFORMATION

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): D1 Mini (ESP8266)
- [x] Tasmota binary firmware version number used: 6.7.0 (Build Date:20191025 Time:19:00 GMT+2) 
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: NodeMCU-PyFlasher-4.0-x64
- [x] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
20:03:11 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO"
20:03:12 SRC: Backlog
20:03:12 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:03:12 RSL: stat/sonoff/RESULT = {"NAME":"PZEM004TV3","GPIO":[0,62,0,63,6,5,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
20:01:37 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:01:37 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:23:03","StartupUTC":"2019-10-25T18:38:34","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":20,"SaveAddress":"3F8000"}}
20:01:37 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.7.0(6529820-sonoff)","BuildDateTime":"2019-10-25T19:01:53","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
20:01:37 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MEO-B7DCE0",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
20:01:37 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":3512,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
20:01:37 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3490","IPAddress":"192.168.1.82","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"80:7D:3A:3B:ED:A2","Webserver":2,"WifiConfig":4}}
20:01:37 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3BEDA2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:01:37 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 25 19:01:37 2019","Local":"Fri Oct 25 20:01:37 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:25","Sunset":"17:42"}}
20:01:37 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:01:37 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-25T20:01:37","ENERGY":{"TotalStartTime":"2019-10-25T19:24:30","Total":0.002,"Yesterday":0.000,"Today":0.002,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":236,"Current":0.000}}}
20:01:37 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-25T20:01:37","Uptime":"0T00:23:03","UptimeSec":1383,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"MEO-B7DCE0","BSSId":"00:06:91:B7:DC:E0","Channel":11,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:07"}}}
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
20:02:10 RSL: stat/sonoff/RESULT = {"WebLog":4}
20:02:11 DMP: A3 00 00 3F 00 00 E2
20:02:11 CFG: Saved to flash at 3F7, Count 21, Bytes 4096
20:02:11 DMP: A0 00 EC 02 00 00 8E
20:02:12 DMP: A1 00 00 00 00 00 A1
20:02:12 DMP: A2 00 06 00 00 00 A8
20:02:13 DMP: A3 00 00 3F 00 00 E2
20:02:13 DMP: A0 00 EC 06 00 00 92
20:02:13 RSL: tele/sonoff/STATE = {"Time":"2019-10-25T20:02:13","Uptime":"0T00:23:39","UptimeSec":1419,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"MEO-B7DCE0","BSSId":"00:06:91:B7:DC:E0","Channel":11,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:07"}}
[... cut ...]
20:02:17 DMP: A3 00 00 3F 00 00 E2
20:02:17 DMP: A0 00 EC 03 00 00 8F
20:02:18 DMP: A1 00 00 00 00 00 A1
20:02:18 DMP: A2 00 06 00 00 00 A8
20:02:19 DMP: A3 00 00 3F 00 00 E2
20:02:19 DMP: A0 00 EC 03 00 00 8F
20:02:20 DMP: A1 00 00 00 00 00 A1
20:02:20 DMP: A2 00 07 00 00 00 A9
20:02:21 DMP: A3 00 00 3F 00 00 E2
20:02:21 DMP: A0 00 EC 06 00 00 92
20:02:22 DMP: A1 00 00 00 00 00 A1
20:02:22 DMP: A2 00 07 00 00 00 A9
 
  
  ```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-10-25:
Try this Template `{"NAME":"PZEM004TV3","GPIO":[0,62,0,98,6,5,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}`

amagr0, 2019-10-25:
With that template I can no longer receive Voltage data, all blank.

[image]

Even after with the ModuleAddress 1 command.

Tks

Jason2866, 2019-10-25:
Ahh, sorry your config was correct you have V2 version!l

amagr0, 2019-10-25:
Sorry, but i didn't found anything in "release notes" that could help me to configure Tasmota.

amagr0, 2019-10-25:
The PZEM is already communicating with D1 Mini, since the Voltage values are continuous changing.

ModuleAddress 1 command was already applied.

Jason2866, 2019-10-25:
Mhh, irritated why didnt you tell you have already wrote here?
https://github.com/arendst/Sonoff-Tasmota/issues/6585#issuecomment-544256105
It is not nice to let us do double work

amagr0, 2019-10-25:
That was a different cenario, on that post I made a mistake with 3V and gnd, and because of that I could not get PZEM communicating with D1mini.

I solved that problem, but now can only receive Voltage data.

I will close the other post.
Thanks.

Jason2866, 2019-10-25:
It seems your module is defect

amagr0, 2019-10-25:
I tested with two PZEM and two different ESP modules (D1 Mini and NodeMCU), and the result was always the same.

ascillato2, 2019-10-25:
Is correctly wired your PZEM ? Can you share a Photo?
Is there any current passing ?

amagr0, 2019-10-25:
Yes, there's a light bulb of 10W.

[image]

amagr0, 2019-10-25:
I found that the switch in the light bulb cable is causing this issue.
[image]

I connected another lamp without the switch, and everything seems now to be ok.

[image]

[image]

closed 2019-10-25 by amagr0

ascillato, 2019-10-25:
Great to know that now it is working for you on v6.7.0

Thanks for reporting back :+1: