arendst/Tasmota #6747: pzem still buggy

Smandurlo, 2019-10-25:
After updating the new 6.7 (minimal first) I have this situation:

- pzem 004T reads the voltage (230V ok), but 0A and 0W

- pzem 016 0V, 0A and 0W. Plus, after the reboot I get some weird data showing:
this one is after the reboot: 
[image]
this one after 1-2 seconds:
[image]
and finally, after some seconds again:
[image]

Could you please reupload the 6.6 release with 2.5.2 core? They were working good, but the repository is empty now. At least I can use the pzem while waiting for a fix. Thank you!

ascillato2, 2019-10-25:
Closing this issue as it is **duplicated**. Sorry. Please, add your comments to the already open issue #6746 

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-25 by ascillato2

arendst, 2019-10-25:
Congratulations. You just experienced the autoconfigutation of your Pzem environment.

Pls read the releasenotes and enjoy.

Not an issue. This is as designed to service up to three pzems on a bus.