arendst/Tasmota #6748: When selecting module type Blitzwolf SHP, it reverts to Sonoff Basic

RazziaDK, 2019-10-25:
I upgraded my Blitzwolf BW-SHP6 to 6.7.0. First the minial image, then the regular sonoff.bin.

When the device restarted, it started with module type Blitzwolf SHP. Then the relay started turning on/off 5-10 times, and the module type was typed to Sonoff Basic.

If I change the configuration back the same thing happens.
I attached my weblog.
[blitzwolf-shp6-weblog.log](https://github.com/arendst/Sonoff-Tasmota/files/3773618/blitzwolf-shp6-weblog.log)

Jason2866, 2019-10-25:
When did you the upgrade? Flash minimal again. Do a OTA upgrade to sonoff.bin (6.7.0)
Do command `reset 5` Configure device by hand. Does it work? If yes your settings where corrupted

RazziaDK, 2019-10-25:
> When did you the upgrade? Flash minimal again. Do a OTA upgrade to sonoff.bin (6.7.0)
> Do command `reset 5` Configure device by hand. Does it work? If yes your settings where corrupted

I updated about an hour ago. I just reflashed again, and the same thing happend, but `reset 5` seems to have fixed it. Thanks!

closed 2019-10-25 by RazziaDK