arendst/Tasmota #6749: Compilation Error!

qingz2004, 2019-10-26:
### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_

Got error when I try to compile the latest development branch without any altering. So the core is 2.6.x.
Here is the message.

> lib\TasmotaSerial-2.4.1\src\TasmotaSerial.cpp: In constructor 'TasmotaSerial::TasmotaSerial(int, int, int, int, int)':
> lib\TasmotaSerial-2.4.1\src\TasmotaSerial.cpp:115:83: error: 'attachInterruptArg' was not declared in this scope
>        attachInterruptArg(m_rx_pin, callRxRead, this, (m_nwmode) ? CHANGE : FALLING);
>                                                                                    ^
> *** [.pioenvs\sonoff\lib009\TasmotaSerial-2.4.1\TasmotaSerial.cpp.o] Error 1
>  [FAILED] Took 21.31 seconds
>  

Using Platformio IDE with everything up to date.
Tried on Gitpod, works fine.

meingraham, 2019-10-26:
If it compiled on Gitpod, then the problem is your IDE installation. Can't help with that other than uninstall everything, delete any remnants in folders and start over.

qingz2004, 2019-10-26:
But I don't have problem compiling with 6.6.0.12.

qingz2004, 2019-10-26:
Problem fixed!
I can not just use the link "https://github.com/Jason2866/platform-espressif8266.git#Tasmota" directly. I had to download the files and integrate it into my Platformio library.

Jason2866, 2019-10-26:
Just use the provided platformio.ini! 
No need to copy anything by hand!!

qingz2004, 2019-10-26:
The core version from Gitpod and http://thehackbox.org/tasmota shows as the following.
Core/SDK Version	STAGE/2.2.2-dev(38a443e)
Is it pre-2.6?
If I set the platform to core_stage.platform, not core-pre.platform, I got no error, and core version is same as the one from OTA.

ascillato2, 2019-10-26:
Yes, Core/SDK Version STAGE/2.2.2-dev(38a443e) IS pre-2.6

closed 2019-10-26 by ascillato2

Jason2866, 2019-10-26:
files on thehackbox are pre2.6.
Pre2.6 is a freezed version from feature/stage at commit https://github.com/esp8266/Arduino/commit/2a5d21597724faa6d65c420b7263c0f8b6f000ce
So it is normal that `stage` is shown

qingz2004, 2019-10-26:
So I can just change the core_active in platform.ini to stage. And it's the same as core_pre.
```
platform                  = ${core_stage.platform}
build_flags               = ${core_stage.build_flags}

```
I got no error to compile with this.

Jason2866, 2019-10-26:
No, than you use always the latest stage version, which has newer commits (maybe introducing bugs 
or breaking changes)

qingz2004, 2019-10-26:
Why I got error if I just use core_pre, but no error using core_stage? What's the difference? I got the same core after compiling.
Is the problem from my platformio installation?

Jason2866, 2019-10-26:
Yes, your platformio installation is corrupted.
Try to delete as described here https://github.com/arendst/Sonoff-Tasmota/wiki/Flashing#flashing-and-compiling-from-source

Jason2866, 2019-10-26:
commits are NOT shown. You will always get the same info!