arendst/Tasmota #6750: Tasmota v.6.7.0 still does not work for my Sonoff POW R2

adharyanto, 2019-10-26:
### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [v] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [v] Device used (e.g., Sonoff Basic): Sonoff POW R2
- [v] Tasmota binary firmware version number used: v.6.7.0
  - [v] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [v] Flashing tools used: ESP8266 Download Tool V3.6.5
- [ ] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

**(Please, remember to close the issue when the problem has been addressed)**

Hi,

- I have pulled out my Sonoff POW R2 from my electricity power control box, (I connected the electricity directly, not via POW R2), and reflash manually (offline = not via OTA), to Tasmota v.6.7.0
- Switch on, connect to the wifi network, and changed the Configuration Module to Sonoff POW R2 (43), restart, then the results still = "FAIL", it keeps became Sonoff Basic (not POW R2)
- Tried to use RESET 5, then became disconnected to wifi network, after several minutes did not restart automatically, so it could not be accessed....
- Reflash again (offline), fallback to Tasmota v.6.6.0, switch on, connect wifi, changed to POW R2 = everything works fine.....!!!
- Then upgrade OTA to v.6.7.0, first via minimal-upgrade, then full upgrade, result = FAIL...... become Sonoff Basic (not POW R2) ==> upgrade is succesfull to v.6.7.0, but not succesfull change to POW R2
- Re-flashed again (offline) to v.6.6.0 = works 100% fine......
- Then upgrade OTA to v.6.7.0, directly to full-upgrade (not via minimal), result = FAIL..... could not configure POW R2

- For my case, Tasmota v.6.6.0 is still the one that works properly for my Sonoff POW R2
- I still cannot use Tasmota v.6.7.0 for my Sonoff POW R2, as it cannot be configured to POW R2 = always configure the module as Sonoff Basic, it means NO measurements at all
- I am using Tasmota v.6.7.0 for my other ESP8266 based modules (Sonoff Basic/Dual/4ch/ESP01/ESP12/WeMos), but for POW R2 I am using v.6.6.0

This issue is more for information only, incase someone else is experiencing the same case.
If this is not a usefull issue, then it is ok for me to close this issue.
Solution for me is to fallback using v.6.6.0 (for the POW R2)

mike2nl, 2019-10-26:
@adharyanto 
i don't read that you have the **flash erased** before downloding the new firmware to the sonoff pow.

adharyanto, 2019-10-26:
I did several times, also using the same tool (ESP8266 Download Tool V3.6.5). It has an Erase function as well. No effect whether i use Erase first before start Flashing, or directly Flashing without Erasing..... the result is the same. In the mean time I am using v.6.6.0. I do not have other POW R2 module to play or investigate, what is the cause of this issue.

Comaglove, 2019-10-26:
shelly 1pm behaves the same way.

bartbouwens, 2019-10-26:
Same problem over here with 3* POW R2 after updating to 6.7.0 (from 6.6.0)

Jason2866, 2019-10-26:
Do command `reset 5` before you upgrade. Do a OTA upgrade. Configure device (NO import of saved settings). Done

adharyanto, 2019-10-26:
In my case, after doing the RESET 5, then the unit disconnected from the WiFi network and won't restart, and so I totally do not have access to the module.... Unfortunately I have only 1 (one ) POW R2, and it is running live, I have to access this module remotely over wifi, if disconnected then I have to pull it out again, and it will interrupt the electricity supply again...

arendst, 2019-10-26:
@adharyanto Pls provide output of command status 0. Also provide console log when unit falls back to basic.

Needles to say chech your binary compile dat and time to make sure you use the corrected release 6.7 binary.

adharyanto, 2019-10-26:
Unfortunately I have only 1 (one) POW R2, and it is now installed and running back again inside the electricity control box, but I had to switch/fall back to v6.6.0 to run it properly. I will try to dismantle it again, and try to minimize the electricity interrupt, and then double check and re-test again to give you those required feed back. Updating from v.6.6.0 to v.6.7.0 was succesfull, as I can see on the bottom written v.6.7.0....... BUT, configuring to POW R2 fails..... it wont configure, it always set as Sonoff Basic (not POW R2)...

adharyanto, 2019-10-26:
This is the "status 0" result when it is running v.6.6.0
(at the bottom is written = Sonoff-Tasmota 6.6.0 by Theo Arends)

15:21:52 CMD: status 0
15:21:52 MQT: POW_R2/stat/STATUS = {"Status":{"Module":43,"FriendlyName":["POW R2"],"Topic":"POW_R2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:21:53 MQT: POW_R2/stat/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-minimal.bin","RestartReason":"Power on","Uptime":"0T00:03:05","StartupUTC":"2019-10-26T08:18:47","Sleep":50,"CfgHolder":4617,"BootCount":189,"SaveCount":2256,"SaveAddress":"F5000"}}
15:21:54 MQT: POW_R2/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":5,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:21:54 MQT: POW_R2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["omah",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["54000009","280500000100068000000000000000000000","00000000"]}}
15:21:55 MQT: POW_R2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":0,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
15:21:55 MQT: POW_R2/stat/STATUS5 = {"StatusNET":{"Hostname":"POW_R2-7895","IPAddress":"192.168.1.16","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"118.98.44.100","Mac":"80:7D:3A:6A:FE:D7","Webserver":2,"WifiConfig":3}}
15:21:56 MQT: POW_R2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6AFED7","MqttUser":"aspi","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:21:56 MQT: POW_R2/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 26 08:21:56 2019","Local":"Sat Oct 26 15:21:56 2019","StartDST":"Sun Jan 27 00:00:00 2019","EndDST":"Sun Jan 27 00:00:00 2019","Timezone":99,"Sunrise":"13:27","Sunset":"23:41"}}
15:21:57 MQT: POW_R2/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:21:57 MQT: POW_R2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-10-26T15:21:57","ENERGY":{"TotalStartTime":"2018-12-16T16:55:03","Total":0.255,"Yesterday":0.000,"Today":0.255,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":221,"Current":0.000}}}
15:21:58 MQT: POW_R2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-10-26T15:21:57","Uptime":"0T00:03:10","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"POWER1":"ON","Wifi":{"AP":1,"SSId":"omah","BSSId":"0C:37:47:93:63:DF","Channel":8,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:03"}}}

Then upgrading it via OTA:
http://thehackbox.org/tasmota/release/sonoff-minimal.bin
refresh the browser, then followed by
http://thehackbox.org/tasmota/release/sonoff.bin
refresh the browser

This is the "status 0" result when it is running v.6.7.0
(at the bottom is written = Sonoff-Tasmota 6.7.0 by Theo Arends)
This is just immediately after the upgrading, it is set automatically as Sonoff Basic:

15:30:16 CMD: status 0
15:30:16 MQT: POW_R2/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["POW R2"],"Topic":"POW_R2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:30:16 MQT: POW_R2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-minimal.bin","RestartReason":"Exception","Uptime":"0T00:03:12","StartupUTC":"2019-10-26T08:27:04","Sleep":50,"CfgHolder":4617,"BootCount":190,"SaveCount":2260,"SaveAddress":"F9000"}}
15:30:16 MQT: POW_R2/stat/STATUS2 = {"StatusFWR":{"Version":"6.7.0(sonoff)","BuildDateTime":"2019-10-25T12:28:03","Boot":5,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
15:30:16 MQT: POW_R2/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["omah",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["54000009","2805C8000100068000005A00000000000000","00000000"]}}
15:30:16 MQT: POW_R2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:30:16 MQT: POW_R2/stat/STATUS5 = {"StatusNET":{"Hostname":"POW_R2-7895","IPAddress":"192.168.1.16","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"118.98.44.100","Mac":"80:7D:3A:6A:FE:D7","Webserver":2,"WifiConfig":2}}
15:30:16 MQT: POW_R2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6AFED7","MqttUser":"aspi","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:30:16 MQT: POW_R2/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 26 08:30:16 2019","Local":"Sat Oct 26 15:30:16 2019","StartDST":"Sun Jan 27 00:00:00 2019","EndDST":"Sun Jan 27 00:00:00 2019","Timezone":99,"Sunrise":"13:27","Sunset":"23:41"}}
[... cut ...]
15:41:49 MQT: POW_R2/stat/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 26 08:41:49 2019","Local":"Sat Oct 26 15:41:49 2019","StartDST":"Sun Jan 27 00:00:00 2019","EndDST":"Sun Jan 27 00:00:00 2019","Timezone":99,"Sunrise":"13:27","Sunset":"23:41"}}
15:41:50 MQT: POW_R2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-10-26T15:41:49"}}
15:41:50 MQT: POW_R2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-10-26T15:41:50","Uptime":"0T00:00:33","UptimeSec":33,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":26,"MqttCount":1,"POWER1":"ON","Wifi":{"AP":1,"SSId":"omah","BSSId":"0C:37:47:93:63:DF","Channel":8,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:06"}}}

sgofferj, 2019-10-26:
Same with 3x Blitzwolf SHP-6 here, except, they go into a boot loop with an error like this:
switch4/tele/INFO3 {"RestartReason":"Fatal exception:0 flag:2 (EXCEPTION) epc1:0x4000dce5 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}

arendst, 2019-10-26:
@sgofferj provide command ``status 0`` output too

arendst, 2019-10-26:
> "Version":"6.7.0(sonoff)","BuildDateTime":"2019-10-25T12:28:03"

@adharyanto from your status 0 information I can confirm you are using the wrong released binary. 

Any release 6.7.0 binary with a BuildDateTime before 2019-10-25T18:00:00 is faulty with regards to energy monitoring devices. It will report Exception:0 with epc1:-x4000dce5 caused by a divide by zero coding error.

adharyanto, 2019-10-26:
Hi, but i upgrade this using OTA....
Is there any possibility that the source file on the OTA server is the wrong released?

arendst, 2019-10-26:
Pls provide the OTA link

adharyanto, 2019-10-26:
This is the minimal version & full version:
http://thehackbox.org/tasmota/release/sonoff-minimal.bin
http://thehackbox.org/tasmota/release/sonoff.bin

arendst, 2019-10-26:
Thx. When I now download the binary it is the correct version. When did you OTA upload your device?

BIERAAP, 2019-10-26:
> This is the minimal version & full version:
> http://thehackbox.org/tasmota/release/sonoff-minimal.bin
> http://thehackbox.org/tasmota/release/sonoff.bin

these are not working on the SPH6. Bootloop with the error above Exception:0 with epc1:-x4000dce5

adharyanto, 2019-10-26:
> Thx. When I now download the binary it is the correct version. When did you OTA upload your device?

Hi, I downloaded it just in between the time on my "status 0" reports above, (between 15:21 until 15:30 in my time zone).....just few hours ago... I live in location 7 hours earlier than GMT/UTC standard time.

sgofferj, 2019-10-26:
@arendst I have already flashed back to 6.6.0. But I downloaded the binary from Github an hour before my post, i.e. now 90min. ago. If that helps, the MD5 of the Sonoff.bin I was using is 6bfc0d621e11b40e0ed368d8a6458b1e.
I can flash one of the SHP-6s again tomorrow and do some debugging if you like.c

arendst, 2019-10-26:
Just verified it fails on SHP6 as soon as powerdelta > 0

Will investigate.

As a workaround set powerdelta to 0 and report back pls.

arendst, 2019-10-26:
I've got it fixed.

I will pull release 6.7 and release 6.7.1 in the next hours.

arendst, 2019-10-26:
Gents, pls try release 6.7.1 available now for OTA download.

PrathikGopal, 2019-10-26:
That fixed the POW R2 for me. Thanks @arendst.. 👍

adharyanto, 2019-10-26:
Hallo....
Good news.... Now this v.6.7.1 works fine for my Sonoff POW R2.
I upgraded it via OTA, using the same links as previously mentioned, (minimal version and then full version).
Thank you... @arendst

arendst, 2019-10-26:
Great!

Thank you all for your feedback. Much appreciated.

closed 2019-10-26 by arendst

PrathikGopal, 2019-10-26:
The sensor values on the MQTT output is 0 and not returning values. @adharyanto How about yours?

MQT: tele/powr2/SENSOR = {"Time":"2019-10-26T19:38:39","ENERGY":{"TotalStartTime":"2019-01-21T22:22:44","Total":0.040,"Yesterday":2.249,"Today":0.040,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

lobocobra, 2019-10-29:
I can confirm, that 6.7.1 works also for shelly 2.5. Was not the case with 6.7.0
=> Issue has gon with 6.7.1