arendst/Tasmota #6753: No tele/sonoff/SENSOR

thucar, 2019-10-26:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I'm not getting any SENSOR info coming in to my MQTT server. All sensor data is published only locally in the device console, marked with RSL prefix.
stat/sonoff/+ works fine, and so do tele/sonoff/INFO2, INFO3
  
**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff 4CH Pro
- [x] Tasmota binary firmware version number used: 6.7.0
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Arduino IDE
- [x] Flashing tools used: Arduino IDE
- [x] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
13:08:35 CMD: Backlog Template; Module; GPIO
13:08:35 SRC: WebConsole from 192.168.32.204
13:08:35 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO"
13:08:36 SRC: Backlog
13:08:36 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
13:08:36 MQT: stat/heating/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
13:08:36 SRC: Backlog
13:08:36 CMD: Group 0, Index 1, Command "MODULE", Data ""
13:08:36 MQT: stat/heating/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
13:08:36 SRC: Backlog
13:08:36 CMD: Group 0, Index 1, Command "GPIO", Data ""
  
  
  ```
- [x] Provide the output of this command: ``Status 0``:
  ```
  13:09:42 CMD: Group 0, Index 1, Command "STATUS", Data "0"
13:09:42 RSL: stat/heating/STATUS = {"Status":{"Module":23,"FriendlyName":["3T Open","3T Close","Radiator Pump","Boiler Pump"],"Topic":"heating","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"greenhouse","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:09:42 RSL: stat/heating/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:25","StartupUTC":"2019-10-26T10:01:17","Sleep":50,"CfgHolder":4617,"BootCount":35,"SaveCount":73504,"SaveAddress":"F6000"}}
13:09:42 RSL: stat/heating/STATUS2 = {"StatusFWR":{"Version":"6.7.0(sonoff)","BuildDateTime":"2019-10-26T10:33:05","Boot":31,"Core":"2_5_0","SDK":"3.0.0-dev(c0f7b44)"}}
13:09:42 RSL: stat/heating/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Nutiseadmed",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005AFF000000000000","40000000"]}}
13:09:42 RSL: stat/heating/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8F082382","00100000","00000044","000000C0","00000100"],"Drivers":"1,2,3,4,9,10,20,21,27","Sensors":"2,3,5"}}
13:09:42 RSL: stat/heating/STATUS5 = {"StatusNET":{"Hostname":"heating-0784","IPAddress":"10.10.10.10","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"60:01:94:A3:63:10","Webserver":2,"WifiConfig":4}}
13:09:42 RSL: stat/heating/STATUS6 = {"StatusMQT":{"MqttHost":"home.raitwebs.com","MqttPort":9993,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A36310","MqttUser":"Homey","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:09:42 RSL: stat/heating/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 26 10:09:42 2019","Local":"Sat Oct 26 13:09:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+03:00","Sunrise":"09:27","Sunset":"19:41"}}
13:09:42 RSL: stat/heating/STATUS10 = {"StatusSNS":{"Time":"2019-10-26T13:09:42","DS18B20-1":{"Id":"000005957B1D","Temperature":28.3},"DS18B20-2":{"Id":"0000059625CB","Temperature":24.8},"DS18B20-3":{"Id":"00000596501B","Temperature":31.4},"DS18B20-4":{"Id":"0000059691D5","Temperature":48.4},"DS18B20-5":{"Id":"0014517554FF","Temperature":48.7},"DS18B20-6":{"Id":"0014517FE8FF","Temperature":40.4},"Shutter1":{"Position":28,"direction":0},"TempUnit":"C"}}
13:09:42 RSL: stat/heating/STATUS11 = {"StatusSTS":{"Time":"2019-10-26T13:09:42","Uptime":"0T00:08:25","UptimeSec":505,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Nutiseadmed","BSSId":"6E:3B:6B:45:2D:EF","Channel":1,"RSSI":56,"LinkCount":1,"Downtime":"0T00:00:05"}}}
  
  ```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
 13:01:25 MQT: Attempting connection...
13:01:25 MQT: Connected
13:01:25 MQT: tele/heating/LWT = Online (retained)
13:01:25 MQT: cmnd/heating/POWER = 
13:01:25 MQT: Subscribe to cmnd/heating/#
13:01:25 RSL: tele/heating/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"6.7.0(sonoff)","FallbackTopic":"cmnd/DVES_A36310_fb/","GroupTopic":"sonoffs"}
13:01:25 MQT: tele/heating/INFO2 = {"WebServerMode":"Admin","Hostname":"heating-0784","IPAddress":"10.10.10.10"}
13:01:25 MQT: tele/heating/INFO3 = {"RestartReason":"Software/System restart"}
13:01:25 MQT: stat/heating/RESULT = {"POWER1":"OFF"}
13:01:25 MQT: stat/heating/POWER1 = OFF
13:01:25 MQT: stat/heating/RESULT = {"POWER2":"OFF"}
13:01:25 MQT: stat/heating/POWER2 = OFF
13:01:25 MQT: stat/heating/RESULT = {"POWER3":"ON"}
13:01:25 MQT: stat/heating/POWER3 = ON
13:01:25 MQT: stat/heating/RESULT = {"POWER4":"OFF"}
13:01:25 MQT: stat/heating/POWER4 = OFF
13:01:25 QPC: Reset
13:01:25 HTP: Main Menu
13:01:27 APP: Boot Count 35
13:01:28 CFG: Saved to flash at F6, Count 73504, Bytes 4096
13:01:33 RSL: tele/heating/STATE = {"Time":"2019-10-26T13:01:33","Uptime":"0T00:00:16","UptimeSec":16,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":38,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Nutiseadmed","BSSId":"6E:3B:6B:45:2D:EF","Channel":1,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:05"}}
13:01:33 RSL: tele/heating/SENSOR = {"Time":"2019-10-26T13:01:33","DS18B20-1":{"Id":"000005957B1D","Temperature":28.3},"DS18B20-2":{"Id":"0000059625CB","Temperature":24.7},"DS18B20-3":{"Id":"00000596501B","Temperature":31.1},"DS18B20-4":{"Id":"0000059691D5","Temperature":47.8},"DS18B20-5":{"Id":"0014517554FF","Temperature":48.4},"DS18B20-6":{"Id":"0014517FE8FF","Temperature":40.3},"Shutter1":{"Position":28,"direction":0},"TempUnit":"C"}
  
  
  ```

Jason2866, 2019-10-26:
RSL means you have no connection to mqtt broker

ascillato2, 2019-10-26:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-26 by ascillato2

thucar, 2019-10-26:
@Jason2866 if you look at the logs I posted, you can see there are RSL entries with MQT immediately before and after it.

So does that mean there is no MQTT connection for that millisecond only?

@ascillato2 I do believe there is a good chance this is a bug

arendst, 2019-10-26:
Pls update to core pre-2.6.0 as your current core 2.5.0 is know to be buggy.

thucar, 2019-10-27:
I've upgraded to Core pre-2.6.0 but the issue remains the same. 
I did a reset 6, upgraded the firmware, reconfigured my sonoff (Sonoff 4CH Pro with DS18b20's on GPIO2)
Nothing changed.

```
04:01:47 MQT: Attempting connection...
04:01:47 MQT: Connected
04:01:47 MQT: tele/heating/LWT = Online (retained)
04:01:47 MQT: cmnd/heating/POWER = 
04:01:47 RSL: tele/heating/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"6.7.0(sonoff)","FallbackTopic":"cmnd/DVES_A36310_fb/","GroupTopic":"sonoffs"}
04:01:47 MQT: tele/heating/INFO2 = {"WebServerMode":"Admin","Hostname":"heating-0784","IPAddress":"10.10.10.10"}
04:01:47 MQT: tele/heating/INFO3 = {"RestartReason":"Software/System restart"}
04:01:47 MQT: stat/heating/RESULT = {"POWER1":"OFF"}
04:01:47 MQT: stat/heating/POWER1 = OFF
04:01:47 MQT: stat/heating/RESULT = {"POWER2":"OFF"}
04:01:47 MQT: stat/heating/POWER2 = OFF
04:01:47 MQT: stat/heating/RESULT = {"POWER3":"OFF"}
04:01:47 MQT: stat/heating/POWER3 = OFF
04:01:47 MQT: stat/heating/RESULT = {"POWER4":"OFF"}
04:01:47 MQT: stat/heating/POWER4 = OFF
04:01:52 RSL: stat/heating/STATUS8 = {"StatusSNS":{"Time":"2019-10-27T04:01:52","DS18B20-1":{"Id":"000005957B1D","Temperature":28.3},"DS18B20-2":{"Id":"0000059625CB","Temperature":28.0},"DS18B20-3":{"Id":"00000596501B","Temperature":43.3},"DS18B20-4":{"Id":"0000059691D5","Temperature":46.9},"DS18B20-5":{"Id":"0014517554FF","Temperature":48.0},"DS18B20-6":{"Id":"0014517FE8FF","Temperature":38.6},"TempUnit":"C"}}
04:01:52 MQT: stat/heating/RESULT = {"POWER3":"ON"}
04:01:52 MQT: stat/heating/POWER3 = ON
```

This is my `Status0:`
```
04:05:52 RSL: stat/heating/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"heating","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:05:52 RSL: stat/heating/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:13","StartupUTC":"2019-10-27T03:01:39","Sleep":50,"CfgHolder":4617,"BootCount":4,"SaveCount":12,"SaveAddress":"F8000"}}
04:05:52 RSL: stat/heating/STATUS2 = {"StatusFWR":{"Version":"6.7.0(sonoff)","BuildDateTime":"2019-10-27T04:49:41","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
04:05:52 RSL: stat/heating/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Nutiseadmed",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
04:05:52 RSL: stat/heating/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8F082382","00100000","00000044","000000C0","00000100"],"Drivers":"1,2,3,4,9,10,20,21,27","Sensors":"2,3,5"}}
04:05:52 RSL: stat/heating/STATUS5 = {"StatusNET":{"Hostname":"heating-0784","IPAddress":"10.10.10.10","Gateway":"10.10.10.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"60:01:94:A3:63:10","Webserver":2,"WifiConfig":4}}
04:05:52 RSL: stat/heating/STATUS6 = {"StatusMQT":{"MqttHost":"home.raitwebs.com","MqttPort":9993,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A36310","MqttUser":"Homey","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
04:05:52 RSL: stat/heating/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 27 03:05:52 2019","Local":"Sun Oct 27 04:05:52 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:28","Sunset":"17:39"}}
04:05:52 RSL: stat/heating/STATUS10 = {"StatusSNS":{"Time":"2019-10-27T04:05:52","DS18B20-1":{"Id":"000005957B1D","Temperature":28.4},"DS18B20-2":{"Id":"0000059625CB","Temperature":28.2},"DS18B20-3":{"Id":"00000596501B","Temperature":36.8},"DS18B20-4":{"Id":"0000059691D5","Temperature":47.0},"DS18B20-5":{"Id":"0014517554FF","Temperature":48.0},"DS18B20-6":{"Id":"0014517FE8FF","Temperature":38.7},"TempUnit":"C"}}
04:05:52 RSL: stat/heating/STATUS11 = {"StatusSTS":{"Time":"2019-10-27T04:05:52","Uptime":"0T00:04:13","UptimeSec":253,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Nutiseadmed","BSSId":"6E:3B:6B:45:2D:EF","Channel":1,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:05"}}}
04:05:52 RSL: stat/heating/STATUS8 = {"StatusSNS":{"Time":"2019-10-27T04:05:52","DS18B20-1":{"Id":"000005957B1D","Temperature":28.4},"DS18B20-2":{"Id":"0000059625CB","Temperature":28.2},"DS18B20-3":{"Id":"00000596501B","Temperature":36.8},"DS18B20-4":{"Id":"0000059691D5","Temperature":47.0},"DS18B20-5":{"Id":"0014517554FF","Temperature":48.0},"DS18B20-6":{"Id":"0014517FE8FF","Temperature":38.7},"TempUnit":"C"}}
```

thucar, 2019-10-27:
Just finished a more extensive testing. Everything works when using the source files of the latest release. This issue is only present if building from files checked out from github directly.

meingraham, 2019-10-27:
Allison (6.7.1) is the latest release. Pre-compiled binaries are available at http://thehackbox.org/tasmota

arendst, 2019-10-27:
Thx for testing. I was afraid indeed that your MAX_PACKET_SIZE was too.small but from your status info it shows 1000.

I think something goes wrong with the gitpod pubsubclient library as in there it also needs to be set to 1000 as it is set in the github tasmota library too.

Strange...

Jason2866, 2019-10-27:
@thucar  Gitpod ALWAYS uses latest dev/master version since it does fetch Tasmota repo files via git when started. There are NO difference files used