arendst/Tasmota #6755: Sonoff RE5v1C

jonferreira, 2019-10-26:
Hi,

I've recently purchased a couple of the new Sonoff RE5V1C tinny modules and I'm wondering if there's any plans for tasmota to support these?

ascillato2, 2019-10-26:
If the device has an esp8266 and someone wants to donate its time making a new template or driver if needed, it can be supported.

If you want to work on it and need some help for that, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-26 by ascillato2

jonferreira, 2019-10-26:
thanks will do!

danmandle, 2019-12-16:
For those who end up here via searching like I did, there is now a template: https://blakadder.github.io/templates/sonoff_RE5V1C.html

dweston, 2019-12-29:
Wiring arrangement for flashing Sonoff RE5V1C:
[image]

pauljatho, 2020-01-30:
Thanks Danmandle for this, I managed to flash one of these with Tasmota and using your template its working.

I wonder do you know how to change it to work from NO to NC? I would like to use this to control a magnetic door strip.

Thanks agin.

danmandle, 2020-01-30:
@pauljatho You would likely have to do that with software. The downside is that if it is without power, it would be NO.

pauljatho, 2020-01-30:
Yup, while thinking it through I realised that the relay doesn't have a NC pin so it would like you say without power be NO which wont work for me since this is controlling an access door.

I'll get one of the Sonoff 1Channel inching devices as this has both NO and NC.

Thanks again for your post above.

ttweb, 2020-02-10:
Hello everybody,
I also have two of these devices, but I never had such problems soldering the little contacts here compared to other devices. I use standard pb solder.
How did you manage to solder the cable/pins correctly? Any suggestions which flux, temperature and solder to use? Thank you very much.

pauljatho, 2020-02-10:
> Hello everybody,
> I also have two of these devices, but I never had such problems soldering the little contacts here compared to other devices. I use standard pb solder.
> How did you manage to solder [...]

I soldered directly onto the pads but don't use pins use cable, first put a small amount of solder on each of the pads and make sure the cable has solder, needs a very short time to solder.

Also be careful the pads come off very easily I pulled the one ground pad off, luckily it was the ground pad and not TX or RX so I was able to use another ground on the board.

ttweb, 2020-02-10:
Thank you very much, I will try this again later. Maybe I just was too impatient...

clockworkcrow, 2020-08-14:
> I soldered directly onto the pads but don't use pins use cable, first put a small amount of solder on each of the pads and make sure the cable has solder, needs a very short time to solder.
>
> Also be [...]

I lost the pad on the ground too. Which alternative did yo use?

lobmarib, 2021-02-13:
> Thanks Danmandle for this, I managed to flash one of these with Tasmota and using your template its working.
> 
> I wonder do you know how to change it to work from NO to NC? I would like to use this to control a magnetic door strip.
> 
> Thanks again.

Which tasmota build did you use? Regular tasmota.bin? 
Thanks

gaucho1978, 2021-04-10:
With this firmware it is possible to activate sleep mode, is it correct? Which is the consumption of SONOFF Re5v1c during sleep?

danielbrunt57, 2021-09-08:
I just successfully soldered connections to my 2nd unit (botched the first - ERX and ETX pads are gone), flashed with regular tasmota.bin, added template {"NAME":"Sonoff RE5V1C","GPIO":[17,255,255,255,255,255,0,0,21,56,0,0,0],"FLAG":0,"BASE":18}, issued `SwitchMode 13` and `PulseTime 10` in console and inching mode is working!  Woo hoo!  My third Tasmota flash after two Sonoff S31's...