arendst/Tasmota #6759: Support for Stepper Motor 28BYJ-48 + the Driver ULN2003

adharyanto, 2019-10-27:
**Have you looked for this feature in other issues and in the wiki?**  
==> Yes

**Is your feature request related to a problem? Please describe.**  
==> No, not a problem, but additional feature

**Describe the solution you'd like**  
==> I know Tasmota recently has supported the A4988/DRV8255 Stepper Motor Controller that can be connected to the stepper motor 28BYJ-48 or NEMA-17

**Describe alternatives you've considered**  
==> There is alternative, a VERY-CHEAP and VERY-POPULAR Stepper Motor controller/driver, the ULN2003, that also can be connected to 28BYJ-48, in different ways than A4988/DRV8255. The ULN2003 is just more like a driver.

**Additional context**  
==> It would be great if Tasmota accommodates the capability to control 28BYJ-48 Stepper Motors using this ULN2003, (without/not using A4988/DRV8255 controller).
Inside the Tasmota we can give input to command the motor:
- possibility to switch on/off
- how many steps to turn right/left
- how may degree rotation per step
- how fast/slow the motor rotation
- manually control up/down (turn right/left) by a button/sliding switch
- etc...

Jason2866, 2019-10-27:
Sorry no, since the ULN2003 is just a "current" driver. Tasmota has not the resources to do this.

andrethomas, 2019-10-27:
I think the benefits of using A4988/DRV8825 modules far outweigh the cost savings of using ULN2003's to drive stepper motors.

The ULN2003 breaks down at around 500mA and even at that high current levels, it gets pretty darn hot. Perhaps they are fine for what they were originally intended like opening and closing the vent on your split air conditioning system.

That aside it's more complex to implement in the firmware so even there the benefits of the A4988/DRV8825 outweigh the cost of flash and ram on the ESP8266.

I would rather see the flash and ram used to add support for newer generation drivers like the TMC2100 or perhaps the TMC2208 which have the unique added advantage that they drive a stepper motor more quietly and less violently.

ascillato2, 2019-10-27:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-27 by ascillato2

adharyanto, 2019-10-28:
Yes, ULN2003 is more as a current "buffer", so there must be more complicated on the firmware to control the stepper motor. It is OK to close this issue. Then I would rather to use the A4988 instead.