arendst/Tasmota #6760: Possible bug in mqtt for WifiConfig

hawkeyexp, 2019-10-27:
Hi,

starting with Tsmota 6.7.0 setting the option `WifiConfig 7` results in `WifiConfig_7` in mqtt with value `ManagerRst`. I think this is a bug cause before it was reportet as `WifiConfig` with value `ManagerRst` which was correct.

arendst, 2019-10-27:
I cannot reproduce:
```
11:14:27 CMD: wificonfig 7
11:14:27 SRC: Serial
11:14:27 CMD: Group 0, Index 1, Command "WIFICONFIG", Data "7"
11:14:27 MQT: stat/wemos4/WIFICONFIG = {"WifiConfig":{"7":"ManagerRst"}}
11:14:27 CFG: Saved to flash at F9, Count 156, Bytes 4096
11:14:48 CMD: wificonfig 4
11:14:48 SRC: Serial
11:14:48 CMD: Group 0, Index 1, Command "WIFICONFIG", Data "4"
11:14:48 MQT: stat/wemos4/WIFICONFIG = {"WifiConfig":{"4":"Retry"}}
11:14:49 CFG: Saved to flash at F8, Count 157, Bytes 4096
```

ascillato2, 2019-10-27:
Closing as this issue can't be reproduced. If you find more information about this, please add it here. Thanks for reporting. 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-27 by ascillato2