arendst/Tasmota #6762: Hide Alexa discovered objects

prstprst, 2019-10-27:
Hello, 

I'm using an electrodragon in shutter mode and I've added a third output that is a "pwm" directly connected to the buit-in led of the electrodragon. Then, by using rules, I've linked the dimmer of the third output to the command of the shutters. And like this, with alexa,  I can use the dimmer of the dicovered third output and control my shutters with granularity. May be I'm false. But It's the way I found.
But in this case, the problem is that Alexa discover the two others output of the electrodragon. So for each shutter in my house, I have two more dicovered and unusable devices in my alexa interface. 
I repeat, may be I'm false, but I don't understand how I can discover only one device with alexa when I use the tasmota shutter mode with a dual relay real device (like electrodragon or sonoff dual)
Soory if I'm not at the good place. I'm newbee. Thx :-)
Thank to have taken time to read me again 
prst

s-hadinger, 2019-10-27:
Currently there is no way to hide some devices from Alexa. You need to delete the unwanted devices via the Alexa app.

s-hadinger, 2019-10-27:
Duplicate of #6722

closed 2019-10-27 by ascillato2