arendst/Tasmota #6764: RGBW values no longer load correctly on restart

arktronic, 2019-10-27:
### BUG DESCRIPTION
As of v6.7.1, mixed RGBW values do not load correctly after a restart: for example, a hex value of `ff150033` will work correctly after being set, but after a restart, will be loaded as `ff150000`. This was tested with both SM16716- and MY92x1-based bulbs.

Notably, after downgrading to v6.6.0 firmware, the correct mixed RGBW value loaded (after having been saved with v6.7.1) so it appears that the saving is happening correctly, but the loading is problematic.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): multiple light bulbs
- [x] Tasmota binary firmware version number used: `6.7.1(sonoff)`
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: webserver flash
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
```
00:01:29 RSL: RESULT = {"NAME":"MI-BW904-999W","GPIO":[0,0,0,0,140,37,0,0,0,142,141,0,0],"FLAG":0,"BASE":18}
00:01:29 RSL: RESULT = {"Module":{"0":"MI-BW904-999W"}}
00:01:29 RSL: RESULT = {"GPIO":"Not supported"}
```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE

1. Use RGBW or RGBCW bulb
1. Set color to anything with both RGB and W/CW components, e.g., `color ffdd0015`
1. `restart 1`
1. Observe `color` return `ffdd0000` instead

### EXPECTED BEHAVIOUR
Color should still be set to previous value (`ffdd0015`)

s-hadinger, 2019-10-27:
This is expected behavior to have either RGB or White but not both at the same time. Use 'SetOption37 128' if you want to be able to set both independently. 

Can you please set and confirm?

s-hadinger, 2019-10-27:
Also look at this option if that's what you are looking for.
https://github.com/arendst/Tasmota/wiki/White-Blend-Mode

arktronic, 2019-10-27:
Using `setoption37 128` does not fix the issue, unfortunately.

arktronic, 2019-10-27:
To clarify, setting the color to mixed RGB+W works by itself. It's only the reloading of that setting after a restart that does not work correctly.

s-hadinger, 2019-10-27:
Ok. I'm off for the week and will look at it in a few days.

s-hadinger, 2019-10-27:
Actually the behavior changed with #6612 to avoid overloading of newly flashed devices. In any cases it will require ‘SetOption37 128’ to restore both RGB and White.

arktronic, 2019-10-27:
Looks like the settings get restored correctly with `setoption68 1`. Can you confirm whether this is the intended behavior, please?

s-hadinger, 2019-10-27:
It depends. SetOption68 allows you to have 4 independent channels. Is this what you want?

arktronic, 2019-10-28:
I'm not sure it matters for my use case: the `color` command still works as expected, and each channel gets saved and loaded correctly. Am I missing a subtlety?

s-hadinger, 2019-10-28:
If you just use the color command then yes, this is all you need.

ascillato2, 2019-10-28:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-28 by ascillato2

denverpilot, 2019-11-05:
Example use case where the behavior change behaves poorly, as does the workaround... unless I'm missing something: 

1. Set RGBW as desired on this type of bulb that's using real LED drivers:
{"NAME":"Lohas RGBW","GPIO":[0,0,0,0,0,0,0,0,0,143,0,144,0],"FLAG":0,"BASE":18}
(Ex: 255,255,0,150)
2. Next set PowerOnState 1
3. Flip power off and on at a hardwired power switch: 

Bulb reverts to RGB without W setting. 
W is set to 0.
(Ex: 255,255,0,0)

Additional oddities:
4. SetOption 68 1 basically splits the bulb into four, and "typical" RGBW lovelace cards and other things in HomeAssistant (and I assume others) may deal with it poorly thinking it's multiple bulbs. HA autodiscovery breaks in an odd way. The bulb is detected after the option change as "Sonoff 4" and power toggle don't work.

5. SetOption 37 128 appears to split the bulb into two, which also breaks typical UI controls in HA like 68 does. HA autodiscovery breaks similarly but now the entity discovered is called "Tasmota2".

(The oddities are more of an HA/UI/MQTT problem, but does require repair of UI elements, vs the bulb restarting at saved RGBW values.)

Doesn't feel like the expected behavior for items 1 through 3 ...? 
The HA stuff is, what it is, I suppose...

Tests for the above were on Tasmota 7.0.0.3 main binary, and HA 0.101.2

denverpilot, 2019-11-05:
Perhaps items 4 & 5, I was using reset 6, and FriendlyName and Hostname were getting reset. 
This messed with my testing / will have to re-do that part.

s-hadinger, 2019-11-05:
@denverpilot 1/2/3 is the expected behavior, especially since #6612 and #6534.

The bulb will be either RGB, or White **EXCEPT** the following:

- you force levels of each channel with the `Color` command
- `SetOption37` > 128, splitting bulb into RGB + White
- `SetOption68 1`, splitting bulb in individual channels
- you active White Blend Mode with `RGBWWtable`, see here: https://github.com/arendst/Tasmota/wiki/white-blend-mode

denverpilot, 2019-11-05:
Yeah, it's too bad the software has to change a reasonable behavior at power on, for badly designed power supplies. Especially for non-PWM bulbs with real LED drivers. The engineer in me understands why, but for the user in me, it's not intuitive and completely breaks the wall switch use case where a family member might turn off a wall switch --which seems the real purpose of PowerOnState 1 -- for a bulb.

The bulb comes back on a different color than it was set to, when using PowerOnState 1 to keep the family happy with custom flashed bulbs.  I can't really tell my wife to go to a web page and type a color command when someone bumps a wall switch. LOL.. or guests. :-)

About my only option is to write code into my stuff that if it sees any RGBW bulb reboot, check if it's in a scene or normal white-ish light mode, and force the white channel back into it.

Plus in the case of a real power outage (we have many, being rural), the automation server won't be back up yet before the bulbs, and they''ll all come up in some odd RGB only color until the server finishes coming up and tells them they have a white channel value again.

I don't believe any of those options mentioned, help with the breakage of PowerOnState 1.

I'll need to test both options and see if the bulb comes up properly as RGBW mixed if PowerOnState 1 is on, or see if retained MQTT messages work through a power cycle to see if there's a "family can use it" solution to the wall switch and power outage problems.

Thanks for considering it. I started testing that entire matrix of options last night but it was late and I was making mistakes. It didn't appear the W channel was being activated by any of those options after a cold start.  Always 0.  But will attempt a more disciplined test pattern as soon as I have time.

As an aside, I don't believe MQTT Friendly Name nor WiFi Hostname are being retained during a "reset 6", but that could be interaction with recovery modes. That was what derailed my testing... I can play with that more before saying that's always the case, but probably needs a documentation change for reset 5 & 6.  If you agree I could put that on my long to-do list.