arendst/Tasmota #6765: Tasmota using PCF8574

Sequenzial, 2019-10-27:
I2C device "PCF8574" is working generally, all Ports are configured as output:
[image]
- power1-7 work fine
- power8 has no function (on the console I see the change, but the relais is not responding)
- power0 acting on all relais (1-7 - not 8)
so Relais 8 cannot be triggered.

[image]

Tried on 6.7.1 and currently running dev. 7.0.0.1

- [**x**] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [**x**] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [**x**] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [**x**] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [**x**] Device used (e.g., Sonoff Basic): wemos D1 mini
- [**x**] Tasmota binary firmware version number used: **7.0.0.1**
  - [**x**] Self-compiled
    - [**x**] IDE / Compiler used: **Arduino 1.8.10 / esp8266 V2.3.0**
- [**x** ] Flashing tools used: **Arduino 1.8.10**
- [**x**] Provide the output of command: ``Backlog Template; Module; GPIO``:

`16:30:19 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:30:19 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/RESULT = {"Module":{"18":"Generic"}}
16:30:19 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"5":"I2C SCL"},"GPIO13":{"0":"None"},"GPIO14":{"6":"I2C SDA"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}`
  Configuration output here:

- [**x**] Provide the output of this command: ``Status 0``:
  STATUS 0 output here:

`6 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS = {"Status":{"Module":18,"FriendlyName":["ESP-RELAIS-8X-01","ESP-RELAIS-8X-012","ESP-RELAIS-8X-013","ESP-RELAIS-8X-014"],"Topic":"ESP-RELAIS-8X-01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://192.168.1.1/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:15:58","StartupUTC":"2019-10-27T15:16:38","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":50,"SaveAddress":"3FA000"}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS2 = {"StatusFWR":{"Version":"7.0.0.1(tasmota)","BuildDateTime":"2019.10.27 16:11:47","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.1.1","LogPort":514,"SSId":["WLAN","WLAN2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","80000000"]}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS4 = {"StatusMEM":{"ProgramSize":446,"Free":2624,"Heap":9,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000407","0F002186","000083A0","00000085","000000C0","00000200"],"Drivers":"1,2,3,4,9,10,16,28","Sensors":"1,2,3,6"}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS5 = {"StatusNET":{"Hostname":"ESP-RELAIS-8X-01","IPAddress":"192.168.3.111","Gateway":"192.168.1.250","Subnetmask":"255.255.252.0","DNSServer":"192.168.1.3","Mac":"EC:00:00:00:D8:31","Webserver":2,"WifiConfig":4}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.197","MqttPort":1883,"MqttClientMask":"ESP-RELAIS-8X-01","MqttClient":"ESP-RELAIS-8X-01","MqttUser":"USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 27 15:32:36 2019","Local":"Sun Oct 27 16:32:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:10","Sunset":"17:09"}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS10 = {"StatusSNS":{"Time":"2019-10-27T16:32:36"}}
16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS11 = {"StatusSTS":{"Time":"2019-10-27T16:32:36","Uptime":"0T00:15:58","UptimeSec":958,"Heap":9,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","Wifi":{"AP":1,"SSId":"WLAN","BSSId":"E0:00:00:00:83:DE","Channel":12,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:09"}}}`

would be great to have all 8 Relais working!

Thanks in common!

BR
Seq

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-10-27:
This core is NOT supported
`16:32:36 MQT: SmartHome/Garten/stat/ESP-RELAIS-8X-01/STATUS2 = {"StatusFWR":{"Version":"7.0.0.1(tasmota)","BuildDateTime":"2019.10.27 16:11:47","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}} `
Please compile with Core Pre 2.6. See wiki how to use in Arduino IDE https://github.com/arendst/Tasmota/wiki/Arduino-IDE

andrethomas, 2019-10-27:
Have you tried power0 ?

Sequenzial, 2019-10-27:
Yes, power0 switches all (1-7) relais, except 8.
... and yes the relais is ok ;-)
 
I´ll try compiling with vsc -> pre 2.6.

Sequenzial, 2019-10-27:
Ok, I have compiled the firmware using Visual Studio Code with Core PRE 2.6
Now all 8 relais work fine. 

Thanks for the support!

BR
Seq

closed 2019-10-27 by Sequenzial

yogesh102011, 2020-01-23:
if i make all 8 as output..its working fine with relay. but one problem i am facing is if all the 8 relays are on.. while restarting none of the relay is in onstate. but in web interface it show all are on. if i off one relay all are getting on except the one which is off

ITTV-tools, 2020-03-01:
How did you set the PCF8574 up ? I can see it with i2cscan. But I found nothing in the docs and can not control it.

Sequenzial, 2020-03-01:
When you got a firmware with PCF8574 integrated (no in these you can download), you have to set SDC and SDA to the correct GPIOS (those they are connected to).
[image]

When everything is correct, you will get an extra option under settings:
[image]
where you can define if the ports are IN- or OUT-Put (for each):
[image]

That´s it. The Output ports are shown as regular ports:
[image]

Inputs I did not use yet.

Sequenzial, 2020-03-01:
(sorry I only have the german interface available in the pictures ...)

ITTV-tools, 2020-03-01:
german is fine :) The changelog says there is support for this chip. I would compile it myself but I don’t know what to add it in „user_config_override“ for example. How have you done it ? Or wich firmware have you used ?

Sequenzial, 2020-03-01:
Just remove the //
  #define USE_PCF8574                            // [I2cDriver2] Enable PCF8574 I/O Expander (I2C addresses 0x20 - 0x26 and 0x39 - 0x3F) (+1k9 code)
[image]
Then you can use it.

Sequenzial, 2020-03-01:
https://github.com/arendst/Tasmota/wiki/Prerequisites
https://github.com/arendst/Tasmota/wiki/Hardware-preparation

I personally still use Arduino IDE, because I get usable results.

Sequenzial, 2020-03-01:
>> The changelog says there is support for this chip.
I assume the i2c protocoll can see a device, but if the firmware does not contain the correct driver, it won´t work.
You have to compile a firmware yourself or get an modified version somewhere.

Sequenzial, 2020-03-01:
--- > The changelog says there is support for this chip.
I assume the i2c protocoll can see a device, but if the firmware does not contain the correct driver, it won´t work.
You have to compile a firmware yourself or get an modified version somewhere.

Sequenzial, 2020-03-01:
I have compiled the "basic 8.1" and enabled MCP230xx and PCF8574

[tasmota_8.1_MCP230xx+PCF8574.zip](https://github.com/arendst/Tasmota/files/4272099/tasmota_8.1_MCP230xx%2BPCF8574.zip)

With 600 kb it is pretty big, so you need to flash an minimal firmware first.

KostasHelsinki, 2020-06-21:
Hi guys, im experimenting with this but i found unreliable the operation. First, when esp8266 boots it briefly pulls high all pcf8574 pins high. Then, if i power on off a pin from webui or console, sometimes doesnt really do anything. My 3rd problem is that i cannot use the pins as inputs. Has any of you had these problems?

alxferraz, 2020-06-27:
I am also interested in using some of the pins in PCF8574 as inputs. Any ideas?

snaiperr, 2020-09-12:
Dear All, 
please share information how to check the state of PCF8574 as input

1Ombree, 2020-10-05:
> I have compiled the "basic 8.1" and enabled MCP230xx and PCF8574
> 
> [tasmota_8.1_MCP230xx+PCF8574.zip](https://github.com/arendst/Tasmota/files/4272099/tasmota_8.1_MCP230xx%2BPCF8574.zip)
> 
> With 600 kb it is pretty big, so you need to flash an minimal firmware first.

Could you please confirm if I should first firmware the ESP with tasmota-minimal.bin (from i.e. http://thehackbox.org/tasmota/release/) and after that firmware with file from you?
Is it Ok to use first 8.5.0 (minimal) and that 8.1.0 (tasmota_8.1_MCP230xx+PCF8574)?