arendst/Tasmota #6767: Local access security issue due to HTTP CORS header

pseudosavant, 2019-10-27:
### BUG DESCRIPTION
Tasmota has CORS HTTP headers enabled by default. This is a major security issue that could easily be exploited by any website with some simple Javascript, especially since Tasmota does not require a web password by default.

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _Shelly 1 and Sonoff Basic, but this issue isn't device specific_
- [X] Tasmota binary firmware version number used: _6.7_
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: _FlashESP8266.exe_
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:

```
17:36:26 RSL: stat/sonoff/RESULT = {"NAME":"Shelly 1","GPIO":[0,0,0,0,21,82,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
17:36:26 RSL: stat/sonoff/RESULT = {"Module":{"46":"Shelly 1"}}
17:36:27 RSL: stat/sonoff/RESULT = {"GPIO":"Not supported"}
```

- [X] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
```
N/A  
```
  
- [X] Provide the output of this command: ``Status 0``:

```
17:38:36 RSL: stat/sonoff/STATUS = {"Status":{"Module":46,"FriendlyName":["Garage Shelly"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:38:36 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/020502/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:06","StartupUTC":"2019-10-27T16:33:30","Sleep":50,"CfgHolder":4617,"BootCount":765,"SaveCount":785,"SaveAddress":"FB000"}}
17:38:36 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
17:38:36 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ellis-ac",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000"]}}
17:38:36 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
17:38:36 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"garage-door-opener","IPAddress":"192.168.1.175","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A4:CF:12:BA:24:E5","Webserver":2,"WifiConfig":4}}
17:38:36 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"router.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BA24E5","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:38:36 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 27 16:38:36 2019","Local":"Sun Oct 27 17:38:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:28","Sunset":"17:39"}}
17:38:36 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-27T17:38:36","Switch1":"OFF"}}
17:38:36 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-27T17:38:36","Uptime":"0T00:05:06","UptimeSec":306,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ellis-ac","BSSId":"F4:F2:6D:9C:23:B5","Channel":1,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```

- [X] Provide the output of the Console log output when you experience your issue; if applicable: _(Please use_ ``weblog 4`` _for more debug information)_
```
N/A  
```

### TO REPRODUCE
1. Make any HTTP request to a Tasmota device (i.e. `curl --user username:password --head 192.168.1.100`
2. See that `Access-Control-Allow-Origin: *` CORS header is included as an HTTP response header
3. Go to [Tasmota Device Locator](http://tasmota.simplethings.work/)
4. See that TDL is able to make arbitrary HTTP requests to every Tasmota device on my network

### EXPECTED BEHAVIOUR
1. CORS is disabled by default
2. Any public Internet website should *not* be able to query/enumerate Tasmota devices on my local network

### SCREENSHOTS
_N/A_

### ADDITIONAL CONTEXT

It would be very easy to create a script could be dropped into any web page that would:

1. Enumerate every IP in common subnets (192.168.1.x, 192.168.0.x, 10.0.0.x, etc) to discover Tasmota devices
2. Once Tasmota devices have been identified, if they are password protected the script could attempt to bruteforce the password
3. Once a password is discovered, or if the device isn't password protected (default), the site has complete control of the Tasmota device. They can reflash the firmware.
4. By replacing the firmware a malicious actor could:
   1. Setup a proxy server to enable arbitrary access to any local network device (not just Tasmota)
   2. Push malicious payloads (hacked IP webcam firmware, etc) to local network devices
   3. Contact a command-and-control service to wait for further instructions

This attack vector could largely be mitigated by making CORS configurable (disabled by default), and requiring a web password to be set when first configuring the wifi. Those that prefer to control their Tasmota devices using HTTP requests could still use that method by using an HTTP password and/or enabling CORS.

Jason2866, 2019-10-27:
A PR is welcome ;-)

arendst, 2019-10-27:
Once upon a time there was this #5852

arendst, 2019-10-28:
Thx for the elaborate information.

As you suggested I will disable sending Access-Control-Allow-Origin by default and allow a user to enable it with a SetOption command.

BTW Nice tool this Tasmota Device Locator

meingraham, 2019-10-28:
So, with this update to disable CORS by default, users will not be able to use the "live" version of TDL and will have to [deploy TDL on their own local server](https://github.com/arendst/Tasmota/wiki/Tasmota-Device-Locator#deployment)?

I will make a note in the [TDL wiki article](https://github.com/arendst/Tasmota/wiki/Tasmota-Device-Locator) if this is so.

arendst, 2019-10-28:
Just tested it and as expected it doesn't find it (192.168.2.223) on the local network too.

```
Access to XMLHttpRequest at 'http://192.168.2.223/cm?cmnd=Module' from origin 'http://domus1' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.
xhr.js:178 Cross-Origin Read Blocking (CORB) blocked cross-origin response http://192.168.2.223/cm?cmnd=Module with MIME type application/json. See https://www.chromestatus.com/feature/5629709824032768 for more details.
```

This will make TDL useless but increases security ;-)

arendst, 2019-10-28:
I did some research and trial-and-error to look at the consequences of disabling CORS.

Regarding TDL I think the author has to re-think his solution. There must be other ways to scan for Tasmota devices with CORS disabled.

TasmoAdmin is still able to find the device with CORS disabled.

I will release command ``SetOption73 0/1`` with default CORS disabled (0).

ascillato2, 2019-10-28:
Closing this issue as it has been fixed. Thanks for reporting.

pseudosavant, 2019-10-28:
Wow, what a fast turnaround! Thanks! This is such a great project!

bananenfisch, 2021-05-16:
It seems, this security issue is still open. The CORS option can be manipulated by any website itself.

For example:
- disable CORS on your devices
- go to http://tasui.shantur.com/#/findDevices
- hit the checkbox "Enable CORS for TasUI" (Note: another website could just do it without asking)
- the first AJAX request will be: "http://192.168.1.11/cm?cmnd=CORS http://tasui.shantur.com"
- after that, the website can send any command via AJAX requests to the internal web-devices

I think, this is still a huge security issue?!

ascillato, 2021-05-16:
If you can send any command, CORS is not the main issue. If you delete CORS from Tasmota, but you still has no password in your Tasmota device, you can send any command to it. So, CORS is not the problem. CORS is just another communication channel.

The main issue is leaving your device unprotected without password. If you set a password, CORS can not be enabled.

bananenfisch, 2021-05-16:
Setting passwords is a different topic. By default there is no pwd on the devices.
Anyway, IMO it's an issue. If CORS is disabled, it should not be possible to enable it again via CORS. The option to disable is useless, if a website can easily get permission by sending a simple request (cmnd=CORS http://blah.com).

ascillato, 2021-05-16:
If you don't have a password, it is trivial if you have or not CORS in your build.

You can already send any command to your device.

From your example:

> the first AJAX request will be: "http://192.168.1.11/cm?cmnd=CORS http://tasui.shantur.com"

That can already be ANY Tasmota command.

So, if a user want to secure his/her setup, a password must be used AND also the recommendations stated in the docs.

bananenfisch, 2021-05-16:
Again: same origin policy is a DIFFERENT security feature than setting a password. If you have an security issue in one component of a software, it should be fixed... not just say: use another feature.

You miss the point of AJAX and CORS. Its not the question, if you "can already [send] ANY Tasmota command". The security issue is, that with this implementation of CORS ANY website can send a request into your LOCAL network. Yes, you can setup a password - but its a HUGE issue to kick out same origin policy to open that door!

So, lets say a "normal" user, without knowing things about same origin or CORS have his private LAN behind his router. Well, most people would say: "ah, thats ok - there is no way to access my tasmota devices from outside of my LAN". Normally this is true because of same origin. Every website can request "cm?cmnd=CORS" through JavaScript, which runs on the client - in your LAN. This way, it has full control over all devices inside your LAN.

Thats why, every API, which should be used inside a LAN never ever enables CORS! To disable CORS but keep it possible to enable CORS via CORS is weird, sorry.

ascillato, 2021-05-16:
CORS is disabled by default.

https://github.com/arendst/Tasmota/blob/fbb9df987e1fa2bbe92b793487ff782ba50dfe02/tasmota/my_user_config.h#L174

What else do you propose to do?

bananenfisch, 2021-05-16:
> CORS is disabled by default.
> What else do you propose to do?

No - you can still enable it through JavaScript over CORS. Thats the point.

To reproduce:
If you send a HTTP request over AJAX like:
"local-ip/cm?cmnd=toggle" -> same policy works

But the issue, if you send a request over AJAX like:
"local-ip/cm?cmnd=CORS http://mybadwebsite.com" -> ACCEPTED! and CORS enabled after this simple request!

The fix would be: enabling CORS only on the device itself, NOT over the API!

ascillato, 2021-05-16:
Ok, so your fix is just deleting the command `CORS` from Tasmota and allowing only the CORS configuration from the Wi-Fi Configuration Menu ?

I agree to that, but if the Tasmota device don't have a password, you can do whatever you want using the API too, like changing the whole firmware over OTAurl. So, the most important thing is to put a password to your device to avoid all that.

I agree to delete that command, but this will add an extra configuration step for **_TasUI_** and @shantur won't be happy.

Please, can you provide a PR with this removal? We can further discuss this there.

bananenfisch, 2021-05-16:
Easier to fix is the real bug: if CORS is disabled (it is by default), do not send CORS headers when enabling CORS via API. Because this is useless. Its like to set a password and print it on the login page ;-)

**I know that thing with the password**... this is a complete different topic. Its not the normal behavior to provide an API (with disabled CORS) and give a command to enable it from **any** website. There are many users out there, who have not setup a password because they trust, that there is no access inside their LAN.

In fact: all tasmota devices without password (ALSO when behind a router with or without firewall) can be **full accessed** by **any** website, which is opened in a browser from inside the LAN.

ascillato2, 2021-05-16:
Let's reopen for further discussion to find the best way to fix this.

pseudosavant, 2021-05-18:
Maybe this is a dumb question, but can anyone actually reproduce this issue? I am correctly getting denied when CORS is enabled. Unless your browser is unusually configured to not check for the `Access-Control-Allow-Origin` header, the browser should prevent making any requests. The enforcement of CORS is all on the client, so it only affects browsers. Anything else (e.g. `curl`) can make any request it wants to any endpoint.

[image]
[image]

I used this command in the browser dev tools console while on http://tasui.shantur.com/.

```js
(async () => {
    console.log(await fetch('http://192.168.1.107/cm?cmnd=CORS http://tasui.shantur.com'));
})();
```

bananenfisch, 2021-05-18:
> Maybe this is a dumb question, but can anyone actually reproduce this issue?

You should reproduce it by sending the request twice - or just using http://tasui.shantur.com/#/findDevices (take a look to the network console).

But you are right: the same-origin-policy is a security feature of the browser itself (not on the server), but an important one (a website shouldn't be allowed to fetch HTTP requests into your LAN).

IMO, the problem is the architecture of the API. There are 2 types of requests:
- Simple requests*
- Preflighted requests*
* https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS

While preflight requests are useful, because the browser would send an OPTION request to the server to check if CORS is allowed **before** doing the real request, simple requests are fired once and only the result will be blocked by the browser!

All requests of this API are simple requests, since it uses always the GET method and send the data via URL parameter. Normally for changing a resource, its recommended to use POST or PUT and transfer the data in the body as JSON.

Why?... Because a GET to a resource should just read out data. So there is no need to check **before** via preflight request, because a simple request should never change something (it's enough to block the result). So this is the first problem: the request is done, it changes the CORS setting even if the browser blocks the result. You also can try to do some other commands, like power toggle...

I think, if it is too much work to change the Web-API to use POST/PUT requests with request body in an JSON for changing data and settings (which will end in preflight requests and CORS working out of the box), you should force to set a own password on first installation, IMO!

Too many users out there, who trust on their private network... while any website can use the tasmota API to walk into it... doing bad things as you described in the bug description...

pseudosavant, 2021-05-18:
I tried submitting multiple requests and using TasUI, but I have passwords on all my devices. It did reproduce when I removed my password however.

The first request is shown as a CORS error in the dev tools, but the second one does work. Is Tasmota treating the preflight `OPTIONS` request as a `GET`? That way the CORS setting gets set and future requests work as you'd expect with proper a CORS header.

This is definitely a security bug. Even if no password is set, you shouldn't be able to make any requests to a device on a different origin than the page you are on.

bananenfisch, 2021-05-19:
The OPTIONS request is not the main bug (but maybe the 2nd step to solve this issue). I do not see any OPTIONS request, when i try to access the API via JavaScript. The browsers differs between "Simple requests" and "Preflighted requests". A request via JavaScript with method GET and data transfer with mime type text/plain or application/x-www-form-urlencoded are done as simple requests without a preflight OPTIONS request.

So this way, a solution could be to check the same origin on the server side - but this is not the job for the server, little dirty fix.

The cleanest solution would be to change the API:
- GET requests only for reading out data (status, etc.)... the browser will block the response automatically if CORS is disabled.
- changing settings: use POST or PUT requests, and put the parameters into the body... in this case, the browser will automatically send a preflight OPTIONS request to check the CORS status (we need to check, if OPTIONS requests are working here). This way, the browser will block the request before changing any settings or states.

The clean solution will also close the door to other cross site attacks.
For example, its always possible to send a simple GET request with other techniques.

Lets try this - a website could load something like this:
`<?php
for ($i = 0; $i < 256; $i++) {
  echo '[image]';
  echo '[image]';
}
`

If we agree, that some-origin is an important security mechanism, then the API **must use** POST/PUT requests for changing state/settings requests, because with simple GET requests CORS will not work, nor its safe with other kinds of cross-site attacks.

If there is no agreement for implementations against this cross-site things, then IMO it's needed to force all users to set a (good) password (because as you said: bruteforce over JavaScript requests is still possible)!

bananenfisch, 2021-05-19:
I've implemented a little test: https://www.bananenfisch.net/fun.php
Note: there is a warning. If you click the link on this page, this code will be executed:
```
<body>
If you did not set a password on your tasmota devices and they are in the 192.168.1 subnet, then all power states are toggled now... please check your lights / water systems / garage door openers / roller shutter / etc.
<?php
for ($i = 2; $i < 255; $i++) {
  echo '<img src="http://192.168.1.' . $i . '/cm?cmnd=Power%20TOGGLE">';
}
?>
</body>
```

shantur, 2021-05-19:
@bananenfisch : According to https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS#simple_requests a POST request is also a simple request, unless I am not reading it correctly

shantur, 2021-05-19:
Having read https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS#simple_requests , a simple way to make all request preflighted is to add a non-standard header. It could be any static header that Tasmota needs to check before accepting the request

bananenfisch, 2021-05-19:
> @bananenfisch : According to https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS#simple_requests a POST request is also a simple request, unless I am not reading it correctly

The only allowed values for the Content-Type header are:
    application/x-www-form-urlencoded
    multipart/form-data
    text/plain

Normally you put the POST data into an JSON - that way it will become a "preflight request" and the browser will do all security work for you.

> Having read https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS#simple_requests , a simple way to make all request preflighted is to add a non-standard header.

How? Its all about **request headers** ... an attacker can implement his JavaScript requests as he like. The only way to prevent against cross site attacks is:
- use GET **only** for reading data
- use POST/PUT **always** for changing states/settings
- ignore GET requests, that are changing things on the server

That's the minimum standard for RESTful APIs. If you try to create an ugly work-around to get the CORS working, you still have this problem: [image]

This is really a big security issue... If you have a garage door opener with tasmota with no password (default), any person can send you a message: "hey, visit my new website: xxx"... voila! Welcome home!

**please force all users to setup a good password until this issue isn't closed**

pseudosavant, 2021-05-19:
I had never read about the preflight vs simple request aspect of CORS. I thought it always preflighted. Learning something new every day. I realized that I'm already ~~using~~ abusing this issue (via `<img>`) to make API requests to my IR blaster from an external domain: http://tasmota-ir-remote.glitch.me/.

I agree that it would be preferred to take a RESTish approach and use the 'correct' HTTP methods based on if you are reading or writing. It would be a *huge* breaking change though. And doing that will only fix part of the problem as it is still a significant security issue if attackers could extract data from any Tasmota device on a user's network.

The Fetch spec has the exact steps (see below) required to do a CORS check for requests *and* responses. It would seem Tasmota should conform to those same steps for its responses?

> To perform a CORS check for a request and response, run these steps:
> 1. Let origin be the result of getting `Access-Control-Allow-Origin` from response’s header list.
> 2. If origin is null, then return failure.
> 3. If request’s credentials mode is not "include" and origin is `*`, then return success.
> 4. If the result of byte-serializing a request origin with request is not origin, then return failure.
> 5. If request’s credentials mode is not "include", then return success.
> 6. Let credentials be the result of getting `Access-Control-Allow-Credentials` from response’s header list.
> 7. If credentials is `true`, then return success.
> 8. Return failure.

https://fetch.spec.whatwg.org/#concept-cors-check

bananenfisch, 2021-05-19:
> Having read https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS#simple_requests , a simple way to make all request preflighted is to add a non-standard header. It could be any static header that Tasmota needs to check before accepting the request

Hmmm... i was thinking about it a little more. Maybe this could really work (sure, its an ugly work-around, but maybe an option to close the issue quickly).

We should test this scenario:
* No changes in the API architecture, stay at GET
* But one change: all HTTP requests to the API needs a custom header from the client: the server will only respond if the header is set

Theoretically it would result in this (need to test this):
* Every AJAX request will become a preflight request because of the custom header - so if the server can handle the OPTIONS request, all is fine and CORS will work
* If the attacker set no header, the command will be blocked by the server
* If the attacker set the header, the CORS mechanism will block the request after OPTIONS request if CORS is disabled
* If the attacker tries to perform a request via some HTML src attribute: there is no way to set a custom header!

Lets do some tests...

shantur, 2021-05-20:
Indeed.

I have worked with Tasmota code and that's the easiest and least disruptive work-around.
The only thing needs to be figured out what happens with the pages served by Tasmota itself. Would it be possible to add the header there?

ascillato2, 2021-07-09:
Hi, any news on this?

Would be Ok if until a fix is submitted to disable CORS commands and preflights responses?

arendst, 2021-07-10:
OK with me. Let's see what the response is.

bananenfisch, 2021-07-28:
I strongly recommend to force all users to setup a password and not just disable CORS.
The easiest fix would be to set a custom header, which should be checked before the API perform the action.

CORS is just one part of the problem. Every visited website can take control over ALL tasmota devices with no password set, as simple: [image]

Our tasmota devices are used for doors openers, not just for lights - it's a huge security issue. IMO the only possible way is: force to set a initial password until this isn't fixed.

closed 2021-08-05 by arendst

btsimonh, 2023-09-10:
An interesting and informative read.

One observation is that the access to the webcam video via <img> tag works, even though it is actually cross-origin (because /stream is on port 81).  But a `fetch` against :81/stream fails with CORS error, even if the requesting HTML is served from the specific tasmota from which you want to read the stream.

rew1ndustry, 2024-06-01:
i really do not get this - what is the point of having a web API at all, if you will not allow CORS?

could someone please point me to a rational explanation?

s-hadinger, 2024-06-01:
Please open a new discussion with your detailed question. This issue is closed.