arendst/Tasmota #6773: Switches (buttons) not sending autodiscovery topic

gabter1, 2019-10-28:
I have setup a Wemos D1 mini with Tasmota, connected to a dual relay module. I also have two inputs on GPIO2 and GPIO4 which get information from my wall thermostat dry contact relay.
This is what Tasmota setup looks like:
[image]
Everything is working fine, the switches do drive the relays as they are supposed to and I can manually configure the binary_switches in Home Assistant.
But I would like to have the binary_switches to be auto discovered in HASS as well (the relays are discovered nicely).
This is what the console log shows:
```
00:00:00 CFG: Loaded from flash at F7, Count 636
00:00:00 Project sonoff D1mini1 Version 6.6.0(sonoff)-2_4_2
00:00:00 WIF: Connecting to AP1 xxxxxxx in mode 11N as sonoff-1232...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-1232 with IP address 192.168.xxx.xxx
20:30:29 MQT: Attempting connection...
20:30:29 MQT: Connected
20:30:29 MQT: sonoff/tele/LWT = Online (retained)
20:30:29 MQT: sonoff/cmnd/POWER = 
20:30:29 MQT: sonoff/tele/INFO1 = {"Module":"Generic","Version":"6.6.0(sonoff)","FallbackTopic":"cmnd/DVES_35A4D0_fb/","GroupTopic":"sonoffs"}
20:30:29 MQT: sonoff/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1232","IPAddress":"192.168.xxx.xxx"}
20:30:29 MQT: sonoff/tele/INFO3 = {"RestartReason":"Software/System restart"}
20:30:29 MQT: sonoff/stat/RESULT = {"POWER1":"OFF"}
20:30:29 MQT: sonoff/stat/POWER1 = OFF (retained)
20:30:29 MQT: sonoff/stat/RESULT = {"POWER2":"OFF"}
20:30:29 MQT: sonoff/stat/POWER2 = OFF (retained)
20:30:30 MQT: homeassistant/light/35A4D0_LI_1/config =  (retained)
20:30:30 MQT: homeassistant/switch/35A4D0_RL_1/config = {"name":"D1mini1","cmd_t":"~cmnd/POWER1","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER1}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"35A4D0_RL_1","device":{"identifiers":["35A4D0"]},"~":"sonoff/"} (retained)
20:30:30 MQT: homeassistant/light/35A4D0_LI_2/config =  (retained)
20:30:30 MQT: homeassistant/switch/35A4D0_RL_2/config = {"name":"D1mini2","cmd_t":"~cmnd/POWER2","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER2}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"35A4D0_RL_2","device":{"identifiers":["35A4D0"]},"~":"sonoff/"} (retained)
20:30:30 MQT: homeassistant/light/35A4D0_LI_3/config =  (retained)
20:30:30 MQT: homeassistant/switch/35A4D0_RL_3/config =  (retained)
20:30:30 MQT: homeassistant/light/35A4D0_LI_4/config =  (retained)
20:30:30 MQT: homeassistant/switch/35A4D0_RL_4/config =  (retained)
20:30:30 MQT: homeassistant/light/35A4D0_LI_5/config =  (retained)
20:30:30 MQT: homeassistant/switch/35A4D0_RL_5/config =  (retained)
20:30:30 MQT: homeassistant/light/35A4D0_LI_6/config =  (retained)
20:30:30 MQT: homeassistant/switch/35A4D0_RL_6/config =  (retained)
20:30:30 MQT: homeassistant/light/35A4D0_LI_7/config =  (retained)
20:30:30 MQT: homeassistant/switch/35A4D0_RL_7/config =  (retained)
20:30:30 MQT: homeassistant/light/35A4D0_LI_8/config =  (retained)
20:30:30 MQT: homeassistant/switch/35A4D0_RL_8/config =  (retained)
20:30:30 MQT: homeassistant/sensor/35A4D0_status/config = {"name":"D1mini1 status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"35A4D0_status","device":{"identifiers":["35A4D0"],"name":"D1mini1","model":"Generic","sw_version":"6.6.0(sonoff)","manufacturer":"Tasmota"},"~":"sonoff/tele/"} (retained)
20:30:37 MQT: sonoff/tele/STATE = {"Time":"2019-10-27T20:30:37","Uptime":"0T00:00:14","Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxxxxx","BSSId":"xxxxxxxxxxxxxxxxx","Channel":9,"RSSI":20,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:30:37 MQT: sonoff/tele/SENSOR = {"Time":"2019-10-27T20:30:37","Switch1":"OFF","Switch2":"OFF"}
20:35:25 MQT: sonoff/tele/HASS_STATE = {"Version":"6.6.0(sonoff)","BuildDateTime":"2019-07-06T13:35:03","Core":"2_4_2","SDK":"2.2.1(cfd48f3)","Module":"Generic","RestartReason":"Software/System restart","Uptime":"0T00:05:02","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:04","MqttCount":1,"BootCount":44,"SaveCount":637,"IPAddress":"192.168.xxx.xxx","RSSI":"18","LoadAvg":19}
20:35:37 MQT: sonoff/tele/STATE = {"Time":"2019-10-27T20:35:37","Uptime":"0T00:05:14","Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SSId":"xxxxxxxxxxx","BSSId":"xxxxxxxxxxxxxxxxx","Channel":9,"RSSI":18,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:35:37 MQT: sonoff/tele/SENSOR = {"Time":"2019-10-27T20:35:37","Switch1":"OFF","Switch2":"OFF"}'''
```

As you can see, only the relays are sent to be auto discovered, the binary switches are not. According to the wiki , those should be auto discovered, too. Any ideas on how to get them auto discovered. It would help a lot in getting their states after a restart.

effelle, 2019-10-28:
Hi @gabter1 ,

`Button` and `Swithces` autodiscovery was re-introduced on 6.7.1 (you are running the old stable 6.6.0).
If you  have any other question please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 
Thanks.

Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

effelle, 2019-10-28:
Additional note: don't use `GPIO2` for a switch, it has [restrictions](https://github.com/arendst/Tasmota/wiki/Expanding-Sonoffs#restrictions) (can't be low on boot).
Use `GPIO13` and `GPIO14` instead.

gabter1, 2019-10-28:
Thanks Federico, I will try to update when I get home and refer back. I don't yet see the change log for 6.7.1 and I was keeping to "don't fix what's not broken" :)

effelle, 2019-10-28:
Lot of changes on latest release, I pushed the fix the same day of 6.7 announcement. :P

ascillato2, 2019-10-28:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-28 by ascillato2

gabter1, 2019-10-29:
Works fine with the 6.7.1 firmware, thank you.
I change the GPIO as well, thanks for the heads up.

effelle, 2019-10-29:
Your're welcome!