arendst/Tasmota #6776: RFrecv stopped working

bjeram, 2019-10-28:
I have Sonoff Dual R2 (esp8285) with added a RF receiver  module to pin GPI09.
REceiving data worked with Tasmota 6.3.0, but it does not work anymore with the latest version of it.
As requested in the newer version I needed to enable it in my_user_config.h:
#define USE_RC_SWITCH                     
#define USE_RF_SENSOR                     
  #define USE_THEO_V2    
  #define USE_ALECTO_V2
Any idea what could be the source of the problem?

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Dual R2 (esp8285) with RF receiver  module
- [x ] Tasmota binary firmware version number used: 6.6.0
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [x ] IDE / Compiler used: PlatformIO and Arduino IDE
- [x] Flashing tools used: provided with IDE

meingraham, 2019-10-28:
Use the latest release. The current version is 7.0.1

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-28 by ascillato2

bjeram, 2019-12-16:
I've done some more investigation.
First I thought it is related to using esp8285 (Sonoff Dual R2), but it was the same on other "raw" ESP8266 like Wemos.
The problem was/is that pin where RF module sends data is not set to INPUT.
I've fixed this by adding command pinMode( .... , INPUT); to: xdrv_17_rcswitch.ino.
I'll creat a PR with the mod.

PR #7236 (2019-12-16): setting pinMode for the pin used by RFRecv to INPUT
## Description:
After a certain version receiving RF stopped working, what looks like is a problm due to not set pin to INPUT.
**RFrecv stopped working:** fixes  #6776

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob [... cut]