arendst/Tasmota #6779: Custom Topic Appended To GroupTopic

w35m17h, 2019-10-29:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

meingraham from Discord asked me to open this bug report.

When deviating from the standard topic (%prefix%/%topic%/) the standard use of group topic ie cmnd/tasmotas/somecommand no longer works. But appending the additional terms used in the topic to the group topic works. Not sure if that is the way it is intended to work. For example, If I change my topic to %prefix%/kitchen/%topic%/ I have to publish to cmnd/kitchen/tasmotas/somecommand to make group topic work. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _Wemos, sonoff S31, S20 and Basic
- [ x] Tasmota binary firmware version number used: 6.5.0.12, 6.6 and 7.0.0.1 
  - [ x] Pre-compiled
  - [ x] Self-compiled
    - [ x] IDE / Compiler used: Atom/Platformio
- [ ] Flashing tools used: Atom/Platformio
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
01:57:24 CMD: Backlog Template; Module; GPIO
01:57:25 MQT: stat/wemos/kitchen/fridge/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
01:57:25 MQT: stat/wemos/kitchen/fridge/RESULT = {"Module":{"18":"Generic"}}
01:57:25 MQT: stat/wemos/kitchen/fridge/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"21":"Relay1"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

  ```
- [x ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here: N/A

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
01:58:25 CMD: Status 0
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS = {"Status":{"Module":18,"FriendlyName":["Fridge Water"],"Topic":"fridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T02:01:30","StartupUTC":"2019-10-28T22:56:55","Sleep":50,"CfgHolder":4617,"BootCount":1,"SaveCount":21,"SaveAddress":"F7000"}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS2 = {"StatusFWR":{"Version":"7.0.0.1(9b0f16f-tasmota)","BuildDateTime":"2019-10-28T22:00:12","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS5 = {"StatusNET":{"Hostname":"kitchen-fridge-water","IPAddress":"192.168.4.44","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"5C:CF:7F:F0:E0:08","Webserver":2,"WifiConfig":4}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.4.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F0E008","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 29 00:58:25 2019","Local":"Tue Oct 29 01:58:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:32","Sunset":"17:35"}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS10 = {"StatusSNS":{"Time":"2019-10-29T01:58:25"}}
01:58:25 MQT: stat/wemos/kitchen/fridge/STATUS11 = {"StatusSTS":{"Time":"2019-10-29T01:58:25","Uptime":"0T02:01:30","UptimeSec":7290,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iot","BSSId":"02:9F:C2:71:C6:55","Channel":6,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:05"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

N/A
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Change default topic from %prefix%/%topic%/ to something like this %prefix%/test/%topic%/

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

When publish a command to cmnd/tasmotas/somecommand I expect all devices with the grouptopic = tasmotas to respond

Instead I have to publish to cmnd/test/tasmotas/somecommand to accomplish this

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-10-29:
It's intended and I see the issue here.

I'll need some thought how to tackle this other than decide to keep the grouptopic fixed as ``cmnd/tasmotas/somecommand``.

In that case I forsee the HAss users complaining as their default fulltopic is ``tasmotas/cmnd/somecommand``

I think I take a long holiday....

joba-1, 2019-10-29:
how about publishing the source code along with a good wiki documenting everything? 
Then everybody should be happy, right? OMG :D

meingraham, 2019-10-29:
>  I forsee the HAss users complaining

I'm missing the problem :wink:

ascillato2, 2019-10-29:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

meingraham, 2019-10-29:
So, GroupTopic just replaces the %topic% portion of whatever FullTopic is. Is that accurate?

arendst, 2019-10-30:
Correct.

Perhaps I better change the grouptopic like the fallbacktopic. In that case it would always become ``cmnd/tasmotas/command``

arendst, 2019-10-31:
Added command ``SetOption75 0/1`` to switch between two grouptopic layouts.

- 0 = default grouptopic built from fulltopic replacing %topic% with grouptopic
- 1 = fixed grouptopic of ``cmnd/<grouptopic>``

After executing command ``SetOption`` a restart is performed to unsubscribe current group topic and re-subscribe new group topic

ascillato2, 2019-10-31:
Closing this issue as this feature has been added. Thanks :+1:

closed 2019-10-31 by ascillato2