arendst/Tasmota #6780: Dimmer command doesn't work with 2 byte ranges

igorbernstein, 2019-10-29:
// cc @shantur 

### BUG DESCRIPTION
Dimmer command doesn't work below 71% for a tuya based Moes DS01C.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):Moes DS01C
- [x] Tasmota binary firmware version number used: 7.0.0.1(9b0f16f-tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
04:56:01 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO"
04:56:01 SRC: Backlog
04:56:01 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
04:56:01 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
04:56:01 SRC: Backlog
04:56:01 CMD: Group 0, Index 1, Command "MODULE", Data ""
04:56:01 RSL: stat/sonoff/RESULT = {"Module":{"54":"Tuya MCU"}}
04:56:01 SRC: Backlog
04:56:01 CMD: Group 0, Index 1, Command "GPIO", Data ""
04:56:01 RSL: stat/sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
04:59:41 CMD: Group 0, Index 1, Command "STATUS", Data "0"
04:59:41 RSL: stat/sonoff/STATUS = {"Status":{"Module":54,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:59:41 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:28:15","StartupUTC":"2019-10-29T03:31:26","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":59,"SaveAddress":"F9000"}}
04:59:41 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"7.0.0.1(9b0f16f-tasmota)","BuildDateTime":"2019-10-28T22:00:12","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
04:59:41 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
04:59:41 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
04:59:41 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3475","IPAddress":"10.0.3.13","Gateway":"10.0.3.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.3.1","Mac":"XXXXXXX","Webserver":2,"WifiConfig":2}}
04:59:41 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA0D93","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
04:59:41 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 29 03:59:41 2019","Local":"Tue Oct 29 04:59:41 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:32","Sunset":"17:35"}}
04:59:41 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-29T04:59:41"}}
04:59:41 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-29T04:59:41","Uptime":"0T00:28:15","UptimeSec":1695,"Heap":26,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"MqttCount":0,"POWER":"ON","Dimmer":71,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXX","BSSId":"XXXXX","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}

  ```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  ```
05:02:49 HTP: Console
05:02:57 CMD: DimmerRange
05:02:57 SRC: WebConsole from 10.0.3.12
05:02:57 CMD: Group 0, Index 1, Command "DIMMERRANGE", Data ""
05:02:57 RSL: stat/sonoff/RESULT = {"DimmerRange":{"Min":180,"Max":1000}}
05:02:58 TYA: Send "55aa00000000ff"
05:02:58 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
05:02:58 TYA: Heartbeat
05:03:04 CMD: Dimmer 100
05:03:04 SRC: WebConsole from 10.0.3.12
05:03:04 CMD: Group 0, Index 1, Command "DIMMER", Data "100"
05:03:04 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100}
05:03:04 TYA: Send dim value=1000 (id=2)
05:03:04 TYA: Send "55aa0006000802020004000003e800"
05:03:04 {"TuyaReceived":{"Data":"55AA0307000802020004000003E804","Cmnd":7,"CmndData":"02020004000003E8","DpId":2,"DpIdType":2,"DpIdData":"000003E8"}}
05:03:04 TYA: fnId=21 is set for dpId=2
05:03:04 TYA: RX Dim State=1000
05:03:04 CFG: Saved to flash at F8, Count 60, Bytes 4096
05:03:07 CMD: Dimmer 80
05:03:07 SRC: WebConsole from 10.0.3.12
05:03:07 CMD: Group 0, Index 1, Command "DIMMER", Data "80"
05:03:07 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":80}
05:03:07 TYA: Send dim value=800 (id=2)
05:03:07 TYA: Send "55aa00060008020200040000032038"
05:03:07 {"TuyaReceived":{"Data":"55AA0307000802020004000003203C","Cmnd":7,"CmndData":"0202000400000320","DpId":2,"DpIdType":2,"DpIdData":"00000320"}}
05:03:07 TYA: fnId=21 is set for dpId=2
05:03:07 TYA: RX Dim State=800
05:03:07 CFG: Saved to flash at F7, Count 61, Bytes 4096
05:03:09 TYA: Send "55aa00000000ff"
05:03:09 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
05:03:09 TYA: Heartbeat
05:03:10 CMD: Dimmer 71
05:03:10 SRC: WebConsole from 10.0.3.12
05:03:10 CMD: Group 0, Index 1, Command "DIMMER", Data "71"
05:03:10 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":71}
05:03:10 TYA: Send dim value=710 (id=2)
05:03:10 TYA: Send "55aa0006000802020004000002c6dd"
05:03:10 CFG: Saved to flash at F6, Count 62, Bytes 4096
05:03:10 {"TuyaReceived":{"Data":"55AA0307000802020004000002C6E1","Cmnd":7,"CmndData":"02020004000002C6","DpId":2,"DpIdType":2,"DpIdData":"000002C6"}}
05:03:10 TYA: fnId=21 is set for dpId=2
05:03:10 TYA: RX Dim State=710
05:03:17 CMD: Dimmer 70
05:03:17 SRC: WebConsole from 10.0.3.12
05:03:17 CMD: Group 0, Index 1, Command "DIMMER", Data "70"
05:03:17 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":70}
05:03:17 TYA: Send dim value=706 (id=2)
05:03:17 TYA: Send "55aa0006000802020004000002c2d9"
05:03:17 {"TuyaReceived":{"Data":"55AA0307000802020004000002C2DD","Cmnd":7,"CmndData":"02020004000002C2","DpId":2,"DpIdType":2,"DpIdData":"000002C2"}}
05:03:17 TYA: fnId=21 is set for dpId=2
05:03:17 TYA: RX Dim State=706
05:03:17 CFG: Saved to flash at F5, Count 63, Bytes 4096
05:03:20 TYA: Send "55aa00000000ff"
05:03:20 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
05:03:20 TYA: Heartbeat
05:03:25 CMD: Dimmer 50
05:03:25 SRC: WebConsole from 10.0.3.12
05:03:25 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
05:03:25 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":50}
05:03:25 TYA: Send dim value=706 (id=2)
05:03:25 TYA: Send "55aa0006000802020004000002c2d9"
05:03:25 CFG: Saved to flash at F4, Count 64, Bytes 4096
05:03:25 {"TuyaReceived":{"Data":"55AA0307000802020004000002C2DD","Cmnd":7,"CmndData":"02020004000002C2","DpId":2,"DpIdType":2,"DpIdData":"000002C2"}}
05:03:25 TYA: fnId=21 is set for dpId=2
05:03:25 TYA: RX Dim State=706
05:03:25 SRC: Switch
05:03:25 CMD: Group 0, Index 1, Command "DIMMER", Data "71"
05:03:25 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":71}
05:03:25 TYA: Send dim skipped value=710
05:03:26 CFG: Saved to flash at FB, Count 65, Bytes 4096
05:03:31 CMD: Dimmer 25
05:03:31 SRC: WebConsole from 10.0.3.12
05:03:31 CMD: Group 0, Index 1, Command "DIMMER", Data "25"
05:03:31 RSL: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":25}
  ```

### TO REPRODUCE
1. Flash Moes DS01C switch with 7.0.0.1(9b0f16f-tasmota)
2. Configure as Tuya MCU (54): GPIO1-> TuyaTX,  GPIO3 -> TuyaRx
3. Map dpId: `TuyaMCU 21,2`
4. Movie slider all the way up and down to get range
5. Configure dimmer range: DimmerRange 180,1000
6. Set Dimmer below 71%: Dimmer 50

### EXPECTED BEHAVIOUR
Dimmer would be set to 50%, Tuya TX would send the value 500 (in hex)

### ACTUAL Behavior
The dimmer value doesnt go below 71%

blakadder, 2019-10-29:
Turn on `weblog 4` and run command `TuyaSend2 2,000003E8` and `TuyaSend2 2,00000000` and let me know the values are reported in Data for that dpId

shantur, 2019-10-29:
@igorbernstein : Fixed https://github.com/arendst/Tasmota/pull/6785

ascillato2, 2019-10-29:
Closing this issue as it has been fixed with proposed @shantur's PR .

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

PR #6785 (2019-10-29): TuyaMCU: Change dimmer scale before comparing with dimmer ranges. Fix…
Change dimmer scale before comparing with dimmer ranges Fixes #6780

## Description:

Dimmer range are hardware values not UI values.

**Related issue (if applicable):** fixes #6780 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core pre-2.6
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](htt [... cut]
commit 94fcff67 (2019-10-29): TuyaMCU: Change dimmer scale before comparing with dimmer ranges. Fixes #6780
closed 2019-10-29 by ascillato2