arendst/Tasmota #6781: Full topic resetting

webtop, 2019-10-29:
### BUG DESCRIPTION
After an OTA update to 7.0.0.1, MQTT Full Topic is set to "%topic%/%prefix%". Changing it yields no change after device restart. As you will see in the logs below, the topic comes before the prefix.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff T1 3CH Module
- [x] Tasmota binary firmware version number used: 7.0.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
12:03:54 MQT: gazebo-shade/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:03:54 MQT: gazebo-shade/stat/RESULT = {"Module":{"30":"Sonoff T1 3CH"}}
12:03:54 MQT: gazebo-shade/stat/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"}}
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
12:05:28 MQT: gazebo-shade/stat/STATUS = {"Status":{"Module":30,"FriendlyName":["gazebo-shade","Gazebo-Shade-Mid","Gazebo-Shade-Down"],"Topic":"gazebo-shade","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:05:28 MQT: gazebo-shade/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/tasmota-4m.bin","RestartReason":"Software/System restart","Uptime":"0T00:18:23","StartupUTC":"2019-10-29T03:47:05","Sleep":50,"CfgHolder":4617,"BootCount":30,"SaveCount":199,"SaveAddress":"F6000"}}
12:05:28 MQT: gazebo-shade/stat/STATUS2 = {"StatusFWR":{"Version":"7.0.0.1(9b0f16f-tasmota)","BuildDateTime":"2019-10-28T22:00:12","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
12:05:28 MQT: gazebo-shade/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ALLSOPP-2-4G",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["000A8109","2805C8000100060000005A00000000000000","00000200"]}}
12:05:28 MQT: gazebo-shade/stat/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:05:28 MQT: gazebo-shade/stat/STATUS5 = {"StatusNET":{"Hostname":"gazebo-shade-01","IPAddress":"192.168.0.36","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"75.75.75.75","Mac":"60:01:94:8A:6B:14","Webserver":2,"WifiConfig":4}}
12:05:28 MQT: gazebo-shade/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.19","MqttPort":1883,"MqttClientMask":"gazebo-shade","MqttClient":"gazebo-shade","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:05:28 MQT: gazebo-shade/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 29 04:05:28 2019","Local":"Tue Oct 29 12:05:28 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+08:00","Sunrise":"22:45","Sunset":"09:03"}}
12:05:28 MQT: gazebo-shade/stat/STATUS10 = {"StatusSNS":{"Time":"2019-10-29T12:05:28"}}
12:05:28 MQT: gazebo-shade/stat/STATUS11 = {"StatusSTS":{"Time":"2019-10-29T12:05:28","Uptime":"0T00:18:23","UptimeSec":1103,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"ALLSOPP-2-4G","BSSId":"F8:CF:C5:FF:1E:68","Channel":1,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:08"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
Unknown or OTA update firmware to:
http://thehackbox.org/tasmota/tasmota.bin

### EXPECTED BEHAVIOUR
Full topic should not have changed at all. Changing it back to its original value, or clearing the field completely should revert it to "%prefix%/%topic%/"

### SCREENSHOTS
Online image folder access: https://1drv.ms/u/s!Arg9f0GH8EvIhZMfN9Kn9SgQsGDvdg?e=eE5jv3
tas_err_01: view of config
tas_err_02: console view after device resets: full topic has reset to incorrect order
tas_err_03: console view showing device now cannot work with MQTT because messages are not understood (might work if I change the name)
tas_err_04: information screen

### ADDITIONAL CONTEXT
As it is, if I keep the name the same as it has been (with a hyphen) the MQTT broker does not understand the command requests, and so will not change the device state, or allow status updates.
As can be seen in the tas_err_03 image, calling "stat/gazebo-shade/POWER" causes the firmware to think the command is "GAZEBO" no matter whether topic precedes prefix or follows it.

Thx

arendst, 2019-10-29:
If you use HAss and enabled the corresponding SetOption (I don't know by heart) the fulltopic will change to %topic%/%prefix% as documented in some issues here and is related to HAss autodiscovery.

If you do not use HAss set the corresponding SetOption off as it might have been set before your upgrade to the latest version while it was not used.

The hyphen issue needs further investigation but might have something to do with the change to command decoding where additional separaters were introduced between command and parameter; a hyphen or minus is such a separater and results in your issue. If so the workaround is remove the hyphen from your topic. Longterm solution is re-evaluate command/parameter separation but as you can imagine the minus character was a big plus to introduce this command/separator code.

To be continued...

effelle, 2019-10-29:
Yes, if `SetOption19` was enabled, even after disabling it a manual revert of the order of  %topic%/%prefix%  will be necessary. 
As @arendst said, you can't use minus on topic name:
```
12:35:57 SRC: WebConsole from 192.168.15.199
12:35:57 CMD: gazebo-shade/stat/POWER
12:35:57 SRC: WebConsole from 192.168.15.199
12:35:57 CMD: Group 0, Index 1, Command "GAZEBO", Data "-shade/stat/POWER"
12:35:57 MQT: gazebo-shade/stat/RESULT = {"Command":"Unknown"}
```
Bu you can use the underscore:
```
12:38:27 CMD: gazebo_shade/stat/POWER
12:38:27 SRC: WebConsole from 192.168.15.199
12:38:27 CMD: Group 0, Index 1, Command "POWER", Data ""
12:38:27 MQT: gazebo_shade/stat/RESULT = {"POWER":"OFF"}
12:38:27 MQT: gazebo_shade/stat/POWER = OFF
```

ascillato2, 2019-10-29:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-29 by ascillato2

webtop, 2019-10-29:
Thank you guys, HAss auto-discovery was indeed the issue. Just an FYI on the hyphen issue, I use it on another device which I know doesn't have HAss auto-discovery enabled, and there are no issues with it.

effelle, 2019-10-29:
I don't remember on what revision we changed the decode. Perhaps you have a older version still not affected. 
What happens if you enable HAss discovery on the other (working) device? And what firmware is loaded on it?

webtop, 2019-10-29:
Version is 6.6.0, the device name is back-porch-light. SetOption19 to 1 changes the full topic as expected. Changing it back to 0, and then resetting the full topic causes no issues: everything back and working. Odd, but hey my issue is resolved :)

effelle, 2019-10-29:
Well,  is good for us to keep track of eventual issues. 
Thanks for your feedback!