arendst/Tasmota #6788: Motion sensor actuate relay and report status to Domoticz

kokangit, 2019-10-29:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  

I want to be able to control a light with a motion sensor. The light should turn on when motion is detected and turn off after a while. The status of the light as well as the status of the motion detector should be visible in Domoticz. It should also be possible to turn on and off the light from Domoticz.
I have configured 2 virtual devices i Domoticz, one for the light (relay) and another for the motion detector (switch). Both are type "Light/Switch" in Domoticz. They are configured in Tasmota as Idx 1 and Switch idx 1.
I defined a rule like this:
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":417,"Rules":"on switch1#state do backlog power1 1; ruletimer1 30 endon on rules#timer=1 do power1 off endon"}

In this configuration everything works but the state of the motion detector is not reported to Domoticz.

**Describe the solution you'd like**  

When the motion detector is triggered, the state should be sent to Domoticz, the relay should be triggered, the rule should be triggered.

**Describe alternatives you've considered**  

I did try with "switchtopic 1", but then the relay was not triggered. I also did setoption61 1 but the rule was never triggered.

**Additional context**  

I'm not sure if this is a bug or not, therefor describing it as a feature request.
I am using a sonoff mini with tasmota sonoff 6.7.1.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-10-29:
Closing this issue as it is **duplicated**. Sorry.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-29 by ascillato2

kokangit, 2019-10-29:
Sorry, is this a duplicate of someone requesting the same thing? I really tried to find the answer in the wiki, in the issues, asked in the forum group and the chat.

ascillato, 2019-10-29:
This is duplicated because was already addressed.

Please, check the wiki at rules. It is an example for PIR and rules. If you need further help, please address this to the chat.