arendst/Tasmota #6789: impossible to upgrade from 6.6.0.18 to newer firmwares

scargill, 2019-10-29:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
starting from 6.6.0.18 (2.3.0 based), can't go to anything newer as it says not enough memory, even going to minimal...

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic):  blitzwolf ss1
- [ ] Tasmota binary firmware version number used: 6.6.0.18 and 6.7.1 and 7.0.0.1 and 6.6.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: ota or file upload
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
16:12:15 CMD: Backlog Template; Module; GPIO
16:12:15 SRC: WebConsole from 192.168.14.126
16:12:15 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO"
16:12:16 SRC: Backlog
16:12:16 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
16:12:16 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
16:12:16 SRC: Backlog
16:12:16 CMD: Group 0, Index 1, Command "MODULE", Data ""
16:12:16 RSL: stat/sonoff/RESULT = {"Module":{"1":"Sonoff Basic"}}
16:12:16 SRC: Backlog
16:12:16 CMD: Group 0, Index 1, Command "GPIO", Data ""
16:12:16 RSL: stat/sonoff/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}
16:12:17 WIF: Checking connection...
16:12:17 WIF: Connected

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
16:13:00 CMD: Backlog Rule1; Rule2; Rule3
16:13:00 SRC: WebConsole from 192.168.14.126
16:13:00 CMD: Group 0, Index 1, Command "BACKLOG", Data "Rule1; Rule2; Rule3"
16:13:00 SRC: Backlog
16:13:00 CMD: Group 0, Index 1, Command "RULE", Data ""
16:13:00 RSL: stat/sonoff/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
16:13:00 SRC: Backlog
16:13:00 CMD: Group 0, Index 2, Command "RULE", Data ""
16:13:00 RSL: stat/sonoff/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
16:13:01 SRC: Backlog
16:13:01 CMD: Group 0, Index 3, Command "RULE", Data ""
16:13:01 RSL: stat/sonoff/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
16:13:22 CMD: Status 0
16:13:22 SRC: WebConsole from 192.168.14.126
16:13:22 CMD: Group 0, Index 1, Command "STATUS", Data "0"
16:13:22 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:13:22 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-minimal.bin","RestartReason":"Power on","Uptime":"0T00:36:41","StartupUTC":"2019-10-29T14:36:41","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":15,"SaveAddress":"FB000"}}
16:13:22 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.18(fb428a9-sonoff)","BuildDateTime":"2019-10-16T12:00:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:13:22 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["office",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
16:13:22 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":515,"Free":488,"Heap":17,"ProgramFlashSize":4096,"FlashSize":1024,"FlashChipId":"144068","FlashMode":2,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00003081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:13:22 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7528","IPAddress":"192.168.14.117","Gateway":"192.168.14.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.14.1","Mac":"4C:11:AE:15:7D:68","Webserver":2,"WifiConfig":4}}
16:13:22 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_157D68","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:13:22 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 29 15:13:22 2019","Local":"Tue Oct 29 16:13:22 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:32","Sunset":"17:35"}}
16:13:22 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-29T16:13:22"}}
16:13:22 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-29T16:13:22","Uptime":"0T00:36:41","UptimeSec":2201,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"office","BSSId":"00:1B:29:66:D1:B8","Channel":11,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:08"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
15:55:31 HTP: Firmware Upgrade
15:55:32 WIF: Checking connection...
15:55:32 WIF: Connected
15:55:39 CFG: Saved to flash at FB, Count 13, Bytes 4096
15:55:39 UPL: File sonoff-minimal-230-660.bin ...

  ```

### TO REPRODUCE
trying to update from 6.6.0.18 to newer version, going to minimal 6.7.1 before

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-10-29:
@scargill 

Did you upload a 4MB firmware?

If so, you can't do an OTA anymore, you will need to erase and flash by serial. Sorry

arendst, 2019-10-29:
Pete as already noted before your current binary image is 4M while you only have 1M flash.

,"ProgramFlashSize":4096,"FlashSize":1024,"

So indeed the only way to fix is serial uploadthe correct 1M binary.

scargill, 2019-10-29:
Right - I took the plunge and using my blog notes serially flashed the hackbox/tasmota/tasmota.bin file onto the S1, took off GPIO0, booted, filled all in with backlog - just need to remember temlate for the Blitzwolf SS1 - yup I must've programmed as 4 Meg. Now using nodemcu pyflasher.exe ("yes, wipe all data" and DIO, 11520 baud)  so that wont happen again... 

{"NAME":"Generic","GPIO":[255,255,255,255,56,21,255,255,0,17,255,255,255],"FLAG":15,"BASE":18}

So I FLASHED the board serially , used backlog to put in WIFI and MQTT details then manually updated config-other using "generic 18" with ports changed - and saved - then replicated the ports changes in config - module - no idea why both - but it works - and Alexa BELKIN doesnt work but Alexa ue Bridge DOES. All working- may thanks to Theo, Antonio, ALL.

closed 2019-10-29 by scargill