arendst/Tasmota #6792: All new Tuyas with intermediate microcontroller

irmi29, 2019-10-30:
hello there,

I use a lot of SONOFF and TUYA switches with TASMOTA firmware
The latest TUYA switches are of outstanding quality, one of the best fire resistance and assembly quality on the market. Probably the better choice for quality and safe switches for domestic.
But ... In recent months, all new models related to TUYA components are working through another microcontroller, which prevents the use of TASMOTA or any other firmware that is not original by TUYA. you can easily download tasmota to the ESP, but you still need to communicate with the other microcontroller.
[image]
[image]
[image]
[image]

Most components have deleted the component identity, so apart from the Tx/Rx link, the component type cannot be identified.

In another switches, the **nuvoton n76e003at20** microcontroller is used, but I suppose identity doesn't help figuring out how to operate the switches either.

Is this the end of using open firmware with TUYA?
It's probably a long shot, but is it possible to control these MCUs using Tasmota?

And another important issue and request, is it possible to add option to the official and excellent versions of TASMOTA, which will control any pair of relays as cover/shutter/curtain?
That is, to use 2 relays for cover(motor), or 4 relays for 2 covers(or one cover and 2 switches)?

Many thanks
Irmi.

andrethomas, 2019-10-30:
@shantur Isnt this the same as the TuyaMCU stuff you have been working on?

Jason2866, 2019-10-30:
As long Tuya protocol is used it does not matter which MCU is used.
You have to look at. Described in wiki https://github.com/arendst/Tasmota/wiki/TuyaMCU

blakadder, 2019-10-30:
I'm pretty sure TuyaMCU works with that unless its an unstandardized implementation of Tuya's Serial Protocol

shantur, 2019-10-30:
@andrethomas : Yes this will work with TuyaMCU 

@irmi29 : These are supported in Tasmota, check the link Jason posted above.

irmi29, 2019-10-30:
Thanks friends for the comment!

Currently I have only used version 6.6.0.10 since this version does not have a dead time of 5 seconds when manually turning the switch when there is no WIFI connection.
In any case, for testing I installed version 7.0.0.1, I set 3 relay switch as TUYAMCU and already had one relay initially works straight out of the box.

Here are the results:

Manually Switching on relay 1
> {"TuyaReceived":"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpId":1,"DpIdType":1,"DpIdData":"01"}}
> 11:47:50 TYA: fnId=11 is set for dpId=1
> 11:47:50 TYA: RX Relay-1 --> MCU State: On Current State:Off
> 11:47:50 SRC: Switch
> 11:47:50 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
> 11:47:50 RSL: stat/sonoff/POWER = ON
> 11:47:50 CFG: Saved to flash at FB, Count 134, Bytes 4096

 Switching off relay 1

> {"TuyaReceived":"Data":"55AA0007000501010001000E","Cmnd":7,"CmndData":"0101000100","DpId":1,"DpIdType":1,"DpIdData":"00"}}
> 11:49:30 TYA: fnId=11 is set for dpId=1
> 11:49:30 TYA: RX Relay-1 --> MCU State: Off Current State:On
> 11:49:30 SRC: Switch
> 11:49:30 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
> 11:49:30 RSL: stat/sonoff/POWER = OFF
> 11:49:30 CFG: Saved to flash at FA, Count 135, Bytes 4096

Assuming it does not support relay 2 and 3, so the result is different
 Switching on relay 2

> {"TuyaReceived":"Data":"55AA00070005020100010110","Cmnd":7,"CmndData":"0201000101","DpId":2,"DpIdType":1,"DpIdData":"01"}}
> 11:51:16 TYA: fnId=0 is set for dpId=2

and switching off relay 2

> 11:56:58"TuyaReceived":"Data":"55AA0007000502010001000F","Cmnd":7,"CmndData":"0201000100","DpId":2,"DpIdType":1,"DpIdData":"00"}}
> 11:56:58 TYA: fnId=0 is set for dpId=2

Will the version still in development and in future it will support even more than a single relay?

shantur, 2019-10-30:
@irmi29 : Do command `backlog TuyaMcu 12,2; TuyaMcu 13,3` in console. Let it reboot and then read

https://github.com/arendst/Tasmota/wiki/tuyamcu-configurations

ascillato2, 2019-10-30:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-30 by ascillato2

shantur, 2019-10-31:
@irmi29 : I haven't tried shutters switch with TuyaMCU. It needs to be looked at.

rt400, 2020-07-23:
@irmi29 @shantur any new with this module ?
i have the same and cannot handle with it..