arendst/Tasmota #6796: Command for wifi init (recalibration)

Jason2866, 2019-10-30:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**  
Yes.

**Is your feature request related to a problem? Please describe.**  
Some devices have bad wifi performance after upgrading

**Describe the solution you'd like**  
User command to erase SDK parameter area at end of linker memory model (0x0FDxxx - 0x0FFFFF) solving possible wifi errors

**Describe alternatives you've considered**  
Using `reset 5` or `reset 6` but needs reconfiguration of device

arendst, 2019-10-30:
Did you try ``reset 3``?

Jason2866, 2019-10-31:
Thx, how could i miss ;-)

closed 2019-10-31 by Jason2866