arendst/Tasmota #6797: Blitzwolf SHP6 with no load reports apparent power value > 0

tiagofreire-pt, 2019-10-30:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

Blitzwolf SHP6 with no load reports apparent power value > 0

```
Voltage | 234 V
0.112 A
0 W
26 VA
26 VAr
0.01
0.000 kWh
0.000 kWh
0.000 kWh
```
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Blitzwolf SHP6
- [X] Tasmota binary firmware version number used: 6.60
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: esphome.py
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
stat/blitzwolf_shp6_1/RESULT = {"NAME":"BlitzWolf SHP","GPIO":[57,255,56,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
stat/blitzwolf_shp6_1/STATUS = {"Status":{"Module":0,"FriendlyName":["Blitzworf_SHP6_1"],"Topic":"blitzwolf_shp6_1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
stat/blitzwolf_shp6_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:32","StartupUTC":"2019-10-30T22:56:03","Sleep":50,"CfgHolder":4617,"BootCount":14,"SaveCount":127,"SaveAddress":"F5000"}}
stat/blitzwolf_shp6_1/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
stat/blitzwolf_shp6_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["PSA_WIFI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
stat/blitzwolf_shp6_1/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
stat/blitzwolf_shp6_1/STATUS5 = {"StatusNET":{"Hostname":"blitzwolf_shp6_1-1567","IPAddress":"192.168.10.139","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.72","Mac":"84:F3:EB:EC:A6:1F","Webserver":2,"WifiConfig":4}}
stat/blitzwolf_shp6_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.61","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ECA61F","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
stat/blitzwolf_shp6_1/STATUS7 = {"StatusTIM":{"UTC":"Wed Oct 30 23:01:36 2019","Local":"Thu Oct 31 00:01:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:35","Sunset":"17:32"}}
stat/blitzwolf_shp6_1/STATUS9 = {"StatusPTH":{"PowerDelta":20,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":10000}}
stat/blitzwolf_shp6_1/STATUS10 = {"StatusSNS":{"Time":"2019-10-31T00:01:36","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":4,"ApparentPower":29,"ReactivePower":28,"Factor":0.13,"Voltage":234,"Current":0.122}}}
stat/blitzwolf_shp6_1/STATUS11 = {"StatusSTS":{"Time":"2019-10-31T00:01:36","Uptime":"0T00:05:33","Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"PSA_WIFI","BSSId":"2C:FD:A1:3E:31:58","Channel":9,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:06"}}}
00:01:40 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T00:01:40","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":28,"ReactivePower":28,"Factor":0.01,"Voltage":234,"Current":0.119}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
No load used.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Apparent power and current values should be as much close to zero as possible.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

tiagofreire-pt, 2019-10-30:
This might be solved with some kind of non linear function, being able to template within the GUI.

ascillato, 2019-10-30:
Please, update. Your Tasmota version is old and this has been fixed. Thanks

tiagofreire-pt, 2019-10-31:
Despite upgrading to 6.7.1, the issue remains:

```

Voltage | 232 V
-- | --
0.134 A
1 W
31 VA
31 VAr
0.03
0.000 kWh
0.000 kWh
0.000 kWh
```
On which version was this corrected?

meingraham, 2019-10-31:
Have you performed a power calibration?

tiagofreire-pt, 2019-10-31:
> Have you performed a power calibration?

Yes, I have. Both with a 60 W incandescent type bulb and with a 2000 W heater.

Jason2866, 2019-10-31:
Please read https://github.com/arendst/Tasmota/issues/4727#issuecomment-451741078 or better the complete issue.
Short story: The SHP6 is pitty. Needs a **perfect!!** calibration or you have to change the measuring resistor.
Calibration has to be (and can only be) done for this device with [method 2](https://github.com/arendst/Tasmota/wiki/Power-Monitoring-Calibration#method-2)

tiagofreire-pt, 2019-10-31:
> 
> 
> Please read [#4727 (comment)](https://github.com/arendst/Tasmota/issues/4727#issuecomment-451741078) or better the complete issue.
> Short story: The SHP6 is pitty. Needs a **perfect!!** calibration or you have to change the measuring resistor.
> Calibration has to be (and can only be) done for this device with [method 2](https://github.com/arendst/Tasmota/wiki/Power-Monitoring-Calibration#method-2)
> You need 2 DMM to measure Voltage and Current at the SAME time

I managed to re-do the calibration with the method 2 setup. Using both the 2000 W heater and the 60 W incandescent bulb, again the results seems to be quite close to the ones achieved with the method 1. 

The issue remains for low loads, the reactive power is quite high if calibrated with a high load (as the 2000W heater). On the opposite site, calibrating with the 60 W bulb, the 2000W heater is way off the chart, specially the reactive power measured (almost a power factor of 0.5!).

tiagofreire-pt, 2019-10-31:
The issue seems to be related to a single point calibration, instead of a feature for fitting the non-linearity with multiple points (3, e.g.).

Jason2866, 2019-10-31:
SHP 6 with no load connected
[image]

tiagofreire-pt, 2019-10-31:
What parameters have you right now for power, current and voltage?

tiagofreire-pt, 2019-10-31:
After method 2 calibration, with a 60W bulb ON:

```
11:00:52 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:00:52","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":13,"ApparentPower":60,"ReactivePower":58,"Factor":0.22,"Voltage":228,"Current":0.262}}
11:00:55 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:00:55","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":24,"ApparentPower":59,"ReactivePower":54,"Factor":0.41,"Voltage":228,"Current":0.257}}
11:01:02 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:02","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":52,"ApparentPower":52,"ReactivePower":0,"Factor":1.00,"Voltage":228,"Current":0.226}}
11:01:04 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:04","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":44,"ApparentPower":44,"ReactivePower":8,"Factor":0.98,"Voltage":228,"Current":0.194}}
11:01:07 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:07","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":25,"ApparentPower":59,"ReactivePower":54,"Factor":0.42,"Voltage":228,"Current":0.260}}
11:01:10 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:10","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":56,"ApparentPower":56,"ReactivePower":0,"Factor":1.00,"Voltage":228,"Current":0.245}}
11:01:12 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:12","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":39,"ApparentPower":56,"ReactivePower":40,"Factor":0.70,"Voltage":228,"Current":0.245}}
11:01:15 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:15","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":26,"ApparentPower":60,"ReactivePower":54,"Factor":0.44,"Voltage":228,"Current":0.262}}
11:01:18 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:18","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":50,"ApparentPower":50,"ReactivePower":0,"Factor":1.00,"Voltage":228,"Current":0.219}}
11:01:19 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:19","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":58,"ApparentPower":58,"ReactivePower":0,"Factor":1.00,"Voltage":228,"Current":0.254}}
11:01:20 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:20","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":58,"ApparentPower":58,"ReactivePower":0,"Factor":1.00,"Voltage":228,"Current":0.254}}
11:01:22 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:22","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":43,"ApparentPower":51,"ReactivePower":27,"Factor":0.84,"Voltage":228,"Current":0.224}}
11:01:25 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:25","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":20,"ApparentPower":49,"ReactivePower":44,"Factor":0.41,"Voltage":228,"Current":0.213}}
11:01:30 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:30","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":56,"ApparentPower":56,"ReactivePower":0,"Factor":1.00,"Voltage":228,"Current":0.247}}
11:01:31 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:31","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":43,"ApparentPower":56,"ReactivePower":36,"Factor":0.77,"Voltage":228,"Current":0.245}}
11:01:35 MQT: tele/blitzwolf_shp6_1/SENSOR = {"Time":"2019-10-31T11:01:35","ENERGY":{"TotalStartTime":"2019-10-30T23:42:34","Total":0.168,"Yesterday":0.000,"Today":0.168,"Power":19,"ApparentPower":57,"ReactivePower":54,"Factor":0.33,"Voltage":228,"Current":0.249}}
```

The reactive power is waving all over the place...

tiagofreire-pt, 2019-10-31:
And, for some weird reason, the PowerCal command is no working on 7.0.0.1:

```
11:05:26 CMD: PowerCal 320000
11:05:26 MQT: stat/blitzwolf_shp6_1/RESULT = {"PowerCal":321419}
```

Jason2866, 2019-10-31:
Radiator
[image]
Measured with voltmeter 226V and 6 A
power, current and voltage values will differ and using mine will not help...
```
12:06:12 CMD: powercal
12:06:12 MQT: stat/sonoff-8C2A9696/RESULT = {"PowerCal":12827}
12:06:21 CMD: currentcal
12:06:21 MQT: stat/sonoff-8C2A9696/RESULT = {"CurrentCal":2323}
12:06:26 CMD: voltagecal
12:06:26 MQT: stat/sonoff-8C2A9696/RESULT = {"VoltageCal":1930}
```

Jason2866, 2019-10-31:
I remember it took me long to get it calibrated....
Maybe it is easier (or needed for your device?) to change the measurement resistor to a higher value.
It is a damn small value....

tiagofreire-pt, 2019-10-31:
Although the voltage, current and active power readings are not that far away, the **reactive** power is. The reactive power is proportional not so much to current and voltage, but to their phase difference.

@arendst, could you help explaining how reactive power is estimated

arendst, 2019-10-31:
The SHP6 uses the ``xnrg_01_hlw8012`` driver. The hardware provides voltage, current and active power.

Formulas in ``xdrv_03_energy`` calculate reactive power based on parameters provided by the hlw8012 driver.

Now if you use command ``wattres 2`` to get som ghosts wattage you'll see where the reactive power comes from.

tiagofreire-pt, 2019-10-31:
@ascillato + @arendst, there is defenetly a non linearity on this plug, as for a 60W bulb I get a currentcal value of 2500 and the oil heater of 3097, for reasonably accurate active power readings.

There is any way to add a template feature for best fitting the calibration for each device characteristics? Like a currentcal_low and currentcal_high?

tiagofreire-pt, 2019-10-31:
> 
> Now if you use command `wattres 2` to get som ghosts wattage you'll see where the reactive power comes from.

It seems that come from a high current reading of nearly 130 mAr. That is weird, maybe some new board revision? I recall from the flashing steps the board has some extra hardware for better voltage insulation and some other version written, different from the images in the wiki.

tiagofreire-pt, 2019-10-31:
This syntax is correct, please?

```
Rule1
  ON ENERGY#Power>2500 DO Backlog CurrentCal 3402 BREAK
  ON 2000>ENERGY#Power<=2500 DO Backlog CurrentCal 3249 BREAK
  ON 1500>ENERGY#Power<=2000 DO Backlog CurrentCal 3095 BREAK 
  ON 1000>ENERGY#Power<=1500 DO Backlog CurrentCal 2942 BREAK 
  ON 500>ENERGY#Power<=1000 DO Backlog CurrentCal 2788 BREAK
  ON 100>ENERGY#Power<=500 DO Backlog CurrentCal 2635 BREAK 
  ON ENERGY#Power<=100 DO Backlog CurrentCal 2500 ENDON
Rule1 1
```

ascillato2, 2019-10-31:
Have you tried it? Should work. Anyway, could be simplier as `backlog` is not needed (`backlog` is used when executing more than 1 command) and the last line is not needed to be `BREAK` and can be a normal `ENDON`

```lua
Rule1
  ON energy#power<=100 DO Currentcal 2500 BREAK 
  ON energy#power<=500 DO Currentcal 2635 BREAK 
  ON energy#power<=1000 DO Currentcal 2788 BREAK 
  ON energy#power<=1500 DO Currentcal 2942 BREAK 
  ON energy#power<=2000 DO Currentcal 3095 BREAK 
  ON energy#power<=2500 DO Currentcal 3249 BREAK 
  ON energy#power>2500 DO Currentcal 3402 ENDON
```

All together for pasting it into the console:

```lua
Rule1 1 
Rule1 ON energy#power<=100 DO Currentcal 2500 BREAK ON energy#power<=500 DO Currentcal 2635 BREAK ON energy#power<=1000 DO Currentcal 2788 BREAK ON energy#power<=1500 DO Currentcal 2942 BREAK ON energy#power<=2000 DO Currentcal 3095 BREAK ON energy#power<=2500 DO Currentcal 3249 BREAK ON energy#power>2500 DO Currentcal 3402 ENDON
```

ascillato2, 2019-10-31:
Closing this issue as you have made a workaround for the nonlinearity of your own device.

closed 2019-10-31 by ascillato2