arendst/Tasmota #6801: New Device: Blitzwolf BW-LT20

sgofferj, 2019-10-31:
Nice and bright cold/warm LED ceiling lamp .

Link: https://www.banggood.com/Blitzwolf-BW-LT20-24W-Smart-LED-Ceiling-Light-Mount-WiFi-APP-Control-Work-with-Amazon-Echo-Google-Home-AC100-240V-p-1508691.html

Based on Tuya TYWE3S:
[image]

Flashing with https://github.com/ct-Open-Source/tuya-convert works without problems.

Here's the template I figured out:
{"NAME":"Blitzwolf BW-LT20","GPIO":[255,255,255,255,255,37,255,255,255,38,255,255,255],"FLAG":15,"BASE":18}

meingraham, 2019-10-31:
@sgofferj Please contribute this template to the [template repository](https://blakadder.github.io/templates/) if you have not already. Thanks.

ascillato2, 2019-10-31:
@sgofferj 

Thanks for sharing. :+1:

closed 2019-10-31 by ascillato2

sgofferj, 2019-10-31:
@meingraham Done

stofakiller, 2020-02-11:
I tried the settings but, i can only get cold white light... Can you share your info from within the config in the blitzwolf?

Regards Jan

sgofferj, 2020-02-11:
[image]

sgofferj, 2020-02-11:
@stofakiller 
Maybe you compare the connection of the LED ring to the PCB with my photo. Could be that there's different versions with the LEDs connected to different GPIOs.

stofakiller, 2020-02-11:
It worked, thanks :-)

finipini, 2020-03-30:
Have any one backed up the original firmware? I tried to flash it with an ftdi ( tuya convert seems dificult to me ) and now it doesnt do nothing.

Forget to back it up my self. aprreciate your help.

jobsidi, 2020-07-25:
I used espeasy before and dont know tasmota well. I flashed the same lamp with tuya convert well. 
How to control the lamp? What gpio is used for on\off relay? I tried to use "Power" command according to https://tasmota.github.io/docs/Commands/#channel but got error in log.
Channel1 with SetOption68 1 also fails. Any recomendations?
Ty in advance.

jobsidi, 2020-07-26:
Im new with tasmota.
Is there on\off command? 
I tried any variations of PowerX but got nothing. SetOption68 is 1
Only when Channel1, Channel2 are set to 0 then lamp is off.
Also how to make presets like, for example, ch1 80+ch2 30, ch1 10+ch2 50 and send it via 1 command? I still learn https://tasmota.github.io/docs/Commands/
Please, give me some advices. ty

ascillato, 2020-07-26:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

DanielG80, 2021-02-02:
what happens when you turn off the Lamp??
When I turn off the lamp there still light some LEDS (dark, but they on)

pegelf, 2022-11-18:
> what happens when you turn off the Lamp?? When I turn off the lamp there still light some LEDS (dark, but they on)

@DanielG80 Were you able to fix the problem? Mine also doesn't turn off completely when I toggle it. Instead, it is the same as setting the brightness to the lowest value. So the lamp still glows a little bit, which can be very annoying in sleeping rooms.

DanielG80, 2022-11-20:
I don't find a resolution for this Problem. If I want it completely off I must go to the switch on the wall. :-(

Sloneczna17, 2023-02-15:
> Have any one backed up the original firmware? I tried to flash it with an ftdi ( tuya convert seems dificult to me ) and now it doesnt do nothing.
> 
> Forget to back it up my self. aprreciate your help.

Do you have a orginal firmware for bw-lt20?

pegelf, 2023-02-15:
I think I've made a backup of the lamp before erasing the flash. The .bin file inside this zip should work:
[Blitzwolf_LT_20.zip](https://github.com/arendst/Tasmota/files/10750559/Blitzwolf_LT_20.zip)

I did that in April 2021.

sfromis, 2023-02-15:
There is a pretty high risk that a binary backup from one device will not work on another, as it is very common for the firmware to be coded for the MAC of the individual ESP chip.

finipini, 2023-02-16:
> There is a pretty high risk that a binary backup from one device will not work on another, as it is very common for the firmware to be coded for the MAC of the individual ESP chip. This very much includes devices originally sold with Tuya firmware.

Nice you anwser almost 3 years later 😂.

I have it now on my balcony working. Back then the problem was that i had shorten One pin from the PCB with gnd. I only discovered that some months later.