arendst/Tasmota #6802: Device H801 channels CW and WW cannot be used independently

MarcelBTTF, 2019-10-31:
### ISSUE DESCRIPTION - TROUBLESHOOTING
The channels for CW and WW cannot be influenced 100% independently of each other. If, for example, channel CW is set to 80 and channel WW switches between the values 10 and 160, the brightness of channel CW also becomes slightly brighter and darker. 

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): H801
- [ ] Tasmota binary firmware version number used: v6.7.1
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: FlashESP8266

### TO REPRODUCE
  RGBWWTable 255,255,255,255,255
  Color 255,255,255,80,10
  Color 255,255,255,80,160
  Color 255,255,255,80,10
  Color 255,255,255,80,160
  ...

### EXPECTED BEHAVIOUR
  In the version Sonoff-Tasmota v6.6.1 the same issue.
  In the version Sonoff-Tasmota v6.2.1 it works perfectly!

s-hadinger, 2019-10-31:
It may surprise you but it is the expected behavior, due to Gamma Correction.

You can always remove Gamma correction with `LedTable 0`.

With some additional logs:
`Color 255,255,255,80,10`
actually drives the following PWM values:
`R255 G255 B255 C12 W2`

`Color 255,255,255,80,160`
actually drives the following PWM values:
`R255 G255 B255 C72 W144`

The second command actually increases the power to CW as well.

**Explanation**
Normal use would be to set `CT` (white temperature) first and then adjust the `Dimmer` for the white channel. The gamma correction must be calculated on the total white power (CW+WW) and then split between CW/WW based on the CT value. This is what drives the most natural behavior when you move the Dimmer.

MarcelBTTF, 2019-10-31:
Thank you very much for the quick feedback and the very good explanation.

closed 2019-10-31 by MarcelBTTF