arendst/Tasmota #6803: Variable Substitution in Rules is Case Sensitive (Contrary to the Documentation)

mphilipp, 2019-10-31:
### BUG DESCRIPTION
Hello,

the [rules documentation](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#rule-syntax) states "A rule is not case sensitive." but it turns out that there is one exception to this.

The variable substitution in the function `RuleSetProcess()` in xdrv_10_rules.ino is implemented case sensitive and only recognizes the lower case names like `%var1%` and `%value%`. Hence, using `%Var1%` or `%VALUE%` to retrieve variable values in rules does not yield the expected behavior.

The [documentation on Rule Variables](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#rule-variables) even lists a wrong example: 

- Set Var4 to Var2's value - `ON Event#temp DO Var4 %Var2% ENDON`

This rule actually sets Var4 to the string "%Var2%", which is not the intended effect.

It took me some time to figure out that this caused my rules to not work as expected and I would like to spare others the same experience. Either the documentation should be changed to correctly reflect the implementation or vice-versa.

Personally I would favor to change the implementation because, in my view, the current behavior is inconsistent. It does not make sense, that the command to set a variable is not case sensitive, i.e. the command `VAR1 <value>` can be used to set var1 but it is not possible to use `%VAR1%` to read the value. But maybe there is a good reason for the current implementation. In either case, I'd be happy to help fixing this.

Regards,
Matthias

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 1
- [x] Tasmota binary firmware version number used: tasmota-basic.bin 23cef04
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
  RSL: stat/tasmota/RESULT = {"Module":{"46":"Shelly 1"}}
  RSL: stat/tasmota/RESULT = {"GPIO":"Not supported"}
  ```
- [x] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  RSL: stat/tasmota/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":477,"Rules":"ON Event#temp DO Var4 %Var2% ENDON"}
  RSL: stat/tasmota/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
  RSL: stat/tasmota/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
  ```
- [x] Provide the output of this command: ``Status 0``:
  ```
  RSL: stat/tasmota/STATUS = {"Status":{"Module":46,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
  RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:03","StartupUTC":"2019-10-31T17:14:45","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":10,"SaveAddress":"FA000"}}
  RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"7.0.0.1(23cef04-basic)","BuildDateTime":"2019-10-31T15:00:57","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
  RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Phiro",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
  RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":457,"Free":544,"Heap":29,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8F082383","041083A0","20021706","010001C0","00000000"],"Drivers":"1,2,3,4,9,10,16,20,21","Sensors":"3"}}
  RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7046","IPAddress":"192.168.178.50","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"DC:4F:22:77:1B:86","Webserver":2,"WifiConfig":4}}
  RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_771B86","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
  RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 31 17:28:48 2019","Local":"Thu Oct 31 18:28:48 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:35","Sunset":"17:32"}}
  RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-10-31T18:28:48","Switch1":"OFF"}}
  RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-10-31T18:28:48","Uptime":"0T00:14:03","UptimeSec":843,"Vcc":3.222,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXX","BSSId":"44:4E:6D:XX:XX:XX","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}
  ```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  See below

### TO REPRODUCE

  ```
CMD: Rule1 ON Event#temp DO Var4 %Var2% ENDON
RSL: stat/tasmota/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":477,"Rules":"ON Event#temp DO Var4 %Var2% ENDON"}
CMD: rule1 1
RSL: stat/tasmota/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":477,"Rules":"ON Event#temp DO Var4 %Var2% ENDON"}
CMD: var2 foo
RSL: stat/tasmota/RESULT = {"Var2":"foo"}
CMD: var2
RSL: stat/tasmota/RESULT = {"Var2":"foo"}
CMD: var4 bar
RSL: stat/tasmota/RESULT = {"Var4":"bar"}
CMD: var4
RSL: stat/tasmota/RESULT = {"Var4":"bar"}
CMD: event temp
RSL: stat/tasmota/RESULT = {"Event":"Done"}
RUL: EVENT#TEMP performs "Var4 %Var2%"
RSL: stat/tasmota/RESULT = {"Var4":"%Var2%"}
CMD: var4
RSL: stat/tasmota/RESULT = {"Var4":"%Var2%"}
  ```

### EXPECTED BEHAVIOUR

After the event, `Var4` should contain the value of `Var2`, i.e.

  ```
RUL: EVENT#TEMP performs "Var4 foo"
RSL: stat/tasmota/RESULT = {"Var4":"foo"}
CMD: var4
RSL: stat/tasmota/RESULT = {"Var4":"foo"}
  ```

ascillato2, 2019-10-31:
Confirmated bug. Thanks for reporting. It shouldn't be case sensitive.

ascillato2, 2019-11-01:
Closing this issue as it had been fixed.

Thanks :+1:

closed 2019-11-01 by ascillato2

mphilipp, 2019-11-02:
Wow. That was fast!
Thank you very much!