arendst/Tasmota #6805: Ability to Incorporate into SmartThings

ncsufire, 2019-11-01:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**  Yes

**Is your feature request related to a problem? Please describe.**  No

**Describe the solution you'd like**  
There is a device handler and Smart App that has been created (with a different .bin file) by erocm1231 to bring a flashed Sonoff device into SmartThings.  I would like to see this program be able to work with erocm1231's code.  https://github.com/arendst/Tasmota/issues?utf8=%E2%9C%93&q=

**Describe alternatives you've considered**  
Installing erocm1231's software and using the firmware with the application; it does not incorporate the SI7021 sensor. 

**Additional context**  
Thank you

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-11-01:
Tasmotas main goal is to be used with standard mqtt brokers.
Including support for more probitary solutions will reduce functions available in Tasmota.
A PR is welcome and maybe! is considered to include if code size increase is minimal.
There will be probably no driver programmed from the developer....
Since Tasmota is Open Source feel free to fork and make the adaptions you want.

andrethomas, 2019-11-01:
The simplest would be to request erocm1231 to include support for the Si7021 sensor in his binary.

ncsufire, 2019-11-01:
Thanks for your reply.  I have done so but thought with the Hue and the
other options this might be a good enhancement.

Jason2866, 2019-11-01:
Tasmota already has Alexa Hue support

ncsufire, 2019-11-01:
Correct.  My hope was for those who wanted another option.  Thank you for
the consideration.

closed 2019-11-01 by ascillato2