arendst/Tasmota #6808: Support for Honeywell humidity sensors (HIHxxxx)

thepotoo, 2019-11-01:
**Have you looked for this feature in other issues and in the wiki?**  
Yes
**Is your feature request related to a problem? Please describe.**  
The current go-to humidity sensor for DIYers is the DHT22, which performs very poorly in high humidity conditions (like a bathroom).  I am looking for a good, reliable sensor that will perform in damp/very humid environments.

The HIH7121 works over a range of 0-100% humidity, and includes a hydrophobic filter that prevents condensation, which on paper seems like it would be ideal for my application (bathroom fan control).

**Describe the solution you'd like**  
Add support for Honeywell humidity sensors on an I2C interface.  It looks like some code exists here that may help: https://github.com/ControlEverythingCommunity/HIH6130/blob/master/Arduino/HIH6130.ino

As far as I can tell, all of these HIHxxxx sensors use the same address (0x27) and protocol, so adding support for any of them should add support for all of them.

I don't see a particular need for command mode (changing address, setting alarms).

**Describe alternatives you've considered**  
I've spent quite a bit of time looking at different humidity sensors and these seem a bit better than the ones Tasmota currently supports.

**Additional context**  
Here is a datasheet: https://www.mouser.com/datasheet/2/187/honeywell-sensing-humidicon-hih7000-series-product-1140774.pdf

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2019-11-01:
Have you tried the Si7021?

I can't find any breakout/module boards for the HIH7121 so it may not be likely for someone to add support for it at this time.

thepotoo, 2019-11-01:
From the datasheet for the Si7021:

> Recommended humidity operating range is 20% to 80% RH (non-condensing) over –10 °C to 60 °C. Prolonged operation beyond these ranges may result in a shift of sensor reading, with slow recovery time.

This suggests that it won't meet my needs (area with 80-95% humidity).

The HIH7121 doesn't need a breakout board, just a 22uF cap.  I have ordered some and will play around with them when they get here.

andrethomas, 2019-11-01:
Sure, if you get it working you're welcome to make a PR :)

arendst, 2019-11-02:
Give it a try and report back please.

thepotoo, 2019-11-02:
@arendst Thank you!  I can't believe how quickly you added this.  I will test it when the sensors arrive (in a week or so) and let you know if there are any issues.

Will this be available in the default sensors build?

andrethomas, 2019-11-02:
> Will this be available in the default sensors build?

It's not included at the moment - The driver is very small though so I'm guessing it will be as soon as you're able to test it and provide feedback :)

arendst, 2019-11-02:
It probably will not as it has the same address as PCF82xx

ascillato2, 2019-11-06:
@thepotoo

Please, after you perform the tests with this sensor, tell us how it works. Thanks.

thepotoo, 2019-11-07:
@arendst Sensors arrived today.  The original sketch I linked to works fine, but for some reason in Tasmota I am getting:
`20:02:17 MQT: tele/sonoff/SENSOR = {"Time":"2019-11-06T20:02:17","HIH":{"Temperature":256.982,"Humidity":100.0},"TempUnit":"F"}`

The temp is always 256.982 and the humidity is always 100; neither ever change.  I notice `delay(300);`
in the example sketch, which is there to prevent the module from going into command mode - could this be happening with Tasmota?

The module is detected fine and shows up correctly, so that is great.

Any suggestions here?

thepotoo, 2019-11-07:
[This](https://www.phanderson.com/arduino/hih6130.html) sketch also works fine, and has some extra diagnostic stuff in it.

arendst, 2019-11-07:
Thx for testing.

Pls report back with the latest fix.

thepotoo, 2019-11-07:
@arendst Latest changes do NOT work - same problem with the sensor readings maxed out.

However, when I uncomment the line:
` delay(40);                                          // Wait for Valid data or use Stale data`
it works fine.

Thank you very much for putting this together.  You made my week!

arendst, 2019-11-07:
OK thx for testing. I hate to add delays but as this is within my limits (;-)) I'll insert it.

Will release soon.

arendst, 2019-11-07:
Final test please.

thepotoo, 2019-11-07:
Gotta go to work, will test tonight and close this if everything works.

I would love for there to be some option to toggle between these and the PCF82xx so I don't have to make my own build.  But I understand if that's asking too much :)

Again thank you so much for this!

arendst, 2019-11-07:
The toggle is already in place but probably not documented yet.

Use command ``I2cDriver2 0`` to disable PCF8574 support. It will restart and should pick up your Honeywell sensor. Unless you have a LCD display also configured. Then you will have to use command ``I2CDriver3 0`` too.

See document I2CDEVICES.md in the root what ``I2cDriver`` indexes you can use to disable as there could be more before the HIH sensor is reached within the I2C startup sequence.

thepotoo, 2019-11-07:
Works like a charm!

It does not seem to be enabled in the standard tasmota-senors build; `i2cdriver2 0` does not seem to have any effect.  Obviously once I enable `#define USE_HIH6` everything works.

I am still experimenting with these sensors, and will try to update the wiki with how things go.  So far it does not seem like the two 2.2k pullup resistors shown in the wiring diagram are needed.

thepotoo, 2019-11-08:
I wrote this up in the wiki [here](https://github.com/arendst/Tasmota/wiki/Honeywell-HIHxxxx-humidity-sensors).  I deployed 8 of these yesterday in various wet areas and so far there have been no issues at all.

closed 2019-11-08 by thepotoo

andrethomas, 2019-11-08:
@thepotoo Good to hear & thanks for the feedback and writing up the wiki article - I think @meingraham will most likely integrate it into the rest of the wiki :)