arendst/Tasmota #6820: Console -> WebPassword causes imposible login loop

jj-uk, 2019-11-02:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
When using the console and WebPassword to CHANGE an existing password, a login prompt pops up for 2 secs, then disappears and immediately pops up again. This is repeated over and over.
2 seconds is not enough time to enter the password.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Lingan SWA1
- [ ] Tasmota binary firmware version number used:6.6.0
  - [ Y] Pre-compiled
  - [ N] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: tuya convert
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
22:42:19 MQT: stat/bens_disco_light/RESULT = {"NAME":"Lingan SWA1","GPIO":[0,0,0,0,158,21,0,0,0,17,0,0,0],"FLAG":0,"BASE":18}
22:42:19 MQT: stat/bens_disco_light/RESULT = {"Module":"0 (Lingan SWA1)"}
22:42:20 MQT: stat/bens_disco_light/RESULT = {"GPIO":"Not supported"}

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
contains personal information.

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
cant, in a password loop.

  ```

### TO REPRODUCE
On a fresh module, set a web password
Log in
Go to console and change the password with WebPassword XXXXX

### EXPECTED BEHAVIOUR
A password prompt appears and accepts a password.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-11-02:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.
 
Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).
 
Thanks.
 
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

jj-uk, 2019-11-02:
I did complete most of the template. The part i left out contains information that is personal.

ascillato, 2019-11-02:
Status 0 is crucial information. You can delete the parts you don't want to share but the rest is very important.

jj-uk, 2019-11-02:
23:55:06 CMD: status 0
23:55:06 MQT: stat/bens_disco_light/STATUS = {"Status":{"Module":0,"FriendlyName":["Bens disco light"],"Topic":"bens_disco_light","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:55:06 MQT: stat/bens_disco_light/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:19","StartupUTC":"2019-11-02T22:54:47","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":7,"SaveAddress":"F5000"}}
23:55:06 MQT: stat/bens_disco_light/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:55:06 MQT: stat/bens_disco_light/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["REMOVED",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["0000A009","140500000100060000000000000000000000","00000000"]}}
23:55:06 MQT: stat/bens_disco_light/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
23:55:06 MQT: stat/bens_disco_light/STATUS5 = {"StatusNET":{"Hostname":"bens_disco_light","IPAddress":"REMOVED","Gateway":"REMOVED","Subnetmask":"REMOVED","DNSServer":"REMOVED","Mac":"REMOVED","Webserver":2,"WifiConfig":4}}
23:55:06 MQT: stat/bens_disco_light/STATUS6 = {"StatusMQT":{"MqttHost":"REMOVED","MqttPort":1883,"MqttClientMask":"bens_disco_light","MqttClient":"bens_disco_light","MqttUser":"REMOVED","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:55:06 MQT: stat/bens_disco_light/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 02 22:55:06 2019","Local":"Sat Nov 02 23:55:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:38","Sunset":"17:29"}}
23:55:06 MQT: stat/bens_disco_light/STATUS10 = {"StatusSNS":{"Time":"2019-11-02T23:55:06"}}
23:55:06 MQT: stat/bens_disco_light/STATUS11 = {"StatusSTS":{"Time":"2019-11-02T23:55:06","Uptime":"0T00:00:19","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"POWER":"OFF","Wifi":{"AP":1,"SSId":"j13","BSSId":"38:D5:47:1E:C8:C0","Channel":6,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:06"}}}

jj-uk, 2019-11-02:
23:55:06 CMD: status 0
23:55:06 MQT: stat/bens_disco_light/STATUS = {"Status":{"Module":0,"FriendlyName":["Bens disco light"],"Topic":"bens_disco_light","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:55:06 MQT: stat/bens_disco_light/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:19","StartupUTC":"2019-11-02T22:54:47","Sleep":50,"CfgHolder":4617,"BootCount":3,"SaveCount":7,"SaveAddress":"F5000"}}
23:55:06 MQT: stat/bens_disco_light/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
23:55:06 MQT: stat/bens_disco_light/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["REMOVED",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["0000A009","140500000100060000000000000000000000","00000000"]}}
23:55:06 MQT: stat/bens_disco_light/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
23:55:06 MQT: stat/bens_disco_light/STATUS5 = {"StatusNET":{"Hostname":"bens_disco_light","IPAddress":"REMOVED","Gateway":"REMOVED","Subnetmask":"REMOVED","DNSServer":"REMOVED","Mac":"REMOVED","Webserver":2,"WifiConfig":4}}
23:55:06 MQT: stat/bens_disco_light/STATUS6 = {"StatusMQT":{"MqttHost":"REMOVED","MqttPort":1883,"MqttClientMask":"bens_disco_light","MqttClient":"bens_disco_light","MqttUser":"REMOVED","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:55:06 MQT: stat/bens_disco_light/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 02 22:55:06 2019","Local":"Sat Nov 02 23:55:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:38","Sunset":"17:29"}}
23:55:06 MQT: stat/bens_disco_light/STATUS10 = {"StatusSNS":{"Time":"2019-11-02T23:55:06"}}
23:55:06 MQT: stat/bens_disco_light/STATUS11 = {"StatusSTS":{"Time":"2019-11-02T23:55:06","Uptime":"0T00:00:19","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"POWER":"OFF","Wifi":{"AP":1,"SSId":"REMOVED","BSSId":"REMOVED","Channel":6,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:06"}}}

ascillato, 2019-11-02:
Thanks for sharing this information.

Version 6.6.0 is very old and also you have it with core 2.3.0 that is also very old and has been dropped due to security issues.

So, please, update to latest Tasmota 7.0.2 that have a huge amount of fixes and besides it uses a newer and secure core.

If you need further assistant on updating, please use the Tasmota support chat.

Thanks

jj-uk, 2019-11-02:
Just updated to latest using the built in OTA updater (7.6.2). The same issue occurs when using WebPassword from console.

AIexBV, 2019-11-03:
I have had the same issue many times. 

Try the following approaches:
- If you use Chrome press CTRL & F5
- If that does not help, delete the cache and restart the browser
- If that does not help use an alternative browser, log in and go back to your usual browser

Hope that helps.

ascillato2, 2019-11-04:
@jj-uk 

Hi,

Any news on this? Have you tried the approach from @AIexBV ?

jj-uk, 2019-11-04:
Once the login box appears, it will continue to disappear and reappear over and over.

Yes, refreshing the page will fix this.

ascillato, 2019-11-04:
> Yes, refreshing the page will fix this.

ok, so that means that your issue is solved?

jj-uk, 2019-11-04:
Not really, it needs fixing in the code.

ascillato2, 2019-11-04:
> Not really, it needs fixing in the code.

I can't reproduce your issue. Sorry. What do you think needs to be modified in order to make it work in your setup?

jj-uk, 2019-11-04:
I'm not sure why it happens... Leave it with me for a day or two, i'll try to figure out why it happens.

jj-uk, 2019-11-04:
I've been messing with this and it looks very much like a bug in Google Chrome (v78).
Refreshing the page fixes it. MS Edge is ok. Have not tried firefox as i don't have it.

I guess this should be marked as cant fix, and closed.

closed 2019-11-05 by ascillato2