arendst/Tasmota #6821: Topic command no longer works

jj-uk, 2019-11-02:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If you have a issue when flashing was done via Tuya Convert
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
In console, on a newly reset device, running backlog with a topic' no longer sets the topic. 
This works perfectly in 6.6.0.
After the backlog command has run and restarted, Topic cannot change the topic from console.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Lingan SWA1
- [ x] Tasmota binary firmware version number used: _____
  - [y] Pre-compiled
  - [n ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: FTDI
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

00:00:35 CMD: gpio
00:00:35 RSL: stat/sonoff/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

no rules
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

23:32:05 CMD: STATUS 0
23:32:05 MQT: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Front room lamp"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:32:05 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:41","StartupUTC":"2019-11-02T22:30:24","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":17,"SaveAddress":"FB000"}}
23:32:05 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":4,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
23:32:05 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["REMOVED",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["0000A009","1405C8000100060000005A00000000000000","00000000"]}}
23:32:05 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440A1","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
23:32:05 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"front_room_lamp","IPAddress":"REMOVED","Gateway":"REMOVED","Subnetmask":"REMOVED","DNSServer":"REMOVED","Mac":"REMOVED","Webserver":2,"WifiConfig":4}}
23:32:05 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"REMOVED","MqttPort":1883,"MqttClientMask":"front_room_lamp","MqttClient":"front_room_lamp","MqttUser":"REMOVED","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:32:05 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Nov 02 22:32:05 2019","Local":"Sat Nov 02 23:32:05 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:38","Sunset":"17:29"}}
23:32:05 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-11-02T23:32:05"}}
23:32:05 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-11-02T23:32:05","Uptime":"0T00:01:41","UptimeSec":101,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"j13","BSSId":"38:D5:47:1E:C8:C0","Channel":6,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:05"}}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

23:36:41 MQT: stat/sonoff/RESULT = {"WebLog":4}
23:36:41 CFG: Saved to flash at FA, Count 18, Bytes 4096

  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I use backlog to set up new devices. The topic is no longer accepted in the 6.7.1 version of tasmota.

Backlog Template {"NAME":"Lingan SWA1","PIO":[0,0,0,0,158,21,0,0,0,17,0,0,0],"FLAG":0,"BASE":18}; Module 0; WebPassword REMOVED; SSID1 REMOVED; Password1 REMOVED; MqttHost REMOVED; MqttUser REMOVED; MqttPassword REMOVED; Hostname front_room_lamp; MqttClient front_room_lamp; Topic front_room_lamp; FriendlyName1 Front room lamp; SetOption13 1; SetOption32 20; TelePeriod 10;

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Topic and all other settings to be set.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jj-uk, 2019-11-02:
And i cant revert back to 6.6.0 via OTA.

Lingan SWA1 Module
Front room lamp
Upload Failed

Upload buffer miscompare

Main Menu
Sonoff-Tasmota 6.7.1 by Theo Arends

jj-uk, 2019-11-02:
I've reproduced this on v6.6.0 too.
Reset a device, run the backlog command as above.
Then (after the reset) in console try to set the topic.

jj-uk, 2019-11-02:
further messing with this. after a reset and the backlog, i can set the topic to anything EXCEPT front_room_lamp

localhost61, 2019-11-03:
Your backlog command is 374 bytes long, to be adjusted with actual REMOVED fields length. You may have got an overflow !!?

rominator1983, 2019-11-03:
Hi,
You are writing "And i cant revert back to 6.6.0 via file upload to 6.6.0.".
Suppose you ment "cannot" ?
Reverting to a previous version without flashing will not be possible on the sonoff-devices I know. Instead you have to go to upload a minimal variant (http://thehackbox.org/tasmota/release/arc/6.6.0/sonoff-minimal.bin) temporary and then upload the desired version (http://thehackbox.org/tasmota/release/arc/6.6.0/sonoff.bin). This is because of the 1M ROM size of the sonoff-devices (at least Basic and S20 I own). Your settings "should" be fine with this procedure. Although I am not sure how Tasmota exactly does this. At least my settings never got overriden when doing so.

localhost61 may be right that your Backlog-command is too long. Does not explain why this was working previously.

So to investigate further: Does the topic-command work when executing the commands separately without backlog?

And a further tipp: I build my own Tasmota variant and do most of the default setup (WIFI, MQTT host etc.) in a config file there (forgot the name but should be easy to find). That way I have minimal setup steps to make each device unique.

jj-uk, 2019-11-04:
I ment " can't ". OK, i'll mess about with the backlog a bit more see if I can figure out at what point the issue occurs.
What's the char limit for backlog? I found '30 consecutive commands' in the docs but I guess these's a buffer limit too.

localhost61, 2019-11-04:
It seems there is no buffer limit in the source code, out of the 30 commands. It implement a LinkedList for that purpose, there is no buffer reservation.

arendst, 2019-11-04:
Using the info from above the topic is changed as expected:
```
10:42:51 CMD: Backlog Template {"NAME":"Lingan SWA1","PIO":[0,0,0,0,158,21,0,0,0,17,0,0,0],"FLAG":0,"BASE":18}; Module 0; WebPassword REMOVED; SSID1 REMOVED; Password1 REMOVED; MqttHost REMOVED; MqttUser REMOVED; MqttPassword REMOVED; Hostname front_room_lamp; MqttClient front_room_lamp; Topic front_room_lamp; FriendlyName1 Front room lamp; SetOption13 1; SetOption32 20; TelePeriod 10;
10:42:51 SRC: WebConsole from 192.168.2.1
10:42:51 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template {"NAME":"Lingan SWA1","PIO":[0,0,0,0,158,21,0,0,0,17,0,0,0],"FLAG":0,"BASE":18}; Module 0; WebPassword REMOVED; SSID1 REMOVED; Password1 REMOVED; MqttHost REMOVED; MqttUser REMOVED; MqttPassword REMOVED; Hostname front_room_lamp; MqttClient front_room_lamp; Topic front_room_lamp; FriendlyName1 Front room lamp; SetOption13 1; SetOption32 20; TelePeriod 10;"
10:42:51 SRC: Backlog
10:42:51 CMD: Group 0, Index 1, Command "TEMPLATE", Data "{"NAME":"Lingan SWA1","PIO":[0,0,0,0,158,21,0,0,0,17,0,0,0],"FLAG":0,"BASE":18}"
10:42:51 MQT: stat/wemos4/TEMPLATE = {"NAME":"Lingan SWA1","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":0,"BASE":18}
10:42:51 SRC: Backlog
10:42:51 CMD: Group 0, Index 1, Command "MODULE", Data "0"
10:42:51 MQT: stat/wemos4/MODULE = {"Module":{"0":"Lingan SWA1"}}
10:42:51 SRC: Backlog
10:42:51 CMD: Group 0, Index 1, Command "WEBPASSWORD", Data "REMOVED"
10:42:51 MQT: stat/wemos4/WEBPASSWORD = {"WebPassword":"REMOVED"}
10:42:52 SRC: Backlog
10:42:52 CMD: Group 0, Index 1, Command "SSID", Data "REMOVED"
10:42:52 MQT: stat/wemos4/SSID = {"SSId1":"REMOVED"}
10:42:52 SRC: Backlog
10:42:52 CMD: Group 0, Index 1, Command "PASSWORD", Data "REMOVED"
10:42:52 MQT: stat/wemos4/PASSWORD = {"Password1":"REMOVED"}
10:42:52 SRC: Backlog
10:42:52 CMD: Group 0, Index 1, Command "MQTTHOST", Data "REMOVED"
10:42:52 MQT: stat/wemos4/MQTTHOST = {"MqttHost":"REMOVED"}
10:42:52 SRC: Backlog
10:42:52 CMD: Group 0, Index 1, Command "MQTTUSER", Data "REMOVED"
10:42:52 MQT: stat/wemos4/MQTTUSER = {"MqttUser":"REMOVED"}
10:42:53 SRC: Backlog
10:42:53 CMD: Group 0, Index 1, Command "MQTTPASSWORD", Data "REMOVED"
10:42:53 MQT: stat/wemos4/MQTTPASSWORD = {"MqttPassword":"REMOVED"}
10:42:53 SRC: Backlog
10:42:53 CMD: Group 0, Index 1, Command "HOSTNAME", Data "front_room_lamp"
10:42:53 MQT: stat/wemos4/HOSTNAME = {"Hostname":"front_room_lamp"}
10:42:53 SRC: Backlog
10:42:53 CMD: Group 0, Index 1, Command "MQTTCLIENT", Data "front_room_lamp"
10:42:53 MQT: stat/wemos4/MQTTCLIENT = {"MqttClient":"front_room_lamp"}
10:42:53 SRC: Backlog
10:42:53 CMD: Group 0, Index 1, Command "TOPIC", Data "front_room_lamp"
10:42:53 MQT: tele/wemos4/LWT =  (retained)
10:42:53 MQT: stat/wemos4/TOPIC = {"Topic":"front_room_lamp"}
10:42:54 SRC: Backlog
10:42:54 CMD: Group 0, Index 1, Command "FRIENDLYNAME", Data "Front room lamp"
10:42:54 MQT: stat/wemos4/FRIENDLYNAME = {"FriendlyName1":"Front room lamp"}
10:42:54 SRC: Backlog
10:42:54 CMD: Group 0, Index 13, Command "SETOPTION", Data "1"
10:42:54 MQT: stat/wemos4/SETOPTION = {"SetOption13":"ON"}
10:42:54 SRC: Backlog
10:42:54 CMD: Group 0, Index 32, Command "SETOPTION", Data "20"
10:42:54 MQT: stat/wemos4/SETOPTION = {"SetOption32":20}
10:42:54 SRC: Backlog
10:42:54 CMD: Group 0, Index 1, Command "TELEPERIOD", Data "10"
10:42:54 MQT: stat/wemos4/TELEPERIOD = {"TelePeriod":10}
10:42:55 CFG: Saved to flash at F7, Count 376, Bytes 4096
10:42:55 MQT: tele/wemos4/STATE = {"Time":"2019-11-04T10:42:55","Uptime":"0T00:02:10","UptimeSec":130,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"REMOVED","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:42:55 MQT: tele/wemos4/SENSOR = {"Time":"2019-11-04T10:42:55","BME680":{"Temperature":23.7,"Humidity":54.5,"Pressure":987.8,"Gas":417.02},"PressureUnit":"hPa","TempUnit":"C"}
10:42:56 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v11c3757b
~ld

00:00:00 CFG: Loaded from flash at F7, Count 376
00:00:00 SRC: Restart
00:00:00 Project tasmota Front room lamp Version 7.0.0.3(tasmota)-STAGE
00:00:00 CFG: Saved to flash at F6, Count 377, Bytes 4096
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP-, SSId indebuurt1, Channel 1, BSSId 24:D3:F2:97:C0:A1, RSSI -74, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt2, Channel 5, BSSId A0:AB:1B:7D:42:AC, RSSI -39, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId indebuurt3, Channel 12, BSSId 60:E3:27:58:77:E6, RSSI -79, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt2 in mode 11N as front_room_lamp...
00:00:04 WIF: Attempting connection...
00:00:05 CFG: Saved to flash at F5, Count 378, Bytes 4096
00:00:05 WIF: Attempting connection...
00:00:06 QPC: Reset
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on front_room_lamp with IP address 192.168.2.223
10:43:04 NTP: Drift 0, (UTC) Mon Nov 04 09:43:04 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
[... cut ...]
10:43:14 RSL: tele/front_room_lamp/STATE = {"Time":"2019-11-04T10:43:14","Uptime":"0T00:00:19","UptimeSec":19,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:43:16 MQT: Attempting connection...
10:43:16 MQT: Connect failed to REMOVED:1883, rc -2. Retry in 10 sec
```
For debugging purposes enable``weblog 3`` or better ``syslog 3``, re-execute your backlog command and verify the output.

jj-uk, 2019-11-04:
Right, I've narrowed it down a little.

The issue seems to occur when the topic is set to the same as the MqttClient.

See here, the mqtt login is VALID. It connects to the MQTT server. The passwords here are real, for testing:

```
I took a D1 mini (v2.2.0) and flashed it with sonoff.bin (6.7.1) using ESPHome-flasher v1.2.

Connect to the AP and enter valid credentials.

Connect to the device via its IP address.

Go to Console - Send Backup Command with VALID credentials:

Backlog MqttHost 192.168.1.230; MqttUser jjj; MqttPassword zzzzz; MqttClient lamp1;

00:00:00 CFG: Loaded from flash at F5, Count 7
00:00:00 Project sonoff Sonoff Version 6.7.1(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 guest2 in mode 11N as sonoff-7280...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-7280 with IP address 192.168.1.178
23:01:44 MQT: Attempting connection...
23:01:44 MQT: Connected
23:01:44 MQT: tele/sonoff/LWT = Online (retained)
23:01:44 MQT: cmnd/sonoff/POWER = 
23:01:44 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.7.1(sonoff)","FallbackTopic":"cmnd/lamp1_fb/","GroupTopic":"sonoffs"}
23:01:44 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-7280","IPAddress":"192.168.1.178"}
23:01:44 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
23:01:44 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
23:01:44 MQT: stat/sonoff/POWER = OFF

topic aaa

00:00:00 CFG: Loaded from flash at FB, Count 9
00:00:00 Project sonoff Sonoff Version 6.7.1(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 guest2 in mode 11N as aaa-7280...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on aaa-7280 with IP address 192.168.1.178
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/aaa/LWT = Online (retained)
00:00:06 MQT: cmnd/aaa/POWER = 
00:00:06 MQT: tele/aaa/INFO1 = {"Module":"Sonoff Basic","Version":"6.7.1(sonoff)","FallbackTopic":"cmnd/lamp1_fb/","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/aaa/INFO2 = {"WebServerMode":"Admin","Hostname":"aaa-7280","IPAddress":"192.168.1.178"}
00:00:06 MQT: tele/aaa/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/aaa/RESULT = {"POWER":"OFF"}
00:00:06 MQT: stat/aaa/POWER = OFF

topic lamp1

00:00:00 CFG: Loaded from flash at F9, Count 11
00:00:00 Project sonoff Sonoff Version 6.7.1(sonoff)-STAGE
00:00:00 WIF: Connecting to AP1 guest2 in mode 11N as sonoff-7280...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-7280 with IP address 192.168.1.178
23:04:39 MQT: Attempting connection...
23:04:39 MQT: Connected
23:04:39 MQT: tele/sonoff/LWT = Online (retained)
23:04:39 MQT: cmnd/sonoff/POWER = 
23:04:39 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.7.1(sonoff)","FallbackTopic":"cmnd/lamp1_fb/","GroupTopic":"sonoffs"}
23:04:39 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-7280","IPAddress":"192.168.1.178"}
23:04:39 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
23:04:39 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
23:04:39 MQT: stat/sonoff/POWER = OFF

```

ascillato2, 2019-11-06:
Hi,

If you check **Line 896** from _xdrv_02_mqtt.ino_ file, you will find that this behaviour is intended.

https://github.com/arendst/Tasmota/blob/bb8d7812907cfda7f7d4c717c249ae33fbf09f53/tasmota/xdrv_02_mqtt.ino#L896

**Mqtt Topic** can't be equal to **Mqtt Client**. Sorry.

closed 2019-11-06 by ascillato2

jj-uk, 2019-11-06:
Are you able to update the docs?

https://github.com/arendst/Tasmota/wiki/Commands#mqtt

ascillato, 2019-11-07:
You are right. Will do. Thanks for pointing that out.

ascillato2, 2019-11-07:
Done