arendst/Tasmota #6822: Shelly 1 - weird buzzing/whistling sound coming from the device

VivantSenior, 2019-11-03:
> **GUIDE**
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>
> Please DO NOT OPEN AN ISSUE:
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Tasmota/wiki/FAQ) and troubleshooting wiki articles
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 1
- [x] Tasmota binary firmware version number used: 6.7.1(sonoff)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [x] Provide the output of this command: ``Backlog Template; Module; GPIO``:

  ```
  Configuration output here:
  08:41:34 MQT: BedroomLampRight/stat/RESULT = {"NAME":"Shelly 1","GPIO":[0,0,0,0,21,82,0,0,0,0,0,0,0],"FLAG":0,"BASE":46}
  08:41:34 MQT: BedroomLampRight/stat/RESULT = {"Module":{"0":"Shelly 1"}}
  08:41:35 MQT: BedroomLampRight/stat/RESULT = {"GPIO":"Not supported"}
  ```

- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  No rules used.
  ```

- [x] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  08:42:46 MQT: BedroomLampRight/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["BedroomRightLamp"],"Topic":"BedroomLampRight","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:02:04","StartupUTC":"2019-11-03T07:40:42","Sleep":50,"CfgHolder":4617,"BootCount":35,"SaveCount":132,"SaveAddress":"F6000"}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Bazinga",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005AFF000000000000","00000200"]}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS4 = {"StatusMEM":{"ProgramSize":564,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","0000F081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,30","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS5 = {"StatusNET":{"Hostname":"Bedroom_right_lamp","IPAddress":"192.168.0.109","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"62.179.1.63","Mac":"A4:CF:12:BA:60:9B","Webserver":2,"WifiConfig":4}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.199","MqttPort":1883,"MqttClientMask":"BedroomLampRight","MqttClient":"BedroomLampRight","MqttUser":"mosquitto","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 03 07:42:46 2019","Local":"Sun Nov 03 08:42:46 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:40","Sunset":"17:27"}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS10 = {"StatusSNS":{"Time":"2019-11-03T08:42:46","Switch1":"OFF"}}
  08:42:46 MQT: BedroomLampRight/stat/STATUS11 = {"StatusSTS":{"Time":"2019-11-03T08:42:46","Uptime":"0T00:02:04","UptimeSec":124,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Bazinga","BSSId":"90:5C:44:03:BA:DF","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}}
  ```

- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  Not a software issue.
  ```

**(Please, remember to close the issue when the problem has been addressed)**

Hi,

I bought Shelly 1 and 1PM and want to use Shelly 1 to make my wall bed lamp "smart". But noticed that Shelly is whistling/buzzing (doesn't matter if it's ON or OFF). I asked the Shelly support for help. Their response is as follows:

``The whistle is coming from AC/DC chip. This chip is managed to deliver from 120mA at 12V, but Shelly1 consumption is 80-100mA.
The AC/DC chip detects under consumption and makes fast sleep cycles which cause induction to whistle. When you switch On the relay, whistle is gone or change because power consumption of the device is increasing.``

I tried to enable and disabled [DynamicSleep](https://github.com/arendst/Sonoff-Tasmota/wiki/Dynamic-Sleep) and also set Sleep to 0 and 250.  The sound is more steady when I set Sleep to 0, otherwise (Sleep 250) the sound has more "beeps" but I can still hear the buzzing. I thought it's a faulty device and check a few other Shelly devices (1 and 1PM) but the result was the same.

Any ideas of how I can configure Tasmota to get rid of the annoying sound?

Best,
Viv

Jason2866, 2019-11-03:
This is hardware related. The power supply does make the noise. 
Tasmota can do nothing from software side.

VivantSenior, 2019-11-03:
"It is not a bug, it is undocumented feature" of Shelly? I checked the decibels, 31 dB when Shelly is connected to main power, 26 dB when it's disconnected. 5 dB difference - it's a loud sound especially if the source is 50 cm from your ears during the night :(

Sonoff Basic doesn't make this sound.

Is there any way to make it quieter/suppress the sound? Changing AC/DC chip?

Jason2866, 2019-11-03:
Ask Shelly ;-) I dont know.

VivantSenior, 2019-11-03:
Shelly support told me that's a normal Shelly 1/1PM working sound...

Closing the ticket.

closed 2019-11-03 by VivantSenior

UBWH, 2021-04-09:
As a comment, this caused me days of lost time diagnosing why my ultrasonic range finder was giving bad readings when connected via a Shelly1 flashed with Tasmota. 

If the Shelly1 was:

- far (300 mm??)  from the range finder => All OK
- near (10mm) to the range finder => erratic range values.

I assumed electrical interference .... but it was sonic.

dresende, 2021-08-05:
If you're using DC, this may be the PSU connected to Shelly vibrating. It's common in Shelly RGBW.