arendst/Tasmota #6825: MLX90614 shows only -999 obj temp

Failx92, 2019-11-03:
### BUG DESCRIPTION
I connect a MLX90614 Sensor but it shows only -999 degree object temp but for the ambient temp it shows the right temp.

I tried serveral sensors and serveral node mcu boards but the issue is the same. I compiled the 6.6 version and the new 6.7 release.

### REQUESTED INFORMATION

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): nodemcu
- [x ] Tasmota binary firmware version number used: 6.6 + 6.7
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [x ] IDE / Compiler used: arduino ide 1.8.10
- [ x] Flashing tools used: espeasyflash
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

13:53:30 RSL: empfangenes topic /Backlog, Datengröße 22, Daten Template; Module; GPIO
13:53:30 RSL: Gruppe 0, Index 1, Befehl BACKLOG, Daten Template; Module; GPIO
13:53:30 SRC: Backlog
13:53:30 RSL: empfangenes topic /Template, Datengröße 0, Daten 
13:53:30 RSL: Gruppe 0, Index 1, Befehl TEMPLATE, Daten 
13:53:30 MQT: stat/nodemcu_v2/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
13:53:31 SRC: Backlog
13:53:31 RSL: empfangenes topic /Module, Datengröße 0, Daten 
13:53:31 RSL: Gruppe 0, Index 1, Befehl MODULE, Daten 
13:53:31 MQT: stat/nodemcu_v2/RESULT = {"Module":"18 (Generic)"}
13:53:31 SRC: Backlog
13:53:31 RSL: empfangenes topic /GPIO, Datengröße 0, Daten 
13:53:31 RSL: Gruppe 0, Index 1, Befehl GPIO, Daten 
13:53:31 MQT: stat/nodemcu_v2/RESULT = {"GPIO0":"5 (I2C SCL)","GPIO1":"0 (None)","GPIO2":"6 (I2C SDA)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"0 (None)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
  ```

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
13:47:16 RSL: Gruppe 0, Index 1, Befehl STATUS, Daten 0
13:47:16 MQT: stat/nodemcu_v2/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"nodemcu_v2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:47:16 MQT: stat/nodemcu_v2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:26:41","StartupUTC":"2019-11-03T12:20:35","Sleep":50,"CfgHolder":4617,"BootCount":26,"SaveCount":42,"SaveAddress":"3F8000"}}
13:47:16 MQT: stat/nodemcu_v2/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019.11.03 11:35:05","Boot":6,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
13:47:16 MQT: stat/nodemcu_v2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["---",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
13:47:16 MQT: stat/nodemcu_v2/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":3508,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashMode":2,"Features":["00000407","0F8AE396","001783A0","22B617CD","01003BC0"]}}
13:47:16 MQT: stat/nodemcu_v2/STATUS5 = {"StatusNET":{"Hostname":"nodemcu_v2-6233","IPAddress":"192.168.1.54","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"---","Mac":"---","Webserver":2,"WifiConfig":4}}
13:47:16 MQT: stat/nodemcu_v2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.250","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_63D859","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:47:16 MQT: stat/nodemcu_v2/STATUS7 = {"StatusTIM":{"UTC":"Sun Nov 03 12:47:16 2019","Local":"Sun Nov 03 13:47:16 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:13","Sunset":"16:55"}}
13:47:16 MQT: stat/nodemcu_v2/STATUS10 = {"StatusSNS":{"Time":"2019-11-03T13:47:16","ANALOG":{"A0":5},"MLX90614":{"OBJTMP":-999.0,"AMBTMP":22.7}}}
13:47:16 MQT: stat/nodemcu_v2/STATUS11 = {"StatusSTS":{"Time":"2019-11-03T13:47:16","Uptime":"0T00:26:41","Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"Wifi":{"AP":1,"SSId":"----","BSSId":"7C:FF:4D:48:EC:66","Channel":7,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:08"}}}
13:47:25 MQT: tele/nodemcu_v2/STATE = {"Time":"2019-11-03T13:47:25","Uptime":"0T00:26:50","Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"-----","BSSId":"7C:FF:4D:48:EC:66","Channel":7,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:08"}}

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
13:50:25 MQT: tele/nodemcu_v2/SENSOR = {"Time":"2019-11-03T13:50:25","ANALOG":{"A0":5},"MLX90614":{"OBJTMP":-999.0,"AMBTMP":22.8}}

  ```

### EXPECTED BEHAVIOUR
Help to get the sensor to work.

ascillato, 2019-11-05:
Hi, you are using Tasmota 6.6.0 that is quite old and core 2.5.2 that it is known to be buggy.

Please, update to latest Tasmota with core pre2.6.0. Thanks.

Failx92, 2019-11-05:
Hi, I posted the outputs from Version 6.6 thats true but I wrote I tried it with Version 6.7 also.
Today I tried Dev Version 7.0 but the result is the same object temp keeps at -999.0

Failx92, 2019-11-06:
Hi, I tested my MLX90614 sensors with the adafruit library MLX60914, they works fine so I can rule out a hardware issue.

ascillato2, 2019-11-07:
@gemu2015

Please, can you take a look at this?

gemu2015, 2019-11-07:
i have only one mlx90614 but it works fine.
the code is pretty small and the only code difference is that adafruit ignores the error bit which causes the -999 reading in Tasmota.
i don't see what should cause the error bit to be set.
i2c address conflict ??

you may simply remove the error check
the error bit is in the MSB of the raw data

uint16_t uval=read_irtmp(1);
  if (uval&0x8000) {
 // error bit set
  //  obj_temp=-999;
  uval&=0x7fff;
   obj_temp=((float)uval*0.02)-273.15;

  } else {
    obj_temp=((float)uval*0.02)-273.15;
  }

ascillato2, 2019-11-08:
@Failx92 

Hi, any news on this?

ascillato2, 2019-11-09:
@Failx92

Hi, when you have time, please tell us if this fix solved your issue. Thanks

Failx92, 2019-11-09:
Hello Again, sorry for late reply I was busy.

No sorry it don't work after I changed the code to the comment from gemu2015 it show all the time 382.2 degree.

Regards
Failx92

gemu2015, 2019-11-10:
@Failx92 
that's what i expected. the error bit is set.
did you ensure that there is no i2c address conflict?
MPR121 and CCS811 must be undefined because they use the same address.

i personally  #undefine all unused sensors and features to save space and avoid conflicts.

Failx92, 2019-11-10:
@gemu2015 
how do I do that?

MPR121 and CCS811 are undefined by default I changed nothing at this point, I only defined mlx90614

arendst, 2019-11-10:
What is the output of command ``I2cscan``?

arendst, 2019-11-10:
What is the output of command ``I2cDriver``?

Failx92, 2019-11-10:
Hello,

11:11:22 CMD: i2cscan
11:11:22 RSL: stat/nodemcu_v2.7.0/RESULT = {"I2CScan":"Device(s) found at 0x5a"}
11:11:40 CMD: i2cDriver
11:11:40 RSL: stat/nodemcu_v2.7.0/RESULT = {"I2CDriver":"7,8,9,10,11,15,20,32"}

gemu2015, 2019-11-10:
@arendst 
i can confirm that mlx90614 is broken on the last dev version. (just compiled default setup)
my fork is still at 6.7.0 and works.

i am short of time and right now can't check what's changed between 6.7.0 and current.

only  couple of calls used ???

arendst, 2019-11-10:
I'll have a look but cannot test as I don't have a MLX90614 device.

gemu2015, 2019-11-10:
@arendst 
ok reverified with 6.7.0 and my user config disabled and mlx fails also!
there must be an interference with another driver.

sorry i must leave now

@Failx92
here are my undefines in case you want to try.

#undef USE_DISPLAY_MODES1TO5                // Enable display mode 1 to 5 in addition to mode 0
#undef USE_DISPLAY_LCD
#undef USE_DISPLAY_MATRIX
#undef USE_LM75AD
#undef USE_SHT                                // Add I2C emulating code for SHT1X sensor (+1k4 code)
#undef USE_HTU                                // Add I2C code for HTU21/SI7013/SI7020/SI7021 sensor (+1k5 code)
#undef USE_BME680                           // Add additional support for BME680 sensor using Adafruit Sensor and BME680 libraries (+6k code)
#undef USE_BH1750
#undef USE_MHZ19
#undef USE_SENSEAIR
#undef USE_PMS5003
#undef USE_IR_RECEIVE
#undef USE_WS2812
#undef USE_ARILUX_RF
#undef USE_NOVA_SDS
#undef USE_ENERGY_SENSOR
#undef USE_DOMOTICZ
#undef USE_HOME_ASSISTANT
#undef USE_DISCOVERY
#undef USE_DS18x20
#undef USE_SERIAL_BRIDGE
#undef USE_PZEM004T
#undef USE_SHT3X
#undef USE_SGP30
#undef USE_SPS0
#undef USE_SCD30
#undef USE_MLX90614
#undef USE_PZEM2
#undef USE_RF_FLASH
#undef USE_APDS9960
#undef USE_ADE7953
#undef USE_EMULATION_HUE
#undef USE_EMULATION_WEMO
#undef USE_DHT
//#undef USE_COUNTER
#undef USE_TX20_WIND_SENSOR
#undef USE_IR_REMOTE
#undef USE_KNX
#undef USE_HOME_ASSISTANT
#undef USE_PCA9685
#undef USE_TUYA_DIMMER
#undef USE_RC_SWITCH
#undef USE_ARMTRONIX_DIMMERS
#undef USE_PS_16_DZ
#undef USE_SONOFF_IFAN
#undef USE_EMULATION_HUE
#undef USE_LIGHT
#undef USE_EMULATION
#undef USE_HX711
#undef USE_CSE7766
#undef USE_SR04
#undef USE_DS18x20
#undef USE_DS18B20
#undef USE_DS18x20_LEGACY
#undef USE_BMP
#undef USE_DS1820
#undef USE_PZEM004T
#undef USE_PZEM_AC
#undef USE_PZEM_DC
#undef USE_TIMERS
#undef USE_TIMERS_WEBINTERFACE

gemu2015, 2019-11-11:
@arendst 

by disabling all other i2c sensors and selectively enabling  i found SEVERAL dependencies.
(i only testest those by default enabled i2c drivers)

1. switching all others off MLX works
2. USE_SHT,USE_HTU,USE_SHT3X,USE_ADE7953 don't harm
3. USE_BMP disables MLX completely  ???
4. USE_LM75AD causes error -999

the mlx driver is extremely easy and i don't see why it shouldn't work and what causes the interferences 

probably we should be able to disable drivers by command

arendst, 2019-11-11:
Disable/enable i2c drivers using command i2cdriver is available now using I2CDEVICES.md

Failx92, 2019-11-11:
@gemu2015 I tried the disabling but it dont work for me, the sensor isn't recognized but with i2cscan I see the device at 0x5a.

I used the 7.0 dev version and activate the BMP and DS18x20

I'm busy this week, my next tries are on the weekend.

gemu2015, 2019-11-11:
@arendst 
a nice, i missed that because i was busy with other things. 
still no idea what's going wrong here

@Failx92 
so you simply compile with mlx enabled
and then disable one i2c driver after the other until it works.

at least disable these two

i2cdriver10 0
i2cdriver20 0

ascillato2, 2019-11-13:
@Failx92

Hi, any news on this? Could you test with latest version and @gemu2015's suggestions?

ascillato, 2019-11-18:
@Failx92

Hi, any news on this? Could you test with latest version and @gemu2015's suggestions?

Failx92, 2019-11-19:
Hello,

Sorry for my late reply,
I try the latest version tomorrow.

Staars, 2019-11-23:
Okay, joining the party now :)

I can confirm that:

> at least disable these two

> i2cdriver10 0

> i2cdriver20 0

[image]

This is a self compiled firmware with build target "tasmota" and uncommenting  #define USE_MLX90614.

I will try to dig a little bit.

Staars, 2019-11-23:
This is more complex, than I expected.
The sensor normally needs this special restart message after transmission, but at the moment it happily reads without this, which is unexpected for me. (I already turned the reading routine upside-down for testing).
I need more time for this, as (virtually) all driver combinations must be checked. Even then it might result in coincidence.

Staars, 2019-11-25:
@Failx92 

Are you able to compile and test this?

ascillato2, 2019-11-25:
Fix merged.

@Failx92 

Can you confirm if your is solved? Thanks

closed 2019-11-28 by ascillato2

Failx92, 2019-12-08:
Hello,

issue is solved. Thanks.

Karl-opec, 2020-10-11:
Just for Info:

I tried today e self compiled version 9.0.0.2 with
"platformio run -e tasmota-sensors " and enabled MLX90614 
I got -999 OBJ Temp.

with
i2cdriver10 0
i2cdriver20 0

I Got a correct Reading.

The first Reading after Boot is also correct, before it changed to -999

NickyNL, 2023-11-02:
A few years later, running into a similar issue with OBJ temp -999 while the AMB temp is working. I am trying it with the latest tasmota release (13.2.0). Can`t get older versions to work properly yet in gitpod otherwise I would have tried 6.7.0 as that sounded like it worked in the past.

My I2Cscan does find the device; {"I2CScan":"Device(s) found at 0x5a"}
My I2Cdriver usually only shows this; {"I2CDriver":"7,32"}

Does anybody have a version that still works with the MLX90614?

Solar320, 2023-11-08:
> A few years later, running into a similar issue with OBJ temp -999 while the AMB temp is working. I am trying it with the latest tasmota release (13.2.0). Can`t get older versions to work properly [...]
@ESP32 it Work:
[image]

@ESP8266 12E  I got also -999 and CRC errors.  The only workarroud was to use the old Method to readout the MLX90614 . This should work without CRC .

I changed something back into the xsns_46_MLX90614.ino 
void MLX90614_Every_Second(void)
void MLX90614_Every_Second(void)
{
    //mlx90614.i2c_buf = I2cRead24(I2_ADR_IRT, MLX90614_TOBJ1);  //Enable <----------
    mlx90614.value = MLX90614_read16(I2_ADR_IRT, MLX90614_TOBJ1); //Disable <-------
    if (mlx90614.value & 0x8000) {
      mlx90614.obj_temp = -999;
    } else {
      mlx90614.obj_temp = ((float)mlx90614.value * 0.02f) - 273.15f;
    }
    //mlx90614.i2c_buf = I2cRead24(I2_ADR_IRT,MLX90614_TA);   //enable <---------
    mlx90614.value = MLX90614_read16(I2_ADR_IRT, MLX90614_TA); //Disable <--------
    if (mlx90614.value & 0x8000) {
      mlx90614.amb_temp = -999;
    } else {
      mlx90614.amb_temp = ((float)mlx90614.value * 0.02f) - 273.15f;
    }
}

Till now there are still bugs inside. The console shows many sensor timeouts and the Thermostat driver do not detect the sensor values at the moment. "Current Temperature = 0°C"  I would create an IR-Thermostat but no luck.
With BMP180 and DS18B20 it work.

I don't know why it work for him:

quote from Christian Baars
"Finally found my sensor and tested it with ESP8266 and ESP32-C3.
Both work with the current driver for me too.

@Solar320 I fear you have a non-optimal hardware setup, as you see the check of a potential error condition - sadly revealing an error. If you leave out the CRC check (like in the old Tasmota driver or Adafruits library example), then everything seems to work, but in fact it does not. Even consistent wrong values that look plausible are wrong.

Solar320, 2023-11-10:
@ Christian Baars
OK, I found mysterious things. I am now relativ sure that your code has nothing to do with this CRC errors.

Today I tested again this ESP8266-12E without or with very few CRC errors. I do not change something, I only restart this device.
Also I saw that this Obj Temp was jumping up and down 3 Kelvin alle the time. (Sensor was not moved and was directed away from me.)

The Oszilloskop do not bring any news, the voltage spikes at 3,3v were arround +-10mV
Then I Tested an ESP8266-MOD,  again with only +-10mV at 3,3V. 
But the big difference is that the ObjTemp and AMBTemp are stable now, no jumping arround and only <1K difference between the two values. (Same Sensor, same cableconnectors!)
I will make an test over night to confirm this. May be an bug of this ESP  12E Modul.........