arendst/Tasmota #6828: Serial transparent bridge

Matze2018, 2019-11-03:
Is a transparent serial bridge with tasmota possible?
A Transparent serial bridge like 192.168.1.145:23 that can be used as an complete serial port over wifi?
At the moment i can only send some parameters over a terminal program to the tasmota console?

Jason2866, 2019-11-03:
Serial bridge does not work with ports. It uses mqtt for the serialbridge function

ascillato2, 2019-11-03:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-11-03 by ascillato2

Matze2018, 2019-11-03:
Sorry i can´t find a discription how to use mqtt for the serialbridge function.
What i need to set on the mqtt broker?

Jason2866, 2019-11-03:
Nothing :-) you receive the serial received stuff as mqtt message
To send just send the payload with serialsend command over mqtt.
See wiki for usage

Matze2018, 2019-11-04:
I got it! Tank you!!!
My Serial device sends only data if i send a SerialSend X21 to it.
Can i do this on startup once?

Jason2866, 2019-11-04:
Something like that
`Rule1 on system#boot do backlog baudrate 9600; serialsend X21 endon on`

Matze2018, 2019-11-04:
Thank you ... but it is not working :-(
If i put this in the console is it saved automaticly?
Where can i see the saved rules?

Jason2866, 2019-11-04:
You have to enable with `rule1 1` Please read wiki section rules

Matze2018, 2019-11-05:
Now it works!!!
Thank You for your Help!!!