arendst/Tasmota #6830: Please correct template for Maxcio YX-DE04 in Device Wiki

EnerGeh, 2019-11-03:
I bought a pack of brand new Maxcio YX-DE04 plugs (that one with the USB-socket) and found out that the current template is incorrect concerning the GPIOs that control the LED PWM signals.
With the following template you can use the **Color <x>** option to control the LED and it results in correct colors (red, green, blue, white etc.) `{"NAME":"Maxcio YX-DE04","GPIO":[0,17,0,21,56,40,0,0,37,38,39,0,0],"FLAG":1,"BASE":18}`

As i found no relay option that could control the USB-Ports, could you please also add the following additional information: `USB-Ports are always on and not switchable`

And last but not least: the ADC0 is giving some strange values when configured as Analog (1) but I have no idea what it is... maybe anyone else?

ascillato2, 2019-11-03:
Hi,

Thanks for sharing this information. 

Could you please add this information to the templates repository (http://blakadder.github.io/templates)

Thanks.

closed 2019-11-03 by ascillato2

EnerGeh, 2019-11-03:
Wow, you're very fast! In the meanwhile I was wondering if the PWM Channel 4 is really needed as they describe the LED Color as RGB+White (2700K to 6000K). I found out that it doesn't need the 4th channel but I have currently no idea how to control the multi-white LED...