arendst/Tasmota #6831: New Zigbee features

s-hadinger, 2019-11-03:
I'm opening this new Feature Request for discussions about Zigbee evolutions. It will replace discussions happening in #6095 

Z2T (Zigbee 2 Tasmota) is now fully functional, there are many improvements to be added.
https://github.com/arendst/Tasmota/wiki/Zigbee

Here is a list of features I'm planning to add:

- Ability to Read device Attributes with names instead of ZCL codes
- Ability to Write device Attributes with names instead of ZCL codes
- Add SetOption for non-verbose mode
- Add scanning of registered devices when booting (ZigbeeProbe on each device)
- Add persistence of Zigbee devices characteristics
- Add Friendly names to Zigbee devices

In longer term:

- Add support for Zigbee routers (for more than 16 devices)

meingraham, 2019-11-05:
> SetOption for non-verbose mode

I thought we were going to rely on Weblog/SysLog/MQTTLog settings to control verbose logging?

s-hadinger, 2019-11-05:
You are right. Forget the SetOption.

tyjtyj, 2019-11-14:
Not really a feature request but is there somewhere we can host precomplied tasmota-zigbee release and beta. I believe this way we can get more people to join into this project.

meingraham, 2019-11-14:
@tyjtyj 

[Gitpod](https://github.com/arendst/Tasmota/wiki/Compiling-Tasmota-on-Gitpod) makes compiling special binaries painless.

tyjtyj, 2019-11-14:
> 
> 
> @tyjtyj
> 
> [Gitpod](https://github.com/arendst/Tasmota/wiki/Compiling-Tasmota-on-Gitpod) makes compiling special binaries painless.

I know but it still pain for lots of people... just suggestion.. I can host the bin file with md5sum if needed.

meingraham, 2019-11-15:
> it still pain for lots of people...

Really? Yes, only someone else doing it is easier. Create a file with one line. Edit one line in platformio.ini. Compile & download. a 5 minute job.

The problem is keeping it current. Yes, the Zigbee might be somewhat static, but with Tasmota itself continually changing, there may be enhancements or fixes that users of Z2T may need. If you can compile and keep it fairly current and host it, post the URL. I can add it to the wiki.

s-hadinger, 2019-11-16:
I believe it is premature to include a precompiled Zigbee firmware. Some people in the community are working on a pcb and case for Z2T. When it's more mainstream and field tested can we think of adding it.

closed 2019-11-22 by s-hadinger

tyjtyj, 2019-11-23:
Not sure if you missed this 

Can we have this feature ?
"mqtt to enable/disable permit join"

s-hadinger, 2019-11-23:
@tyjtyj Just send the command `ZigbeePermitJoin 1` via MQTT.

Thats topic `cmnd/<your device topic>/ZigbeePermitJoin` with a payload `1`.

tyjtyj, 2019-11-23:
> @tyjtyj Just send the command `ZigbeePermitJoin 1` via MQTT.
> 
> Thats topic `cmnd/<your device topic>/ZigbeePermitJoin` with a payload `1`.

Thanks.. so I managed to pair aqara wall switch(QBKG04LM) but does not seem to be correct

Physical press power ON

```
21:59:00 MQT: tele/zigbee/RESULT = {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x344D","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":107,"securityuse":0,"seqnumber":0,"timestamp":2635883,"fc":"0x18","manuf":"0x0000","transact":17,"cmdid":"0x0A","payload":"0000100100F023004D3403"}}
21:59:00 ZigbeeZCLRawReceived: {"ZigbeeReceived":{"0x344D":{"0006/0000":true,"0006/F000":53759232}}}
21:59:00 MQT: tele/zigbee/SENSOR = {"ZigbeeReceived":{"0x344D":{"0006/F000":53759232,"Power":true,"LinkQuality":107}}}
21:59:00 MQT: tele/zigbee/RESULT = {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x344D","srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":99,"securityuse":0,"seqnumber":0,"timestamp":2636772,"fc":"0x18","manuf":"0x0000","transact":18,"cmdid":"0x0A","payload":"0000100000001001"}}
21:59:00 ZigbeeZCLRawReceived: {"ZigbeeReceived":{"0x344D":{"0006/0000":true}}}
21:59:00 MQT: tele/zigbee/SENSOR = {"ZigbeeReceived":{"0x344D":{"Power":true,"LinkQuality":99}}}
21:59:04 MQT: tele/zigbee/RESULT = {"ZigbeeZCLReceived":{"groupid":0,"clusterid":10,"srcaddr":"0x344D","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":99,"securityuse":0,"seqnumber":0,"timestamp":2648387,"fc":"0x10","manuf":"0x0000","transact":19,"cmdid":"0x00","payload":"0000"}}
21:59:04 ZigbeeZCLRawReceived: {"ZigbeeReceived":{"0x344D":{}}}
21:59:04 MQT: tele/zigbee/SENSOR = {"ZigbeeReceived":{"0x344D":{"LinkQuality":99}}}
```

Physical Power off

```
21:59:08 MQT: tele/zigbee/RESULT = {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x344D","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":105,"securityuse":0,"seqnumber":0,"timestamp":2659620,"fc":"0x18","manuf":"0x0000","transact":20,"cmdid":"0x0A","payload":"0000100000F023004D3403"}}
21:59:08 ZigbeeZCLRawReceived: {"ZigbeeReceived":{"0x344D":{"0006/0000":false,"0006/F000":53759232}}}
21:59:08 MQT: tele/zigbee/SENSOR = {"ZigbeeReceived":{"0x344D":{"0006/F000":53759232,"Power":false,"LinkQuality":105}}}
21:59:08 MQT: tele/zigbee/RESULT = {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x344D","srcendpoint":4,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":102,"securityuse":0,"seqnumber":0,"timestamp":2660501,"fc":"0x18","manuf":"0x0000","transact":21,"cmdid":"0x0A","payload":"0000100000001001"}}
21:59:08 ZigbeeZCLRawReceived: {"ZigbeeReceived":{"0x344D":{"0006/0000":true}}}
21:59:08 MQT: tele/zigbee/SENSOR = {"ZigbeeReceived":{"0x344D":{"Power":true,"LinkQuality":102}}}
```

why I receive power true at 21:59:08

tyjtyj, 2019-11-23:
how to use zigbee send to switch on ? 

This line dont work
`ZigbeeSend { "device":"0x344D", "endpoint":"1", "send":{"Power":1} }`

tyjtyj, 2019-11-23:
Status3

```
        {
            "Device": "0x344D",
            "IEEEAddr": "00158D0001F57034",
            "ModelId": "lumi.ctrl_neutral1",
            "Manufacturer": "LUMI",
            "Endpoints": {
                "0x01": {
                    "ProfileId": "0x0104",
                    "ClustersIn": [
                        "0x0000",
                        "0x0003",
                        "0x0001",
                        "0x0002",
                        "0x0019",
                        "0x000A"
                    ],
                    "ClustersOut": [
                        "0x0000",
                        "0x000A",
                        "0x0019"
                    ]
                },
                "0x02": {
                    "ProfileId": "0x0000",
                    "ClustersIn": [],
                    "ClustersOut": []
                },
                "0x03": {
                    "ProfileId": "0x0000",
                    "ClustersIn": [],
                    "ClustersOut": []
                },
                "0x04": {
                    "ProfileId": "0x0000",
                    "ClustersIn": [],
                    "ClustersOut": []
                },
                "0x05": {
                    "ProfileId": "0x0000",
                    "ClustersIn": [],
                    "ClustersOut": []
                },
                "0x06": {
                    "ProfileId": "0x0000",
                    "ClustersIn": [],
                    "ClustersOut": []
                },
                "0x08": {
                    "ProfileId": "0x0000",
                    "ClustersIn": [],
                    "ClustersOut": []
                }
            }
        }
```

s-hadinger, 2019-11-23:
It should work. Can you put loglevel to 3 and paste the logs of your command?

s-hadinger, 2019-11-23:
I found more information here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/356

Can you try endpoint 2 ?

vMeph, 2020-01-07:
@s-hadinger  any news in `Add Friendly names to Zigbee devices`?
adress devices ex: `0x4872` not to friendly with homeassistant so having  friendly name would rox!!!

s-hadinger, 2020-01-08:
It's still planned but I'm working first on persistence of device data. You wouldn't want to reconfigure friendly names at each reboot...