arendst/Tasmota #6834: Continuously reconnecting device

flummy1978, 2019-11-03:
First of all sorry, for my not perfect english... ist not my foreign language... 

I ve an issue, with one of my sonoff 2.5 since the update on the new version. The device is still connecting the WiFi / MQTT and reconnect this after a couple seconds. Tried already the normaly things like, reboot, reflash... nothing helps.  In the console i can see just all the time the same: 

20:47:10 WIF: verbinden mit AP1 WiFI in Modus 11N als Shelly2-FL...
20:47:14 WIF: verbunden
20:47:15 MQT: Verbindungsversuch...
20:47:15 MQT: verbunden
20:47:15 MQT: tele/Shelly2-FL/LWT = Online (beibehalten)
20:47:15 MQT: cmnd/Shelly2-FL/POWER = 
20:47:35 WIF: verbinden mit AP1 WiFI in Modus 11N als Shelly2-FL...
20:47:38 WIF: verbunden
20:47:39 MQT: Verbindungsversuch...
20:47:39 MQT: verbunden
20:47:39 MQT: tele/Shelly2-FL/LWT = Online (beibehalten)
20:47:39 MQT: cmnd/Shelly2-FL/POWER = 
20:47:57 MQT: Verbindungsversuch...
20:47:57 MQT: Verbindung fehlgeschlagen aufgrund von IP:Port, rc -2. Wiederversuch in 10 s
20:47:58 WIF: verbinden mit AP1 WiFI in Modus 11N als Shelly2-FL...
20:48:02 WIF: verbunden
20:48:13 MQT: Verbindungsversuch...
20:48:18 MQT: Verbindung fehlgeschlagen aufgrund von IP:Port, rc -2. Wiederversuch in 10 s

and its still the same way going on ... 

The Wifi AP is next to this and an addiotional Sonoff 2.5 which is working fine.... The "issue" device, works also until the upgrade of the new firmware .... 

Maybe somone has the same Probelems, or got already a fix, or have an idea what i can try.

Thanks in advance
Andreas

Jason2866, 2019-11-03:
Which router do you have?

Jason2866, 2019-11-03:
Have you erased flash with esptool.py?

flummy1978, 2019-11-03:
Thanks for your response: 
- Connection is on an AVM FritzBox (through an UNIFI Access Point ) 
- No i dont erased the flash ( the sonoff is attached behind the wallswitch, not so easy to access - would like to try first, every other option ) 
- the mqtt broker is fhem on a raspberry and this is already updated and rebootet too :(

Jason2866, 2019-11-03:
Have you tried `reset 5` ?  Will erase ALL settings and flash! Just wifi settings will survive.
DO NOT restore Backup settings. Configure all new "by hand".
If this doesnt solve your issue, dissasembly needs to be done.

flummy1978, 2019-11-03:
I ll try to fix it this way .... 
reset 5: Didn't know that before. Seems the perfekt thing, when you need to reset configuration :) Thanks for that note
Just want to know: 
Why shouldn't restory backup settings ? It works with that configuration until the upgrade and i didnt change anything. If its neccessary i would configure it manually, but just like to know why :)

Jason2866, 2019-11-03:
Because restoring a backup could bring back a defect config

localhost61, 2019-11-03:
At home I've several AP sharing the same SSID but on different channels to cover all areas and the resulting WiFi coverage is not that good.
I got far less overall MQTT disconnect after migrating the RPi from WiFi to ethernet. It was rather easy, (make a backup of the SD card), block WiFi in \boot\config.txt, exchange both static leases in the DHCP server and connect the RPi to a switch.

flummy1978, 2019-11-03:
Between 9:30 PM and 10:30 PM i had 1 hour complete a connected device: 

21:28:01 MQT: Verbindung fehlgeschlagen aufgrund von  IP:Port, rc -2. Wiederversuch in 10 s
21:28:14 MQT: Verbindungsversuch...
21:28:14 MQT: verbunden
21:28:14 MQT: tele/Shelly2-FL/LWT = Online (beibehalten)
21:28:14 MQT: cmnd/Shelly2-FL/POWER = 
22:21:02 MQT: tele/Shelly2-FL/SENSOR = {"Time":"2019-11-03T22:21:02","Switch1":"on","Switch2":"off"}
......
.......... switched couple times the POWER1 of that device 
.....
........ doesen't make anything for maybe 20 minutest 
.......
22:23:31 MQT: Verbindungsversuch...
22:23:31 MQT: verbunden
22:23:31 MQT: tele/Shelly2-FL/LWT = Online (beibehalten)
22:23:31 MQT: cmnd/Shelly2-FL/POWER =
.... And it repeats again the connection... 

i guess its might be an hardware issue and just coincidence, that happens after the upgrade, but i ll try the reset 5 and / or flash fresh, with erasing the memory tomorrow again. 

I m wondering about the coincidence just because the two shelly 2.5 are maybe 1 meter apart, and between that (on the room ceiling) is the unifi access point. 

@localhost61 : The RPi is connected with an LAN cable to the switch, so this WiFi is just used for the sonoff and shelly devices

rominator1983, 2019-11-03:
Do you have many access points in your area? Maybe the channel settings of your (or other) access point cause disconnects because of that.

Normally you can either choose a fixed WIFI-channel or let the access point choose one itself. If someone is using the same channel and powering on the access point only at some times this "could" (should not normally) lead to a problem. This could explain why you had one "random" hour when not seeing problems.
So you could try changing from a manual selected channel to automatic selection or vice-verse. Or you could choose another manual selected channel.

Jason2866, 2019-11-03:
Using Auto Channel is in general not a good idea. Use a wifi analyzer with your smartphone and search the best channel (s) for your APs. Set this channels in you APs fix.

ascillato2, 2019-11-04:
If fixing the wifi channel did not work for you, please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

jj-uk, 2019-11-04:
Are you using a unique 'MqttClient' on the mqtt server? e.g. are there any other devices that might have the same 'MqttClient' connected to the mqtt server?

localhost61, 2019-11-05:
If your MQTT server is mosquitto running on the RPi, then why not monitoring what happens on the server side?
In a terminal session close current mosquitto service and launch it manually with option --verbose, you'll receive on the console every event. When you're done, close the process with Ctrl+C and restart the service...
```
$ sudo su
# systemctl status mosquitto
● mosquitto.service - LSB: mosquitto MQTT v3.1 message broker
   Loaded: loaded (/etc/init.d/mosquitto; generated; vendor preset: enabled)
   Active: active (running) since Wed 2019-10-30 02:23:55 CET; 5 days ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/mosquitto.service
           └─401 /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf

# systemctl stop mosquitto
# mosquitto -v
1572911188: mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000) starting
1572911188: Using default config.
1572911188: Opening ipv4 listen socket on port 1883.
1572911188: Opening ipv6 listen socket on port 1883.
1572911189: New connection from 192.168.2.121 on port 1883.
1572911189: New client connected from 192.168.2.121 as DVES_9E8001 (c1, k30, u'openhab').
1572911189: Sending CONNACK to DVES_9E8001 (0, 0)
1572911189: Received PUBLISH from DVES_9E8001 (d0, q0, r1, m0, 'tele/tasmota/Etg/Ch3/009/shutter/LWT', ... (9 bytes))
1572911189: Received PUBLISH from DVES_9E8001 (d0, q0, r0, m0, 'cmnd/tasmota/Etg/Ch3/009/shutter/POWER', ... (0 bytes))
....
^C1572911392: mosquitto version 1.4.10 terminating
# systemctl start mosquitto
```

flummy1978, 2019-11-05:
First, thanks for all the answers and possible solutions / ways to found the problem... I ll try not to forget any answer, if I do pls remind me about the missing answer ;)

@rominator1983 @Jason2866:
I don't beleave (yeah, I know that's should be done in a church ;)....) the problem is the configuration of the AP, Channel or something else with the wifi. Since I'm using the unifi access points, Ive try all the way to find the best channels, using a analyze tool for the wifi. But nothing works better than using the unifi controller (on another RPi) and keep the channel config as simple as possible. It is only that one device, with a Problem like this.
I would like to try, put another tasmota device in the same place and check what happens. Like I said: I have about 30 different devices in the same wifi with two access points and that one is the only one with mqtt / wifi issues.

@ascillato2 I ll do that, when I try all the possible ways. Because of a long working day, I hope to find enough time today

@jj-uk like I said. About 35 devices summary in the wifi and maybe 30 which using mqtt on the same (FHEM) Server.

@localhost61 I m not a real pro about the configuration and using the RPi. Most of the time I m using copy and paste to config something BUT I try to understand what I m doing. I m not sure if I remember right, but there might be a similar way to check it in fhem too.
Anyway I ll like to try your way too and hope to see a little more, than I know already.

@all like i said... Thanks in advance for the ways for solution and answers. I ll try to check all asap, in case my work let me do that :)

ihavenonick, 2019-11-05:
Dann mal auf deutsch:
Beim Core Wechsel, was bei der neuen Tasmota ab 6.7 der Fall ist, ist es zwingend notwendig ein Reset (blank.bin) durchzuführen oder zumindest ein Reset der Settings, sonst kommt es auf alle Fälle zu seltsamen Verhalten.
Steht aber so auch in allen gängigen Foren die sich ein wenig mit Tasmota beschäftigen und war beim Wechsel des Cores schon immer so.

flummy1978, 2019-11-05:
Mhmmm hätte ich jetzt nicht so erwartet.

Zumal es ja n update von 6.5 bzw 6.6 auf 6.7 war und nicht irgendwie 8 Versionen übersprungen.. Aber auf jeden Fall herzlichen Dank für den Hinweis. Ich werds auch mit der Variante mal testen. Ausbauen muss ich das Ding leider eh :(

Andreas

ascillato2, 2019-11-06:
@flummy1978

Hi, Any news on this? 

If using latest Tasmota from the precompiled bins (that have latest core - pre2.6.0) don't solve your issue, please, complete the troubleshooting template in order to have more information so as to properly help you. Thanks.

closed 2019-11-06 by ascillato2

flummy1978, 2019-11-07:
@ascillato2 and all: 
Sorry i can't give you that fast a result of this. I ve a lot of work this days, so i m very happy when i can say "its couch time" after work, The next 3 days are quite the same, but after that i have some time left to fix it. 
I m very happy to get that much possible answers and sollutions. Thank you all very much. I ll try it an answer asap.

flummy1978, 2019-11-08:
It's solved :) 

First, THANKS A LOT for every single answer, try to help me and sorry it takes that long. My working time makes me very busy, so i cant find any time to try and fix it before ...... 

It seems like @ihavenonick  was right (otherwise I can not explain the coincidence) : 

- tried to reset 5, reconfig manually (not uploading a backup) => Not working
- tried to flash new sonoff-minimal.bin or sonoff.bin or sonoff-DE.bin => Not working

- Remove the shelly from switch can 
- flash a blank.bin
- flash  fresh sonoff-DE.bin
- restore settings =====> It works since 2 hours. 

I m pretty sure it works now correctly, because compared to the last time (when i thought it works again) i have in moren than 2 hours not even one reconnecting try in the log. 

So thanks again for your help, this issue can be closed completely :)

ihavenonick, 2019-11-10:
Wobei das Rückspielen der "alten" Einstellungen genau das Problem wieder verursachen kann!
Besser ist es alle Einstellung frisch zu tätigen.

tomasbedrich, 2022-01-27:
I've just hit the same problem. In my case restricting Tasmota to use b/g Wifi only worked. Command `Wifi 3`