arendst/Tasmota #6835: Displaytext commands for rounded rectangle do not work.

smcgann99, 2019-11-04:
### BUG DESCRIPTION

The two display commands -  up:p:p = draws a rounded rectangle and Up:p:p = draws a filled rounded rectangle, detailed on the wiki are not actually implemented.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic):_wemos D1 mini with ILI9341 LCD display
- [ ] Tasmota binary firmware version number used: __6.7 firmware ___
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:

  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:

  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

  ```

### TO REPRODUCE
displaytext up:p:p 

### EXPECTED BEHAVIOUR
should draw a rounded rectangle 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
both commands are missing from xdrv_13_display and xdsp_04_ili9341

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-11-04:
Hi,

> The two display commands - up:p:p = draws a rounded rectangle and Up:p:p = draws a filled rounded rectangle, detailed on the wiki are not actually implemented.

Sorry. Those are already implemented in the code.

Please, check line 603 of _xdrv_13_display.ino_ file:

https://github.com/arendst/Tasmota/blob/d411f7219d032b9e7f0179af7aac5a45ec53a653/tasmota/xdrv_13_display.ino#L603-L617

Please, try again with latest Tasmota version from dev branch. If still does not work for your, please share the exact commands you are executing or also you can address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

smcgann99, 2019-11-06:
I have now reflashed with tasmota 7.0.0.3 and re-tested 
displaytext [x60y185U72:42:3] - nothing is displayed on screen
displaytext [x60y185u72:42:3] - nothing is displayed on screen
displaytext [x60y185r72:42] - draws a rectangle as expected
displaytext [x60y185R72:42] - draws a filled rectangle as expected

gemu2015, 2019-11-06:
rounded rectangles are only supported by display drivers of class "renderer"
this is not the case with xdsp_04_ili9341!
this remark should be added in the wiki

Jason2866, 2019-11-06:
Wiki entry added

blakadder, 2019-11-06:
Resolved. Closing issue.

closed 2019-11-06 by blakadder