arendst/Tasmota #6837: MQTT Home Assistant not updating config topic

StickeyTape, 2019-11-04:
### BUG DESCRIPTION

With HomeAssistant Auto Discovery enabled (SetOption19), the firmware no longer updates the config topic correctly for the device in MQTT (homeassistant/type/device/config )

Browsing the MQTT server the config topic does not populate and is empty using version beyond 6.6.0.21

Re-flashing devices with version 6.6.0 (17 through to 21) works. Version 6.7.1 stops working, the same for version 7.1.

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Basic
- [X] Tasmota binary firmware version number used: 6.6.0.17 (WORKS), 6.6.0.21 (WORKS), 6.7.1 (NOT WORKING) , 7.1 (NOT WORKING)
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Arduino 1.8.9 and 10.8.10 with 2.5.2 or 2.5.3 ESP library
- [X] Flashing tools used: OTA via Web, flashing minimal, then full firmware
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:

```
 11:23:25 CMD: Backlog Template; Module; GPIO
11:23:25 MQT: frontdoor/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:23:25 MQT: frontdoor/stat/RESULT = {"Module":{"1":"Sonoff Basic"}}
11:23:25 RSL: frontdoor/stat/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}
```

- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``: 
``` N/A```

- [X] Provide the output of this command: ``Status 0``: 
```11:24:06 CMD: status 0
11:24:06 RSL: frontdoor/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Front Door Light"],"Topic":"frontdoor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:24:06 RSL: frontdoor/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"outlights","OtaUrl":"http://192.168.1.252/tasmota/tasmota.ino.generic.bin","RestartReason":"Software/System restart","Uptime":"0T16:18:28","StartupUTC":"2019-11-03T17:05:38","Sleep":50,"CfgHolder":4618,"BootCount":5,"SaveCount":16,"SaveAddress":"F4000"}}
11:24:06 RSL: frontdoor/stat/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-11-03T17:09:45","Boot":31,"Core":"STAGE","SDK":"2.2.1(cfd48f3)"}}
11:24:06 RSL: frontdoor/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"192.168.1.252","LogPort":514,"SSId":["XXXXXXXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400A8009","2805C8000100060000005A00000000000000","00000200"]}}
11:24:06 RSL: frontdoor/stat/STATUS4 = {"StatusMEM":{"ProgramSize":595,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","8FDAE397","003683A1","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
11:24:06 RSL: frontdoor/stat/STATUS5 = {"StatusNET":{"Hostname":"frontdoor-0193","IPAddress":"192.168.1.20","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:3B:80:C1","Webserver":2,"WifiConfig":4}}
11:24:06 RSL: frontdoor/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.252","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B80C1","MqttUser":"DVES_USER","MqttCount":7,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:24:06 RSL: frontdoor/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Nov 04 09:24:06 2019","Local":"Mon Nov 04 11:24:06 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"04:58","Sunset":"18:19"}}
11:24:06 MQT: frontdoor/stat/STATUS10 = {"StatusSNS":{"Time":"2019-11-04T11:24:06"}}
11:24:06 RSL: frontdoor/stat/STATUS11 = {"StatusSTS":{"Time":"2019-11-04T11:24:06","Uptime":"0T16:18:28","UptimeSec":58708,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":7,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXXXX","BSSId":"D8:32:14:53:2C:E1","Channel":2,"RSSI":66,"LinkCount":7,"Downtime":"0T00:01:15"}}} 
```

- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
   N/A
  ```

### TO REPRODUCE

1. Flash device with version 6.6.0.x
2. SetOption19 1
3. Check config branch of device in MQTT
     Note :- config topic present - and has value
4. Flash device with version 6.7.1 or 7.1 minimal
5. Flash device with version 6.7.1 or 7.1 generic
6. Check config branch of device in MQTT
     Note :- config topic - has no value

### EXPECTED BEHAVIOUR

Home Assistant discovery/config is not working in newer version. After upgrading this functionality should still be in place and the MQTT topics updated accordingly. The result is that Homeassistant does not populate with the device.

### SCREENSHOTS

N/A

### ADDITIONAL CONTEXT

Home Assistant versions - tested on 99.1 100, 101 - all exhibit the same behaviour

MQTT - Eclipse Mosquitto

A full erase configuration and complete reconfiguration of the device has also been done

Reverting the configuration to a compiled release < 6.6.21 restores the functionality and populates the config topic correctly.

effelle, 2019-11-04:
Can't replicate, working fine here. 
Tested on `6.6.0.21`, `6.7.1` and latest dev available at the moment (`7.0.0.3`) on 3 different devices/configurations. In all case MQTT Explorer and HAss are receiving and using the `json` correctly. 

```json
08:58:26 MQT: test_hass/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.21(sonoff)","BuildDateTime":"2019-10-31T17:39:21","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}

08:56:46 CMD: setoption19 1
08:56:46 MQT: test_hass/stat/RESULT = {"SetOption19":"ON"}
08:56:46 MQT: homeassistant/light/471078_LI_1/config =  (retained)
08:56:46 MQT: homeassistant/switch/471078_RL_1/config = {"name":"Sonoff","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"471078_RL_1","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]]},"~":"test_hass/"} (retained)
08:56:46 MQT: homeassistant/light/471078_LI_2/config =  (retained)
08:56:46 MQT: homeassistant/switch/471078_RL_2/config =  (retained)
08:56:46 MQT: homeassistant/light/471078_LI_3/config =  (retained)
08:56:46 MQT: homeassistant/switch/471078_RL_3/config =  (retained)
08:56:46 MQT: homeassistant/light/471078_LI_4/config =  (retained)
08:56:46 MQT: homeassistant/switch/471078_RL_4/config =  (retained)
08:56:46 MQT: homeassistant/light/471078_LI_5/config =  (retained)
08:56:46 MQT: homeassistant/switch/471078_RL_5/config =  (retained)
08:56:46 MQT: homeassistant/light/471078_LI_6/config =  (retained)
08:56:46 MQT: homeassistant/switch/471078_RL_6/config =  (retained)
08:56:46 MQT: homeassistant/light/471078_LI_7/config =  (retained)
08:56:46 MQT: homeassistant/switch/471078_RL_7/config =  (retained)
08:56:46 MQT: homeassistant/light/471078_LI_8/config =  (retained)
08:56:46 MQT: homeassistant/switch/471078_RL_8/config =  (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_BTN_1/config =  (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_BTN_2/config =  (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_BTN_3/config =  (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_BTN_4/config =  (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_SW_1/config = {"name":"Sonoff Switch1","stat_t":"~stat/RESULT","value_template":"{{value_json.SWITCH1}}","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"471078_SW_1","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]]},"~":"test_hass/","frc_upd":true,"pl_off":"OFF"} (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_SW_2/config =  (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_SW_3/config =  (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_SW_4/config =  (retained)
08:56:46 MQT: homeassistant/binary_sensor/471078_SW_5/config =  (retained)
08:56:47 MQT: homeassistant/binary_sensor/471078_SW_6/config =  (retained)
08:56:47 MQT: homeassistant/binary_sensor/471078_SW_7/config =  (retained)
08:56:47 MQT: homeassistant/binary_sensor/471078_SW_8/config =  (retained)
08:56:47 MQT: homeassistant/sensor/471078_status/config = {"name":"Sonoff status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","ic":"mdi:information-outline","uniq_id":"471078_status","device":{"identifiers":["471078"],"connections":[["mac","CC:50:E3:47:10:78"]],"name":"Sonoff","model":"Generic","sw_version":"6.6.0.21(sonoff)","manufacturer":"Tasmota"},"~":"test_hass/tele/"} (retained))
```
```json
09:02:03 MQT: sonoff/stat/STATUS2 = {"StatusFWR":{"Version":"6.7.1(sonoff)","BuildDateTime":"2019-10-26T13:19:13","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}

09:03:00 MQT: homeassistant/light/DDDE94_LI_1/config =  (retained)
09:03:00 MQT: homeassistant/switch/DDDE94_RL_1/config = {"name":"Sonoff","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"DDDE94_RL_1","device":{"identifiers":["DDDE94"],"connections":[["mac","CC:50:E3:DD:DE:94"]]},"~":"sonoff/"} (retained)
09:03:00 MQT: homeassistant/light/DCPE18_LI_2/config =  (retained)
09:03:00 MQT: homeassistant/switch/DDDE94_RL_2/config =  (retained)
09:03:00 MQT: homeassistant/light/DDDE94_LI_3/config =  (retained)
09:03:00 MQT: homeassistant/switch/DDDE94_RL_3/config =  (retained)
09:03:00 MQT: homeassistant/light/DDDE94_LI_4/config =  (retained)
09:03:00 MQT: homeassistant/switch/DDDE94_RL_4/config =  (retained)
09:03:00 MQT: homeassistant/light/DDDE94_LI_5/config =  (retained)
09:03:00 MQT: homeassistant/switch/DDDE94_RL_5/config =  (retained)
09:03:00 MQT: homeassistant/light/DDDE94_LI_6/config =  (retained)
09:03:00 MQT: homeassistant/switch/DDDE94_RL_6/config =  (retained)
09:03:00 MQT: homeassistant/light/DDDE94_LI_7/config =  (retained)
09:03:00 MQT: homeassistant/switch/DDDE94_RL_7/config =  (retained)
09:03:00 MQT: homeassistant/light/DDDE94_LI_8/config =  (retained)
09:03:00 MQT: homeassistant/switch/DDDE94_RL_8/config =  (retained)
09:03:00 MQT: homeassistant/binary_sensor/DDDE94_BTN_1/config =  (retained)
09:03:00 MQT: homeassistant/binary_sensor/DDDE94_BTN_2/config =  (retained)
09:03:00 MQT: homeassistant/binary_sensor/DDDE94_BTN_3/config =  (retained)
[... cut ...] ormation-outline","uniq_id":"DFCA18_status","device":{"identifiers":["DFCA18"],"connections":[["mac","CC:50:E3:DD:DE:94"]],"name":"Tasmota","model":"Sonoff Basic","sw_version":"7.0.0.3(b46c6ef-tasmota)","manufacturer":"Tasmota"},"~":"tasmota ... (retained)
```
The latest output is from a Sonoff Basic R2, should match your configuration. 
Since you didn't paste here the output of a working/not working `Setoption 19` I can't really see the issue here. 
Care to explain a bit more what is happening?

StickeyTape, 2019-11-04:
Below is sample of running version 6.6.0.17.

The instance the device boots with this code, it populates MQTT Explorer correctly:

##homeassistant/light/3B80C1_LI_1/config

`{"name":"Front Door Light","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"3B80C1_LI_1","device":{"identifiers":["3B80C1"],"connections":[["mac","5C:CF:7F:3B:80:C1"]]},"~":"frontdoor/"}`

##homeassistant/sensor/3B80C1_status/config

`{"name":"Front Door Light status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"3B80C1_status","device":{"identifiers":["3B80C1"],"connections":[["mac","5C:CF:7F:3B:80:C1"]],"name":"Front Door Light","model":"Sonoff Basic","sw_version":"6.6.0.17(sonoff)","manufacturer":"Tasmota"},"~":"frontdoor/tele/"}`

## Console Output for 6.6.0.17

```
14:44:27 MQT: frontdoor/cmnd/POWER = 
14:44:27 MQT: frontdoor/tele/INFO1 = {"Module":"Sonoff Basic","Version":"6.6.0.17(sonoff)","FallbackTopic":"cmnd/DVES_3B80C1_fb/","GroupTopic":"outlights"}
14:44:27 MQT: frontdoor/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"frontdoor-0193","IPAddress":"192.168.1.20"}
14:44:27 MQT: frontdoor/tele/INFO3 = {"RestartReason":"Software/System restart"}
14:44:27 MQT: frontdoor/stat/RESULT = {"POWER":"OFF"}
14:44:27 MQT: frontdoor/stat/POWER = OFF
14:44:28 MQT: homeassistant/switch/3B80C1_RL_1/config =  (retained)
14:44:28 MQT: homeassistant/light/3B80C1_LI_1/config = {"name":"Front Door Light","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"3B80C1_LI_1","device":{"identifiers":["3B80C1"],"connections":[["mac","5C:CF:7F:3B:80:C1"]]},"~":"frontdoor/"} (retained)
14:44:28 MQT: homeassistant/switch/3B80C1_RL_2/config =  (retained)
14:44:28 MQT: homeassistant/light/3B80C1_LI_2/config =  (retained)
14:44:28 MQT: homeassistant/switch/3B80C1_RL_3/config =  (retained)
14:44:28 MQT: homeassistant/light/3B80C1_LI_3/config =  (retained)
14:44:28 MQT: homeassistant/switch/3B80C1_RL_4/config =  (retained)
14:44:28 MQT: homeassistant/light/3B80C1_LI_4/config =  (retained)
14:44:28 MQT: homeassistant/switch/3B80C1_RL_5/config =  (retained)
14:44:28 MQT: homeassistant/light/3B80C1_LI_5/config =  (retained)
14:44:28 MQT: homeassistant/switch/3B80C1_RL_6/config =  (retained)
14:44:28 MQT: homeassistant/light/3B80C1_LI_6/config =  (retained)
14:44:28 MQT: homeassistant/switch/3B80C1_RL_7/config =  (retained)
14:44:28 MQT: homeassistant/light/3B80C1_LI_7/config =  (retained)
14:44:28 MQT: homeassistant/switch/3B80C1_RL_8/config =  (retained)
14:44:28 MQT: homeassistant/light/3B80C1_LI_8/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_BTN_1/config = {"name":"Front Door Light BTN","stat_t":"0/cmnd/POWER","pl_on":"TOGGLE","avty_t":"frontdoor/tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"3B80C1_BTN_1","device":{"identifiers":["3B80C1"],"connections":[["mac","5C:CF:7F:3B:80:C1"]]},"off_delay":1} (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_BTN_2/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_BTN_3/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_BTN_4/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_SW_1/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_SW_2/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_SW_3/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_SW_4/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_SW_5/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_SW_6/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_SW_7/config =  (retained)
14:44:28 MQT: homeassistant/binary_sensor/3B80C1_SW_8/config =  (retained)
14:44:28 MQT: homeassistant/sensor/3B80C1_status/config = {"name":"Front Door Light status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"3B80C1_status","device":{"identifiers":["3B80C1"],"connections":[["mac","5C:CF:7F:3B:80:C1"]],"name":"Front Door Light","model":"Sonoff Basic","sw_version":"6.6.0.17(sonoff)","manufacturer":"Tasmota"},"~":"frontdoor/tele/"} (retained)
14:44:35 MQT: frontdoor/tele/STATE = {"Time":"2019-11-04T14:44:35","Uptime":"0T00:00:16","UptimeSec":16,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXXXXX","BSSId":"CC:2D:21:67:BB:81","Channel":1,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:06"}}
```

## status 2
```
14:46:54 CMD: status 2
14:46:54 MQT: frontdoor/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.17(sonoff)","BuildDateTime":"2019-10-10T17:29:03","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
```

The same device is then flashed with a 6.7.1 version. I deleted all entries for this device out of MQTT before it rebooted.
1. in MQTT the following topics are missing : 
    * homeassistant/light/3B80C1_LI_1/config
    * homeassistant/sensor/3B80C1_status/config
2. All the other topics created when the device booted, have empty config topics.

[... cut ...] ms like it could be truncated, unless that is just the way the console displays it.  

In the new version the _status/config topic  is 528 chars in total including the date, but the old one 375 chars.  This still does not explain why the  _RL_1/config item does not create in the new version, as it does not have the '...' and is only 11 characters inclusive of the date.

One other observation, in the working version the log displays {time} MQT:  but in the non working version it shows {time }RSL:

Jason2866, 2019-11-04:
Please use a precompiled bin. It could be that your selfcompiled firmware is faulty

effelle, 2019-11-04:
I've used all pre-compiled to test all works fine. Puzzling. 
The tests you done were all made using self-compiled (including 6.7.1)? 
Could you please test the latest prebuild from [here](http://thehackbox.org/tasmota/tasmota.bin)? Did you try to compile your own binaries with [GitPod](https://github.com/arendst/Tasmota/wiki/Compiling-Tasmota-on-Gitpod) (or PlatformIO locally) to see if is an issue with Arduino IDE?
Firmwares 6.6.0.21, 6.7, 6.7.1 and 7.0.0.x, all shares the same code for HA Integration but 6.0.0.17 uses older path/settings. Would be better delete the created sensors with `setoption19 0` to force Tasmota to send an empty JSON to all the possible entities and then restart the device and do a `setoption19 1` again. Give it a try. ;-) 

The truncate result is due to console buffer char limit, you need to check the JSON under MQTT explorer to see if it was well received. Here's a complete one, even if it was truncated on console:
```json
{
  "name": "Tasmota status",
  "stat_t": "~HASS_STATE",
  "avty_t": "~LWT",
  "pl_avail": "Online",
  "pl_not_avail": "Offline",
  "json_attributes_topic": "~HASS_STATE",
  "unit_of_meas": " ",
  "val_tpl": "{{value_json['RSSI']}}",
  "ic": "mdi:information-outline",
  "uniq_id": "XXXXXX_status",
  "device": {
    "identifiers": [
      "XXXXXX"
    ],
    "connections": [
      [
        "mac",
        "CC:50:E3:XX:XX:XX"
      ]
    ],
    "name": "Tasmota",
    "model": "Sonoff Basic",
    "sw_version": "7.0.0.3(b46c6ef-tasmota)",
    "manufacturer": "Tasmota"
  },
  "~": "tasmota/tele/"
}
```
> One other observation, in the working version the log displays {time} MQT: but in the non working version it shows {time }RSL:

This is the only thing I don't understand, as you can see I don't have `RSL` but just `MQT` as should be. If happens just on self-compiled you have something wrong on your configuration.

StickeyTape, 2019-11-04:
Just an update.  I have recreated build environment from scratch:
- Arduino 1.8.10  (No updates installed)
- ESP8266 Boards Manager pointing to 2.5.3 as per documentation (https://github.com/arendst/Tasmota/wiki/Arduino-IDE)
- I have not done a library update (recall having done that on old build environment)
- Only copied the libraries folder across to the Arduino libraries folder as extracted.

Built 7.0.0.3 and uploaded to Sonoff Basic device.

All is working fine with this build.

Looking through the old build log and the new build log, some of the libraries in the old environment were of _newer_ versions.  Seems like possibly some of the newer libraries is to blame for this issue.

In device console I now see the messages correctly reflecting with `MQT:` and not `REL:`

MQTT Explorer now shows the missing elements that were sent using the `REL:` messaging before.

```homeassistant/light/3B80C1_LI_1/config```

```{"name":"Front Door Light","cmd_t":"~cmnd/POWER","stat_t":"~tele/STATE","val_tpl":"{{value_json.POWER}}","pl_off":"OFF","pl_on":"ON","avty_t":"~tele/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"3B80C1_LI_1","device":{"identifiers":["3B80C1"],"connections":[["mac","5C:CF:7F:3B:80:C1"]]},"~":"frontdoor/"}```

Home Assistant is identifying devices correctly.

Looks like build environment versions of libraries were the issue.

{facepalm} should have tried recreating my build environment when I discovered this.

closed 2019-11-04 by StickeyTape

effelle, 2019-11-04:
That's ok, I'm glad you sorted out. 
We started to drop some support for Arduino IDE so I strongly suggest you to start to build on VSC/PlatformIO or using Gitpod. 
For every other question please join us on [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)