arendst/Tasmota #6840: BHT-002-GCLW Thermostat with Tuya Convert

31warren, 2019-11-04:
I have flashed two BHT-002-GCLW (Beca/Tuya) thermostats using Tuya Convert and am now attempting to find the configuration between the ESP and the Thermostat MCU.

The thermostat is sending regular keep alive messages (only) to the configured MQTT broker.

I have configured the module as a Tuya Dimmer(54).

Using Weblog I have captured the messages from the Tuya MCU and ESP as shown below. These have a similar format to those shown in the Wiki but the header and Tuya configuration part of each message is different.

I would be grateful for any advice on the next steps needed to complete the communication with this device.

Thanks.

20:08:23 MQT: tele/sonoff/STATE = {"Time":"2019-11-04T20:08:23","Epoch":1572894503,"Uptime":"0T00:19:25","UptimeSec":1165,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"POWER":"OFF","Dimmer":54,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SKYB8E2E","BSSId":"90:21:06:93:26:ED","Channel":6,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:08:24 TYA: Send "55aa00000000ff"
20:08:24 TYA: RX Packet: "55aa010000010102"
20:08:24 TYA: Heartbeat
20:08:26 TYA: RX Dim State=0
20:08:26 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:26 TYA: RX Dim State=51
20:08:26 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:26 TYA: RX Dim State=0
20:08:26 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:26 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0
20:08:27 TYA: RX Packet: "55aa0107000803020004000000334b"
20:08:27 TYA: RX Dim State=51
20:08:27 TYA: RX Packet: "55aa0107000866020004000000007b"
20:08:27 TYA: RX Dim State=0

Jason2866, 2019-11-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

31warren, 2019-11-04:
Moved to the support chat

closed 2019-11-04 by 31warren