arendst/Tasmota #6842: Tele message will not send after set TelePeriod

qingz2004, 2019-11-05:
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: 7.0.0.3
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: GitPod
- [x] Flashing tools used: OTA
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:

>   10:48:13 MQT: stat/Sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,37,255,255,255,139,255,138,255,255],"FLAG":1,"BASE":18}
> 10:48:14 MQT: stat/Sonoff/RESULT = {"Module":{"18":"Generic"}}
> 10:48:14 MQT: stat/Sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"37":"PWM1"},"GPIO5":{"9":"Switch1"},"GPIO12":{"139":"PN532 Rx"},"GPIO13":{"0":"None"},"GPIO14":{"138":"PN532 Tx"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
> 

  Configuration output here:
- [x] Provide the output of this command: ``Status 0``:
  STATUS 0 output here:

> 10:55:17 MQT: stat/Sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"Sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
> 10:55:17 MQT: stat/Sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:20:04","StartupUTC":"2019-11-05T15:35:13","Sleep":50,"CfgHolder":4617,"BootCount":24,"SaveCount":189,"SaveAddress":"F8000"}}
> 10:55:17 MQT: stat/Sonoff/STATUS2 = {"StatusFWR":{"Version":"7.0.0.3(tasmota)","BuildDateTime":"2019-11-05T15:29:39","Boot":28,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
> 10:55:17 MQT: stat/Sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mySSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000009","2805C8000100060000005AFF000000170700","00100000","00000000"]}}
> 10:55:17 MQT: stat/Sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":585,"Free":416,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":3,"Features":["00000809","8F2AA387","04659381","020886C5","001006C4","00007A93"],"Drivers":"1,2,3,4,5,6,9,10,13,16,17,19,20,22,23,24,28,29","Sensors":"1,2,3,4,5,6,8,9,12,16,22,31,33,40,51,54"}}
> 10:55:17 MQT: stat/Sonoff/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-4984","IPAddress":"192.168.xxx.xxx","Gateway":"192.168.xxx.xxx","Subnetmask":"255.255.255.0","DNSServer":"192.168.xxx.xxx","Mac":"84:F3:EB:xx:xx:xx","Webserver":2,"WifiConfig":4}}
> 10:55:17 MQT: stat/Sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.myserver.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_941378","MqttUser":"Sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
> 10:55:17 MQT: stat/Sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Nov 05 15:55:17 2019","Local":"Tue Nov 05 10:55:17 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 03:00:00 2019","Timezone":99,"Sunrise":"06:47","Sunset":"16:45"}}
> 10:55:17 MQT: stat/Sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-11-05T10:55:17","Switch1":"OFF"}}
> 10:55:17 MQT: stat/Sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-11-05T10:55:17","Uptime":"0T00:20:04","UptimeSec":1204,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"PWM":{"PWM1":0},"Wifi":{"AP":1,"SSId":"mySSID","BSSId":"F0:9F:C2:xx:xx:xx","Channel":1,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:05"}}}
> 

- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:

> 10:35:29 MQT: tele/Sonoff/SENSOR = {"Time":"2019-11-05T10:35:29","Switch1":"OFF"}
> 10:35:59 CMD: TelePeriod
> 10:35:59 MQT: stat/Sonoff/RESULT = {"TelePeriod":300}
> 10:36:03 CMD: TelePeriod 10
> 10:36:03 MQT: stat/Sonoff/RESULT = {"TelePeriod":10}
> 10:42:35 CMD: TelePeriod 300
> 10:42:35 MQT: stat/Sonoff/RESULT = {"TelePeriod":300}
> 10:51:26 CMD: TelePeriod
> 10:51:26 MQT: stat/Sonoff/RESULT = {"TelePeriod":300}
> 

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set TelePeriod to 10

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expect to get tele message every 10 seconds. But waited half hour, no tele message.

Jason2866, 2019-11-05:
Cant reproduce
```
18:12:07 CMD: teleperiod 60
18:12:07 MQT: stat/sonoff-53CBF8/RESULT = {"TelePeriod":60}
18:12:07 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-11-05T18:12:07","Uptime":"0T00:34:11","UptimeSec":2051,"Vcc":3.041,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:06"}}
```
```
18:13:24 CMD: teleperiod 10
18:13:24 MQT: stat/sonoff-53CBF8/RESULT = {"TelePeriod":10}
18:13:25 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-11-05T18:13:25","Uptime":"0T00:35:29","UptimeSec":2129,"Vcc":3.065,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:13:35 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-11-05T18:13:35","Uptime":"0T00:35:39","UptimeSec":2139,"Vcc":3.062,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:13:45 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-11-05T18:13:45","Uptime":"0T00:35:49","UptimeSec":2149,"Vcc":3.038,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:13:55 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-11-05T18:13:55","Uptime":"0T00:35:59","UptimeSec":2159,"Vcc":3.039,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:14:05 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-11-05T18:14:05","Uptime":"0T00:36:09","UptimeSec":2169,"Vcc":3.039,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:14:15 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-11-05T18:14:15","Uptime":"0T00:36:19","UptimeSec":2179,"Vcc":3.038,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:06"}}
18:14:25 MQT: tele/sonoff-53CBF8/STATE = {"Time":"2019-11-05T18:14:25","Uptime":"0T00:36:29","UptimeSec":2189,"Vcc":3.061,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}

```

qingz2004, 2019-11-05:
If nobody has the same problem as me, it is probably my user_config_override.h.
Need to find out which define causes this problem.
It used to work with 6.6.0.12. And broken with 6.7.0. Haven't tried anyone in between.

qingz2004, 2019-11-05:
Found the problem.
It is newly added "#define USE_DEEPSLEEP". If I remove this define and recompile, the problem goes away.

The deepsleep was disabled. DeepSleepTime was set to 0.
Don't know why no tele message if compiled with deepsleep function.

Jason2866, 2019-11-05:
@stefanbode can you take a look at?

stefanbode, 2019-11-06:
Hi all, it worked with my fork (stefanbode). I will now check where the integration gets messed up. Agree, it should work. Will keep you posted

qingz2004, 2019-11-06:
Please make sure if it still works after you issue teleperiod command.
In my environment, as soon as I send this command, no more tele message. I had to restart the device, then it started to send tele messages with the teleperiod set before restart.

dcevedo, 2019-11-06:
I was going to report the same problem when adding deep sleep in the configuration override, the solution was to define deepsleeptime 0 and restart.

Tasmota 7.0.0.1 self-compiled on platform.io, and Wemos D1 mini.

qingz2004, 2019-11-06:
Tested it works for me.
So even deepsleeptime is already set to 0 by default, I still need to manually set it. After restart, tele message works. And I can set teleperiod, and it still works.

stefanbode, 2019-11-07:
With resetting the pre_called to false it now works on my side. I was able to reproduce the issue without the fix.

arendst, 2019-11-07:
@stefanbode the FUNC_PREP_BEFORE_TELEPERIOD function is a deprecated function. If you search the code you'll find only a few sensors are still using it. I understand that DeepSleep must have some way of handling sensors after a sleep but this won't work in the long run.

Your latest fix doesn't work for me either as the command ``Teleperiod 60`` was supposed to set teleperiod to 60 AND report a teleperiod status immediatly. This doesn't work if prep_called = false and the command just set the tele_period variable to it's max.

The deprecation of FUNC_PREP_BEFORE_TELEPERIOD was initiated by the introduction of rules that needed more immediate sensor update data. In a non-sleep system all sensors are probed almost within a few seconds. In a deepsleep system when the system awakes the sensor probe will last a few seconds and then reporting should take place independant on teleperiod. In short, we have to deviate from the current tele_period variable and report all sensors as if was rule handling.

So I guess the deepsleep steps need to be changed in:
- Wake up
- Wait a few seconds for the sensor probes to finish initiated by the rules handler which cycles each sensor one sensor at the time every 100mS.
- Report sensors status with command ``status 10``
- Back to sleep.

This basically means remove prep_called from teleperiod and move to rule handler.

stefanbode, 2019-11-07:
Hi Theo, the function FUNC_PREP_BEFORE_TELEPERIOD can be replaced with a rule on SYSTEM:START. The main feature is to switch on a GPIO that power ON the devices. I'm fine with deleting the hook. Quite common on deepsleep is to use a rule to POWER OFF the device AFTER teleperiod and POWER ON on restart. For energy saving, this is mandatory because the GPIO behaves not all the same with FET transistors. Some go down, some stay up and so on.

arendst, 2019-11-07:
@stefanbode Oh now I start to see the light. After connecting GPIO16 to RST the system is restarted as wakeup, does the init cycle, completes the teleperiod message and goes to sleep again.

I missed the wire between GPIO16 and RST. So forget what I brabbled before. I will dive in further and try to get the prep_called function a bit nicer.

Thx so far. You're done for today ;-)

arendst, 2019-11-07:
@stefanbode I think I solved it.

qingz2004, 2019-11-07:
Confirmed it works for me.

closed 2019-11-07 by qingz2004